ifm/libo3d3xx #2: Unable to talk to camera

shaun-edwards, 2015-04-22:
I have installed the library as discussed in the README.  The library builds just fine, but when I run `make check`, the tests fail.  Below is a network `ping` to prove I can see the camera, and a call to `make check`.  I have verified that I can talk to the sensor via the IFM Windows software, so the sensor is working correctly.  I've tried running the various command line utilities, and they fail similarly with a message about being unable to talk to the sensor.  Any help or recommendations of additional steps is appreciated.
```
ros@ros-ThinkPad-T61:~/libo3d3xx/build$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=64 time=0.505 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=64 time=0.449 ms
64 bytes from 192.168.0.69: icmp_seq=3 ttl=64 time=0.231 ms
^C
--- 192.168.0.69 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1999ms
rtt min/avg/max/mdev = 0.231/0.395/0.505/0.118 ms
ros@ros-ThinkPad-T61:~/libo3d3xx/build$ make check
[  5%] Built target gtest
[ 63%] Built target o3d3xx
[ 68%] Built target gtest_main
[100%] Built target o3d3xx-tests
[==========] Running 41 tests from 5 test cases.
[----------] Global test environment set-up.
[----------] 7 tests from ImageBuffers_Tests
[ RUN      ] ImageBuffers_Tests.Cloud
/home/ros/libo3d3xx/test/o3d3xx-image-tests.cpp:25: Failure
Value of: fg->WaitForFrame(img.get(), 1000)
  Actual: false
Expected: true
/home/ros/libo3d3xx/test/o3d3xx-image-tests.cpp:38: Failure
Value of: fg->WaitForFrame(img2.get(), 1000)
  Actual: false
Expected: true
[  FAILED  ] ImageBuffers_Tests.Cloud (2002 ms)
[ RUN      ] ImageBuffers_Tests.DepthImage
/home/ros/libo3d3xx/test/o3d3xx-image-tests.cpp:62: Failure
Value of: fg->WaitForFrame(img.get(), 1000)
  Actual: false
Expected: true
Floating point exception (core dumped)
```

tpanzarella, 2015-04-22:
Hi Shaun,

I just tried this and the tests pass on my machine. The error above indicates a timeout waiting for data from the camera. However more concerning is the core dump at the end of your pasted text. The library writes logs to /tmp by default. It uses the standard scheme provided by glog, so the .INFO file should contain the most info. Can you look to see if there is any relevant information being logged? Maybe paste that in so I can take a look. I'd like to try to duplicate your issue on my end, so I can get it sorted out for you.

Thanks,
Tom

shaun-edwards, 2015-04-22:
Tom, thanks for the quick reply.  Below is the .INFO file.  Not sure what to make of it.
```
Log file created at: 2015/04/22 10:58:44
Running on machine: ros-ThinkPad-T61
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0422 10:58:44.605388 13793 frame_grabber.cpp:128] Camera connection info: ip=192.168.0.69, port=50010
E0422 10:58:44.625613 13793 camera.hpp:647] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/ -> requestSession : RPC response indicates failure.  XMLRPC method not found (D-Bus error 16)
E0422 10:58:44.626071 13793 frame_grabber.cpp:139] Failed to setup camera for image acquisition: XMLRPC method not found
W0422 10:58:45.595967 13792 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0422 10:58:46.596462 13792 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
I0422 10:58:46.607403 13794 frame_grabber.cpp:128] Camera connection info: ip=192.168.0.69, port=50010
E0422 10:58:46.615350 13794 camera.hpp:647] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/ -> requestSession : RPC response indicates failure.  XMLRPC method not found (D-Bus error 16)
E0422 10:58:46.615538 13794 frame_grabber.cpp:139] Failed to setup camera for image acquisition: XMLRPC method not found
W0422 10:58:47.598381 13792 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
```

tpanzarella, 2015-04-22:
This is helpful Shaun.  

So, some background... the camera uses an XML-RPC server as an out-of-band control channel used to configure the camera settings, turn on/off the framegrabber, etc. This is independent of the TCP connection used to stream the image data. That said, as you can see from the log messages, the XML-RPC URLs have a version number encoded in them. As you can also see it looks like the method we are trying to request is not available. I'm thinking the XML-RPC object we are trying to call that on does not exist and is possibly caused by a different version of the protocol. To that end we have a couple of options to try. First, I'll get in touch with IFM to see if I can flash my camera with the latest firmware as the camera I used to do this development was an early beta version. My understanding is that these things are officially released now and may have undergone further software revisions. I'll get on that. Concurrently, we can "backdoor" a solution here (if it is what I am thinking it is). Can you run the IFM software in a VM or something on your machine, you can set it up to just stream in images. Then in a Linux shell, run `tcpdump` to capture the http traffic to the camera. For example, this would work:

$ sudo tcpdump -s 0 -i wlan1 -A host 192.168.0.69 and tcp port http

You will want to set the interface (-i) and IP correct for your environment. This will let us sniff the wire. What you will see is a jumble of HTTP traffic but in there should be things like (for example):

13:18:55.980656 IP 192.168.0.103.50242 > O3D3XX-21-BA-00.local.http: Flags [P.], seq 1:243, ack 1, win 256, length 242
E...z.@........g...E.B.P4.*.&.6.P....v..POST /api/rpc/v1/com.ifm.efector/ HTTP/1.1
User-Agent: ifmVisionAssistant (via QtNetwork)
Content-Type: application/xml
Content-Length: 114
Connection: Keep-Alive
Accept-Encoding: gzip
Accept-Language: en-US,*
Host: 192.168.0.69:80

13:18:55.982829 IP O3D3XX-21-BA-00.local.http > 192.168.0.103.50242: Flags [.], ack 243, win 473, length 0
E..(..@.@..B...E...g.P.B&.6.4.+.P.............
13:18:55.988606 IP 192.168.0.103.50242 > O3D3XX-21-BA-00.local.http: Flags [P.], seq 243:357, ack 1, win 256, length 114
E...z.@........g...E.B.P4.+.&.6.P....C..<?xml version="1.0" encoding="UTF-8"?><methodCall><methodName>getAllParameters</methodName><params/></methodCall>

This will give us a quick check to see if the XML-RPC URLs have changed from when I wrote the code to the official release of the camera. If this is the case (I'm jumping to conclusions), I could give you a quick fix to apply to your code to get you going then I can commit something more permanent to github. But let's see if my intuition is correct here first ... it may be something else too.

shaun-edwards, 2015-04-22:
@ClayFlannigan (including Clay in on the discussion)

tpanzarella, 2015-04-22:
I just spoke with IFM. I should have an official O3D303 camera to look at tomorrow. So, in the event you don't get around to doing the wire sniffing as described above, I'll see what I can find once the new camera is in my hands and update this thread accordingly.

shaun-edwards, 2015-04-22:
Thanks @tpanzarella.  I'll hold off until tomorrow, as my afternoon is pretty busy.

GarrettPlace, 2015-04-22:
Thank you gentlemen.  I have forward this conversation to our colleagues in Germany and expect they will have a response first thing.  I will keep you updated.

tpanzarella, 2015-04-23:
@GarrettPlace Confirming receipt of the sensor. Thanks for expediting this out.

@shaun-edwards I am now able to duplicate your issue on my machine. This is good. I will work on a fix. I am tied up until about mid-afternoon today, but, as soon as I free up, I will take a look at what is going on. If it is more than a "quick fix", I will update this issue with an estimate on how long it should take and an explanation of my findings.

tpanzarella, 2015-04-23:
On the new O3D303 the XML-RPC call to requestSession requires the second parameter -- despite that the documentation states that it is optional. So, I updated the code to pass an empty string as the second parameter and the test is passing now. There were a few other issues I ran into having to do with some new imager types that are defined in the official camera release. In beta there were 3 now there are 8. With that being said, I think you should be able to use the code now. However, I committed these changes on a branch called `O3D303`. I'd like to do a little bit more testing and regression test this against the beta camera before merging the change on to the master branch.

I also did a quick test to see that the new camera will work with the ROS bindings, and it does. I was able to bring up the camera in RVIZ and look at the various images. I do think more testing needs to be done but as stated above, I think if you download and build the code that is currently in the `O3D303` branch you will be good to play around for a while.

I am going to keep this issue open until @shaun-edwards can test on his end.

shaun-edwards, 2015-04-23:
The `make check` command gets further than before.  Tests still fail, though (not sure if this expected)...see below:
```
 RUN      ] AppImagerTest.AppConfig_JSON
[       OK ] AppImagerTest.AppConfig_JSON (937 ms)
[ RUN      ] AppImagerTest.GetAvailableImagerTypes
/home/ros/libo3d3xx/test/o3d3xx-app-imager-tests.cpp:189: Failure
Value of: 8
Expected: imager_types.size()
Which is: 6
[  FAILED  ] AppImagerTest.GetAvailableImagerTypes (958 ms)
[ RUN      ] AppImagerTest.ChangeImagerType
[       OK ] AppImagerTest.ChangeImagerType (1240 ms)
[ RUN      ] AppImagerTest.GetImagerParameters
[       OK ] AppImagerTest.GetImagerParameters (1503 ms)
[ RUN      ] AppImagerTest.ImagerConfig
[       OK ] AppImagerTest.ImagerConfig (2036 ms)
[ RUN      ] AppImagerTest.ImagerConfigValueOutOfRange
[       OK ] AppImagerTest.ImagerConfigValueOutOfRange (1607 ms)
[ RUN      ] AppImagerTest.ImagerConfig_JSON
[       OK ] AppImagerTest.ImagerConfig_JSON (1178 ms)
[----------] 13 tests from AppImagerTest (13679 ms total)

[----------] Global test environment tear-down
[==========] 41 tests from 5 test cases ran. (28523 ms total)
[  PASSED  ] 40 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] AppImagerTest.GetAvailableImagerTypes
```

tpanzarella, 2015-04-23:
@shaun-edwards I just checked in a change that comments out that check. There seems to be an inconsistency in the number of imager configurations that IFM is shipping with this camera. On my beta unit there were 3. On the O3D303 I got today, there were 8. On yours there appears to be 6. I don't think this should affect your usage of the camera too much and I'll speak with @GarrettPlace about IFM's plans for the number of imager configurations that are available.

For now, you can check out the latest code on the `O3D303` branch and give it a try. Please NOTE: I am at home and w/o a camera so I did not test this change but I expect that it is pretty low risk. Let me know if it works for you. I'll also give it a go in the AM when I am back in the lab.

tpanzarella, 2015-04-24:
I've regression tested this against the beta camera and it is working for me. So, I plan to merge this change into the master and push a new master branch up to github imminently.

@GarrettPlace Can the beta hardware be updated to the latest O3D303 software? As you advised earlier, it is probably not worth keeping a "back port" to the beta unit's software revision as it is obsoleted by the O3D303 platform.

tpanzarella, 2015-04-25:
@shaun-edwards Per our conversation, I think you are up-and-running now with the O3D303. Can you confirm that you can stream images from the camera? I'd like to close this issue and move any new issues that arise to a separate thread. I have added a few additional issues to capture what I plan to currently work on in terms of enhancing this library to ensure all the major features of the camera are available to us in the ROS/PCL/OpenCV communities.

tpanzarella, 2015-05-07:
All indications seem that this issue is resolved. Closing for now.

closed 2015-05-07 by tpanzarella