ifm/libo3d3xx #4: PCIC Output Formatting

tpanzarella, 2015-04-24:
The release of the O3D303 provided an implementation of creating a custom PCIC output format. Getting and setting this format is inherently supported  by `libo3d3xx` however, from a practical standpoint, if it is changed, the `libo3d3xx` frame grabber code will not work as it expects the format that the beta cameras provided. This format also appears to be the default on the O3D303. To stay up-to-date with the latest features of the camera, we should implement this feature into the library.

graugans, 2015-08-18:
I recommend to at least send the configuration of the data you expect to receive to be compatible with any future release of the camera/sensor.

tpanzarella, 2015-08-18:
@graugans I think that is a good idea and I will make that update in the coming days.

tpanzarella, 2015-08-25:
[This commit](https://github.com/lovepark/libo3d3xx/commit/ba9407a67adc71d85e53aca3072601dd2bc64385) implements the change suggested by @graugans 

Beyond setting the schema to a known configuration when the frame grabber is running, this commit also saves and restores the original schema (when the FrameGrabber's dtor is run) in the event it is different from what we expect.

This will be mainlined when I merge dev-0.1.8 to the master.

closed 2015-08-25 by tpanzarella