ifm/libo3d3xx #7: PCL crashes with -std=c++11

tpanzarella, 2015-05-22:
This issue is related to: https://github.com/PointCloudLibrary/pcl/issues/780

Technical details of this issue are articulated at the link above. This project is also affected by this problem. There are a couple of options to work around this:

1) Utilize the [o3d3xx-ros](https://github.com/lovepark/o3d3xx-ros) and do your point cloud processing out-of-process in a separate node that is not compiled with C++11 features. This fits my use cases, and so, this is what I do.

2) Recompile PCL with -std=c++11 (has been documented to work ... see thread in PCL issue above)

3) A derivative of 1 (one) above in the case where you do not want to use ROS would be to serialize your point cloud to some out-of-process code that was compiled without C++11 support.

VictorLamoine, 2015-11-18:
PCL trunk is now fully compatible with C++11, however the 1.7.2 packages remains affected by the bug.
Note that compiling/installing PCL 1.7.2 from sources will also fix the problem (the bug only happens with the binaries).

The future 1.8.0 release should ditch the bug.

tpanzarella, 2015-11-18:
Thanks for the update Victor, this is good news indeed! Will the 1.8.0 binaries be pushed to the Ubuntu 14.04 repos? I'm advocating for our industrial users to stick with LTS releases in the field. Would be great to just `apt-get` the C++11 compatible PCL once ready.

VictorLamoine, 2015-11-18:
Yes this will be pushed through the official PCL ppa.
We don't have a dead-line for that release but I think it will land before 2016.

taketwo, 2016-02-11:
This is not a bug in PCL. Boost library which we heavily rely upon is not ABI compatible between C++03 and C++11 (see [here](http://lists.boost.org/Archives/boost/2013/05/203762.php)). Therefore even after release of 1.8.0 the problem will persist, unless PPA binaries will be built with `-std=c++11` flag (but I did not hear about such plans).

tpanzarella, 2016-02-11:
Thanks for the clarification Sergey. Obviously, not good news. Sorry for my ignorance, but do you know how we could track to see how the PPA binaries are being built so we know for sure?

taketwo, 2016-02-11:
I think @jspricke should be able to comment on this.

jspricke, 2016-02-11:
We are pretty much using the default PCL flags, as you can see here: http://anonscm.debian.org/cgit/debian-science/packages/pcl.git/tree/debian/rules

And it will be C++03 unless someone proves that it's no problem to switch.

tpanzarella, 2016-02-11:
@graugans In the embedded build of PCL, do we have `-std=c++11` on?

graugans, 2016-02-12:
@tpanzarella yes we do [pcl_1.7.1.bb](https://github.com/graugans/meta-ros/blob/869b8ea8b225a60b84ccb7eeff01817e1dfe07aa/recipes-extended/pcl/pcl_1.7.1.bb#L28). If needed we can add ``-std=c++11`` for [boost](https://github.com/openembedded/openembedded-core/blob/541315d6c56df6448f64c262f99d43d5c1e9400b/meta/recipes-support/boost/boost_1.60.0.bb) as-well this should be quite simple.

closed 2020-07-27 by theseankelly