ifm/libo3d3xx #9: Compatibility with IFM's JSON Encoding

tpanzarella, 2015-08-17:
Independent of the JSON encoding developed as part of this project, IFM has developed their own JSON encoding for usage with the camera. We should either move 100% to IFM's JSON encoding OR provide "compatibility switches" within our tool set to enable seamless interoperability between this project and the IFM tools.

graugans, 2015-08-18:
What JSON encoding you are referencing? This [one](https://github.com/lovepark/libo3d3xx/blob/master/doc/camera.json)?

tpanzarella, 2015-08-18:
The JSON encoding I was referring to is *not* the one linked to (that is an early version of this project's JSON encoding), but rather the one specified by IFM's schema -- which Javier just sent over to me yesterday, so, I have only known about it for less than 24 hours. BTW, It is extremely helpful to have this and would love to be kept in the loop (i.e., be sent updates) as the schema evolves. 

My initial thoughts were to try to make this project's JSON encoding compatible with the "official one". However, after thinking about it overnight, I think it is better to keep them independent. I believe the intents are different. With respect to this project, I am using our JSON encoding as a way to declaratively configure the camera -- i.e., "make the camera configuration look like this". This is different than using it as a data serialization scheme to "import/export" applications.

For now, my first pass at supporting the IFM zipped-up JSON files will be to treat them as binary "blobs" and utilize the import/export capabilities of the XML-RPC interface to bring data in/out of the camera.

graugans, 2015-08-18:
We are planning to export them by the internal http server as a download option. BTW you are aware of http://<sensor-ip>/api/rpc/v1/com.ifm.efector/ ? Please ensure to always keep the last '/' otherwise it will not work...

tpanzarella, 2015-08-18:
Thanks for the info Christian. Our [XML-RPC URLs](https://github.com/lovepark/libo3d3xx/blob/master/src/libo3d3xx/camera.cpp#L54) have been working well so far.

When you have an on-camera URL for obtaining the JSON schema, please post it here. That will be very helpful. Thanks!

graugans, 2015-08-18:
Sorry, I had been a bit imprecise you can paste this URL in your browser. The signatures shown is compatible with [DBus](http://dbus.freedesktop.org/doc/dbus-specification.html#type-system)

tpanzarella, 2015-08-18:
OK, cool. I was not aware of that before. Thanks.

tpanzarella, 2015-08-26:
For now, this project is going to keep its own/independent JSON serialization of the camera informed by the IFM schema. As for compatibility with the IFM JSON, version 0.1.8 introduces two new command line tools: `o3d3xx-ifm-import` and `o3d3xx-ifm-export` that can bring in/out IFM JSON files. `o3d3xx-dump` and `o3d3xx-config` will continue to use and expect our own JSON serialization of the camera.

closed 2015-08-26 by tpanzarella