ifm/libo3d3xx #11: Cannot connect to camera

kedecuyp, 2015-09-21:
( could be related to https://github.com/lovepark/libo3d3xx/issues/2)
Hello,

I'm trying to install and configure the libo3dxx library to use with a o3d301 TOF camera. I'm currently using Ubuntu 14.04 - 64 bits
I can ping the camera without problems. I have also verified that i can talk to the camera using the IFM windows utility.

When I run the "make check" command, I get an error :
```
[  4%] Built target gtest
[  8%] Built target gtest_main
[ 65%] Built target o3d3xx
[100%] Built target o3d3xx-tests
[==========] Running 50 tests from 7 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from General_Tests
[ RUN      ] General_Tests.Exceptions
[       OK ] General_Tests.Exceptions (0 ms)
[ RUN      ] General_Tests.Version
[       OK ] General_Tests.Version (0 ms)
[----------] 2 tests from General_Tests (0 ms total)

[----------] 7 tests from ImageBuffers_Tests
[ RUN      ] ImageBuffers_Tests.Cloud
/home/kdc/Downloads/libo3d3xx-master/test/o3d3xx-image-tests.cpp:25: Failure
Value of: fg->WaitForFrame(img.get(), 1000)
  Actual: false
Expected: true
/home/kdc/Downloads/libo3d3xx-master/test/o3d3xx-image-tests.cpp:38: Failure
Value of: fg->WaitForFrame(img2.get(), 1000)
  Actual: false
Expected: true
[  FAILED  ] ImageBuffers_Tests.Cloud (3005 ms)
[ RUN      ] ImageBuffers_Tests.DepthImage
/home/kdc/Downloads/libo3d3xx-master/test/o3d3xx-image-tests.cpp:62: Failure
Value of: fg->WaitForFrame(img.get(), 1000)
  Actual: false
Expected: true
Floating point exception (core dumped)
make[3]: *** [test/CMakeFiles/check] Error 136
make[2]: *** [test/CMakeFiles/check.dir/all] Error 2
make[1]: *** [test/CMakeFiles/check.dir/rule] Error 2
make: *** [check] Error 2
```
I also tried the o3d3xx-viewer utility , but all I get is a black frame.

Here are the contents of the .INFO file:
```
Log file created at: 2015/09/21 11:51:23
Running on machine: kdc-XPS-15-9530
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
W0921 11:51:23.659610  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:24.261695  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:24.866829  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:25.468471  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
E0921 11:51:25.900290  4048 camera.hpp:728] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/ -> getParameter : Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  Operation timed out after 3003 milliseconds with 0 bytes received
E0921 11:51:25.900543  4048 frame_grabber.cpp:123] Could not get IP/Port of the camera: XMLRPC call timed out
W0921 11:51:26.069628  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:26.670647  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:27.272006  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:27.873759  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:28.474962  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:29.077205  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
W0921 11:51:29.577576  4045 frame_grabber.cpp:83] Timeout waiting for image buffer from camera
```

Any help would be appreciated, as right now I'm out of ideas to make this work

tpanzarella, 2015-09-21:
A few things to check here: 

When you say that you can ping the camera, what ip address are you pinging? If it is not 192.168.0.69, then you will have to explicitly specify the camera IP address to libo3d3xx. This can be done in two ways. For purposes of running the unit test, you can set the `O3D3XX_IP` environment variable via `export O3D3XX_IP=<the camera IP>` or for running programs like the viewer, you can specify the `--ip` argument on the command line. All of the command line utilities have a `--help` option explaining what other options the program accepts.

You will also want to make sure you are using the latest release of the libo3d3xx code. As you note, in issue #2 a change in the XML-RPC interface was addressed that prevented the library from connecting to the camera. Releases are listed [here](https://github.com/lovepark/libo3d3xx/releases).

You may also want to try rebooting the camera. If either via libo3d3xx or even the IFM utility, if an active edit session is still alive on the camera that was not explicitly shutdown, then you will not be able to connect. AFAIK, the camera only allows for one active XML-RPC edit session at a time.

Let's address these items first. After that, I'd like to understand a bit more about the core dump you report above. That is more concerning to me as I have not had that happen nor have any other users reported that. I am hoping it may be as a result of aborting the unit test with a Ctrl-C or some other premature means and not an actual bug. Please provide details after we get you past the connection issues.

tpanzarella, 2015-09-21:
As a side note, you will also want to make sure you camera is running a supported IFM firmware. Please look at the software compatibility matrix in the README. For convenience, we provide binaries of the supported firmware in the `firmware` subdirectory of the source distribution of `libo3d3xx`. Before applying it to your camera, make sure you validate its checksum.

kedecuyp, 2015-09-21:
Hello,

I have not changed the IP of the camera : it is still configured as 192.168.0.69. As a matter of fact, I have not modified anything in the configuration or the firmware.

As for the version, I downloaded the master archive in july (as a direct download, not using git). I will try  the latest version this afternoon or tomorrow and come back to you.

Thnks for your answer

kedecuyp, 2015-09-21:
Apparently, the camera is running under the 0.06.13 firmware.
Should I update the firmware using the provided files ?

tpanzarella, 2015-09-21:
I have only used the O3D303, however, my understanding is that the O3D301 is "interface compatible" with the O3D303 it just has a different lens. That said, I'd probably want confirmation from someone at IFM to confirm. Looping in @graugans and @GarrettPlace for confirmation. I'll also see if I can make contact off-line to see if we can get an official word.

tpanzarella, 2015-09-21:
Just got word that both cameras run the same firmware. So, I'd say to go forward with updating to the latest via the supplied firmware in this project. Again, be sure to validate against the checksum before flashing. And be sure to flash using the IFM supplied tools.

kedecuyp, 2015-09-21:
I have updated the firmware to 0.06.39 and downloaded the latest version of libo3d3xx. I have verified that the camera still works using the IFM tool.
I still get the same error messages during make check. I power cycled the camera between the tests to ensure that the connection is shut down.

tpanzarella, 2015-09-21:
Please post your latest logs. Also, since you are getting to the stage of running the unit tests, all of the binaries should be built. Can you also post the output of `o3d3xx-dump`. Thanks.

kedecuyp, 2015-09-21:
Hello,

Looking at the logs, I found that there was a problem in my proxy configuration.
When running make check, everything runs fine except for these few lines :
```
[----------] 2 tests from DenseImageTest
[ RUN      ] DenseImageTest.ActiveApplication
unknown file: Failure
C++ exception with description "XMLRPC finished but the called failed" thrown in SetUp().
unknown file: Failure
C++ exception with description "XMLRPC invalid application index" thrown in TearDown().
[  FAILED  ] DenseImageTest.ActiveApplication (1555 ms)
[ RUN      ] DenseImageTest.ImageSize
unknown file: Failure
C++ exception with description "XMLRPC finished but the called failed" thrown in SetUp().
unknown file: Failure
C++ exception with description "XMLRPC invalid application index" thrown in TearDown().
[  FAILED  ] DenseImageTest.ImageSize (1998 ms)
```

The o3d3xx-viewer command now works as expected.

Should I run some additional tests ? 

Thanks again for your help. Now that I have some results I should be able to use the lib for my applications

tpanzarella, 2015-09-21:
The error above is probably OK, however, I never like it when a test fails. It seems to indicate that the loading of the sample 100k pixel application failed and thus this dense imager cannot be tested against. I'm happy to walk through troubleshooting this with you, if you would like. Also, I would still like to see the output of `o3d3xx-dump`.

Also, glad to hear you are up and running. It would be great to add to the issue any other steps taken to diagnose the proxy issue or the specific log message that led you to that conclusion. Just looking to add to our overall knowledge base here.

If you plan to use libo3d3xx directly, i.e., not within ROS, please be aware of #7 If you use any but the most basic of PCL functions, you will have to face this issue and you should have a strategy for dealing with it. The preferred method is to use the ROS wrapper if you can: https://github.com/lovepark/o3d3xx-ros

graugans, 2015-09-21:
Just for the sake of completness here are the "cameras" they only have different opening angles. 
- [O3D301](http://www.ifm.com/products/us/ds/O3D301.htm)
- [O3D303](http://www.ifm.com/products/us/ds/O3D303.htm)

Additionally we provide a so called sensor which has some build in algorithms like box detection or completness applications. 

- [O3D301](http://www.ifm.com/products/us/ds/O3D300.htm)
- [O3D303](http://www.ifm.com/products/us/ds/O3D302.htm)

They provide aswell the same interface like the simpler cameras but do not provide a default application like the camera does. So maybe the libO3D3xx needs some tweaking.

@kedecuyp you can easily check by using your browser and open the camera ip in the browser. You should see a simple HTML page with "IT Works!" on it. nother simple test is opening a telnet connection to port 50010 and type V?<enter>. The second would have worked in case of your proxy issue the first one not.

kedecuyp, 2015-09-22:
@tpanzarella 
I do not have the .INFO file anymore, but what tipped me off is that the line
```
E0921 11:51:25.900290  4048 camera.hpp:728] http://192.168.0.69:80/api/rpc/v1/com.ifm.efector/ -> getParameter : Unable to transport XML to server and get XML response back.  libcurl failed to execute the HTTP POST transaction, explaining:  Operation timed out after 3003 milliseconds with 0 bytes received
```
Contains additional information about the unreachable address in the latest version of libo3d3xx. There, I could see that this address corresponds to my proxy.

Here are the contents of o3d3xx-dump:
```
kdc@kdc-XPS-15-9530:/opt/libo3d3xx/etc$ o3d3xx-dump
{
    "o3d3xx":
    {
        "libo3d3xx": "109",
        "Date": "Tue Sep 22 08:26:02 2015",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_002_12345678_008022169",
            "Illumination": "#!02_I300_001_12345678_008001175",
            "MACAddress": "00:02:01:40:17:90",
            "Frontend": "#!02_F341_002_12345678_008022593",
            "Diagnose": "#!02_D321_001_12345678_008022060",
            "Connector": "#!02_A300_001_12345678_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "0.06.39",
            "Linux": "Linux version 3.11.0-00001-g4e630e2 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Mon Jun 8 10:25:20 CEST 2015",
            "Diagnostic_Controller": "0.3.2",
            "ELDK": "GOLDENEYE_ELDK-2015-07-01_16-26-37-cc5673110f1bff06b66149c8320f6c1823d87891",
            "Calibration_Version": "0.6.2",
            "Calibration_Device": "00:02:01:40:17:90",
            "Main_Application": "0.4.12",
            "Algorithm_Version": "0.6.18"
        },
        "Device":
        {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "1",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "15",
            "ServiceReportFailedBuffer": "15",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
            "DeviceType": "",
            "ArticleNumber": "",
            "ArticleStatus": "",
            "UpTime": "6.95301075392642e-310",
            "ImageTimestampReference": "10422768",
            "TemperatureFront1": "5.149650179128479e-317",
            "TemperatureFront2": "6.952977167208689e-310",
            "TemperatureIMX6": "5.149768754883481e-317",
            "TemperatureIllu": "5.149887330638483e-317"
        },
        "Net":
        {
[... 39 lines cut ...]
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_moderate",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}

```
Thank you to you both for your help

closed 2015-09-22 by tpanzarella