ifm/libo3d3xx #15: Sequence to resume image grabbing after application edition

semsitivity, 2015-10-01:
This is more a question than an issue, and @graugans  would probably have to line in the documentation answering to that.... but after several tries, I don't see how resume image grabbing after going to EDIT mode to edit camera capture settings.
Here is the code I'm using right now:
```
cam_->RequestSession();	
cam_->SetOperatingMode(o3d3xx::Camera::operating_mode::EDIT);		
o3d3xx::DeviceConfig::Ptr dev = cam_->GetDeviceConfig();

int active_idx = dev->ActiveApplication();
int newapp_idx = cam_->ImportIFMApp(bytes); // bytes contains my app configuration file exported from gui
dev->SetActiveApplication(newapp_idx);
		
for (auto& app : cam_->GetApplicationList())
{								
	if (app.index != newapp_idx)
	{
		cam_->DeleteApplication(app.index);
	}
}

cam_->SaveDevice();	
cam_->StopEditingApplication();		cam_->SetOperatingMode(o3d3xx::Camera::operating_mode::RUN);		
cam_->CancelSession();
```

But after that, my image grabber times out to grab images:
```
fg = std::make_shared<o3d3xx::FrameGrabber>(this->cam_);
buff = std::make_shared<o3d3xx::ImageBuffer>();				
if (!fg->WaitForFrame(buff.get(), SAMPLE_READ_WAIT_TIMEOUT))
	std::cout << "I get a timeout here...." << std::endl;

graugans, 2015-10-01:
@semsitivity maybe your application is not activated? This can be checked by the GUI

tpanzarella, 2015-10-01:
You can also check for the active application via `o3d3xx-ls`. A '*' will appear next to the active application. For more introspection, `o3d3xx-dump` could be used. You will want to make sure that the `TriggerMode` for the active application is set to `"1"` which signifies `FREE_RUN`. This is an assumption made by the frame grabber.

tpanzarella, 2015-10-01:
@semsitivity BTW, this call `cam_->SetOperatingMode(o3d3xx::Camera::operating_mode::RUN);` is not necessary at the end of your edit session above. The frame grabber will explicitly set the camera into `RUN`-mode prior to connecting to the socket.

semsitivity, 2015-10-04:
@graugans, you're right. The application is not activated after this sequence.
If I activate it using the GUI, that's works and images are grabbed.
I try to add a dev->SetActiveApplication(newapp_idx); before saving and closing the edit session, but this does not activate the application.
What is the correct way to activate and application (when no application are activated, because actually, I delete all others).

tpanzarella, 2015-10-04:
@semsitivity The `DeviceConfig` is simply a value object that makes it convenient to collect the device parameters in one spot. You have to explicitly set those parameters on the camera to make them stick, so, you could do something like:

```
dev->SetActiveApplication(newapp_idx);
cam->SetDeviceConfig(dev.get());
cam->SaveDevice();
```

Now, just a general comment... typically to do what you are trying to do you would simply use the supplied command line tools. `o3d3xx-ifm-import` could import your application (it will tell you what index it was added on). Then you could use `o3d3xx-config` to set the active application, something like this would work (assuming you wanted to set the application at index 2 as active):

```
$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"2"}}}' | o3d3xx-config
```  

The design of the library and utilities are such that all the "tweaking" you would need to do should be able to be done via the command line tools, then you simply write code to interface with the frame grabber and run your algorithms. If your use case is more complex, please let me know how we can make things work better for you.

graugans, 2015-10-05:
@tpanzarella those tools are really nice for debugging. It would be nice to have some pre build binaries for Windows. 
A lot of customers are used to tweak the device with their own code.

semsitivity, 2015-10-05:
Not that I don't consider od3xx-config and other bin useful, but I'll probably tweak the device during the runtime according temperatures and image quality (based on bad points count detected). For now I start from a reference application saved as config file, but the plan is to switch or tweak during runtime.
But I'll get more inspired by the sourcecode of those binary, thks!

tpanzarella, 2015-10-06:
@graugans I'm going to close this issue for now. However, feel free to create a new "enhancement" issue for the Windows binaries (RE: [this comment](https://github.com/lovepark/libo3d3xx/issues/15#issuecomment-145449983)) if you think we need it or it will be widely used.

closed 2015-10-06 by tpanzarella