ifm/libo3d3xx #18: Issues trying to build

wegunterjr, 2015-10-20:
I am running the make portion of the build, and I am getting this error:
[ 60%] Built target o3d3xx
Linking CXX executable o3d3xx-config
../libo3d3xx/libo3d3xx.so.0.1.11: undefined reference to `cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
../libo3d3xx/libo3d3xx.so.0.1.11: undefined reference to `cv::normalize(cv::_InputArray const&, cv::_InputOutputArray const&, double, double, int, int, cv::_InputArray const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/bin/o3d3xx-config] Error 1
make[1]: *** [src/bin/CMakeFiles/o3d3xx-config.dir/all] Error 2
make: *** [all] Error 2

Any thoughts?

tpanzarella, 2015-10-20:
Make sure you have OpenCV installed -- I'm using version 2.4. In particular you need to be able to link to:  	libopencv_imgproc.so.2.4 and libopencv_core.so.2.4. This is clearly a linker error. If you think you already have this in place, please rerun `make` like `make VERBOSE=1` and paste in what the actual linker command is -- we need to make sure the LDFLAGS are properly set.

I'm working toward a 1.0 release and on the table is tightening up the build scripts. The top-level CMakeLists.txt should have caught this, but, obviously it did not.

wegunterjr, 2015-10-21:
I have OpenCV 3...is that too new?

wegunterjr, 2015-10-21:
/usr/bin/cmake -H/home/viki/libo3d3xx -B/home/viki/libo3d3xx/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/viki/libo3d3xx/build/CMakeFiles /home/viki/libo3d3xx/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory `/home/viki/libo3d3xx/build'
make -f src/libo3d3xx/CMakeFiles/o3d3xx.dir/build.make src/libo3d3xx/CMakeFiles/o3d3xx.dir/depend
make[2]: Entering directory `/home/viki/libo3d3xx/build'
cd /home/viki/libo3d3xx/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/viki/libo3d3xx /home/viki/libo3d3xx/src/libo3d3xx /home/viki/libo3d3xx/build /home/viki/libo3d3xx/build/src/libo3d3xx /home/viki/libo3d3xx/build/src/libo3d3xx/CMakeFiles/o3d3xx.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/viki/libo3d3xx/build'
make -f src/libo3d3xx/CMakeFiles/o3d3xx.dir/build.make src/libo3d3xx/CMakeFiles/o3d3xx.dir/build
make[2]: Entering directory `/home/viki/libo3d3xx/build'
make[2]: Nothing to be done for `src/libo3d3xx/CMakeFiles/o3d3xx.dir/build'.
make[2]: Leaving directory `/home/viki/libo3d3xx/build'
/usr/bin/cmake -E cmake_progress_report /home/viki/libo3d3xx/build/CMakeFiles  3 4 5 6 7 8 9 10 11 12 13 14 15
[ 35%] Built target o3d3xx
make -f src/bin/CMakeFiles/o3d3xx-config.dir/build.make src/bin/CMakeFiles/o3d3xx-config.dir/depend
make[2]: Entering directory `/home/viki/libo3d3xx/build'
cd /home/viki/libo3d3xx/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/viki/libo3d3xx /home/viki/libo3d3xx/src/bin /home/viki/libo3d3xx/build /home/viki/libo3d3xx/build/src/bin /home/viki/libo3d3xx/build/src/bin/CMakeFiles/o3d3xx-config.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/viki/libo3d3xx/build'
make -f src/bin/CMakeFiles/o3d3xx-config.dir/build.make src/bin/CMakeFiles/o3d3xx-config.dir/build
make[2]: Entering directory `/home/viki/libo3d3xx/build'
Linking CXX executable o3d3xx-config
cd /home/viki/libo3d3xx/build/src/bin && /usr/bin/cmake -E cmake_link_script CMakeFiles/o3d3xx-config.dir/link.txt --verbose=1
/usr/bin/c++   -std=c++11  -Wno-deprecated -O3 -DNDEBUG     CMakeFiles/o3d3xx-config.dir/o3d3xx-config.cpp.o  -o o3d3xx-config  -L/home/viki/libo3d3xx/build/lib  -L/usr/local/lib/libxmlrpc_client.so  -L/usr/local/lib/libxmlrpc_server.so  -L/usr/local/lib/libxmlrpc_server_abyss.so  -L/usr/local/lib/libxmlrpc_cpp.so  -L/usr/local/lib/libxmlrpc.so  -L/usr/local/lib/libxmlrpc_xmlparse.so  -L/usr/local/lib/libxmlrpc_xmltok.so  -L/usr/local/lib/libxmlrpc_util.so  -L/usr/lib/x86_64-linux-gnu/libpthread.so -rdynamic ../libo3d3xx/libo3d3xx.so.0.1.11 -lglog -lboost_system -lboost_program_options -lxmlrpc++ -lxmlrpc_client++ -lboost_system -lboost_program_options -lxmlrpc++ -lxmlrpc_client++ -lboost_filesystem -lboost_thread -lboost_date_time -lboost_iostreams -lboost_serialization -lboost_chrono -lpthread -lpcl_common /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ts.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_superres.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ocl.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_gpu.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.8 /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.8 -Wl,-rpath,/home/viki/libo3d3xx/build/lib:/usr/local/lib/libxmlrpc_client.so:/usr/local/lib/libxmlrpc_server.so:/usr/local/lib/libxmlrpc_server_abyss.so:/usr/local/lib/libxmlrpc_cpp.so:/usr/local/lib/libxmlrpc.so:/usr/local/lib/libxmlrpc_xmlparse.so:/usr/local/lib/libxmlrpc_xmltok.so:/usr/local/lib/libxmlrpc_util.so:/usr/lib/x86_64-linux-gnu/libpthread.so:/home/viki/libo3d3xx/build/src/libo3d3xx: 
make[2]: Leaving directory `/home/viki/libo3d3xx/build'
make[1]: Leaving directory `/home/viki/libo3d3xx/build'

wegunterjr, 2015-10-21:
This is my opencv that is installed (i compiled from source OpenCv 3.0) 
viki@c3po:~/libo3d3xx/build$ dpkg --get-selections | grep opencv
libopencv-calib3d-dev:amd64			install
libopencv-calib3d2.4:amd64			install
libopencv-contrib-dev:amd64			install
libopencv-contrib2.4:amd64			install
libopencv-core-dev:amd64			install
libopencv-core2.4:amd64				install
libopencv-dev					install
libopencv-features2d-dev:amd64			install
libopencv-features2d2.4:amd64			install
libopencv-flann-dev:amd64			install
libopencv-flann2.4:amd64			install
libopencv-gpu-dev:amd64				install
libopencv-gpu2.4:amd64				install
libopencv-highgui-dev:amd64			install
libopencv-highgui2.4:amd64			install
libopencv-imgproc-dev:amd64			install
libopencv-imgproc2.4:amd64			install
libopencv-legacy-dev:amd64			install
libopencv-legacy2.4:amd64			install
libopencv-ml-dev:amd64				install
libopencv-ml2.4:amd64				install
libopencv-objdetect-dev:amd64			install
libopencv-objdetect2.4:amd64			install
libopencv-ocl-dev:amd64				install
libopencv-ocl2.4:amd64				install
libopencv-photo-dev:amd64			install
libopencv-photo2.4:amd64			install
libopencv-stitching-dev:amd64			install
libopencv-stitching2.4:amd64			install
libopencv-superres-dev:amd64			install
libopencv-superres2.4:amd64			install
libopencv-ts-dev:amd64				install
libopencv-ts2.4:amd64				install
libopencv-video-dev:amd64			install
libopencv-video2.4:amd64			install
libopencv-videostab-dev:amd64			install
libopencv-videostab2.4:amd64			install
libopencv2.4-java				install
libopencv2.4-jni				install
opencv-data					install
python-opencv					install
ros-indigo-vision-opencv			install

wegunterjr, 2015-10-21:
okay...this is strange..while looking for my issues with the linker, and making sure I 'really' installed OpenCV 3.0, i decided to download the CMake Gui and used it for the Cmake step
- $ cmake ..  
I put the files into a different folder named build2, and then ran make.  I may have skipped teh make step, but i did the make install, and then ran make check (all the tests passed). I am going to go back now to make sure this isn't a false positive...

tpanzarella, 2015-10-21:
No this makes complete sense. Your source build was likely not installing the cmake package infrastructure our build relies upon. You would have also avoided this by installing the supported binaries via `apt-get`.

wegunterjr, 2015-10-21:
Understood.  I am getting some errors from teh command line tools, but putting it in a new issue. Thanks.

closed 2015-11-02 by tpanzarella