ifm/libo3d3xx #19: command line utility o3d3xx-viewer driver related issue

wegunterjr, 2015-10-21:
Here is the terminal window information:
viki@c3po:~/Documents/pocketholProjects$ o3d3xx-ls
* [1] id=1186278907, name=Sample Application, description=
viki@c3po:~/Documents/pocketholProjects$ o3d3xx-viewer 
libGL error: pci id for fd 5: 80ee:beef, driver (null)
OpenGL Warning: glFlushVertexArrayRangeNV not found in mesa table
OpenGL Warning: glVertexArrayRangeNV not found in mesa table
OpenGL Warning: glCombinerInputNV not found in mesa table
OpenGL Warning: glCombinerOutputNV not found in mesa table
OpenGL Warning: glCombinerParameterfNV not found in mesa table
OpenGL Warning: glCombinerParameterfvNV not found in mesa table
OpenGL Warning: glCombinerParameteriNV not found in mesa table
OpenGL Warning: glCombinerParameterivNV not found in mesa table
OpenGL Warning: glFinalCombinerInputNV not found in mesa table
OpenGL Warning: glGetCombinerInputParameterfvNV not found in mesa table
OpenGL Warning: glGetCombinerInputParameterivNV not found in mesa table
OpenGL Warning: glGetCombinerOutputParameterfvNV not found in mesa table
OpenGL Warning: glGetCombinerOutputParameterivNV not found in mesa table
OpenGL Warning: glGetFinalCombinerInputParameterfvNV not found in mesa table
OpenGL Warning: glGetFinalCombinerInputParameterivNV not found in mesa table
OpenGL Warning: glDeleteFencesNV not found in mesa table
OpenGL Warning: glFinishFenceNV not found in mesa table
OpenGL Warning: glGenFencesNV not found in mesa table
OpenGL Warning: glGetFenceivNV not found in mesa table
OpenGL Warning: glIsFenceNV not found in mesa table
OpenGL Warning: glSetFenceNV not found in mesa table
OpenGL Warning: glTestFenceNV not found in mesa table
libGL error: core dri or dri2 extension not found
libGL error: failed to load driver: vboxvideo
init done 
opengl support available 
OpenCV Error: OpenGL API call (Can't load OpenGL extension [glGenerateMipmap]) in IntGetProcAddress, file /home/viki/Downloads/opencv/opencv-master/modules/core/src/gl_core_3_1.cpp, line 141
/home/viki/Downloads/opencv/opencv-master/modules/core/src/gl_core_3_1.cpp:141: error: (-219) Can't load OpenGL extension [glGenerateMipmap] in function IntGetProcAddress

wegunterjr, 2015-10-21:
issue was related to the VM environment. I was able to check out the guest additions and then ran without issue.

wegunterjr, 2015-10-21:
at this point, I downgraded my Vbox Guest additions (no issues with the driver, just the VM setup) to VBoxGuestAdditions-4.3.12 .

tpanzarella, 2015-10-21:
I have not used `o3d3xx-viewer` in a VM, however it is not a surprise that it does not work. `o3d3xx-viewer` wraps the PCL viewer which in turn wraps VTK hence relies upon graphics acceleration. I think the (lack of) graphics capabilities of the VM is at the root here.

wegunterjr, 2015-10-26:
after following the guidance to install ROS Full Desktop, the steps to create this driver were complete and I was able to run the command line utility o3d3xx-viewer with no problems.

Thanks!

tpanzarella, 2015-10-26:
Good news, and thanks for letting us know that the viewer will run w/o h/w acceleration in a VM. I'll close this issue for now.

closed 2015-10-26 by tpanzarella