ifm/libo3d3xx #20: Setting ActiveApplication w/ o3d3xx-config gives "XMLRPC value out of range"

bigcmos, 2015-11-05:
I was unable to set the ActiveApplication in the Device category from the command line using o3d3xx-config.  Am I doing something wrong?  I pulled this example from the docs: https://github.com/lovepark/libo3d3xx/blob/master/doc/utils.md#o3d3xx-config

$ o3d3xx-dump | grep -i 'ifm_software'
            "IFM_Software": "1.1.288",
$ o3d3xx-ls 
  [1] id=1186278907, name=Sample Application, description=
* [2] id=476707713, name=100K Pixel Sample Application, description=
$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config
Failed to configure camera from JSON config '-':
XMLRPC value out of range
x$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"0"}}}' | o3d3xx-config
Failed to configure camera from JSON config '-':
XMLRPC value out of range
$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"2"}}}' | o3d3xx-config
Failed to configure camera from JSON config '-':
XMLRPC value out of range

I was able to change the ActiveApplication by running o3d3xx-dump, modifying the json and passing the full json file to o3d3xx-config.  This caused the device to reboot though and I would like to have a way to set the application without sending the whole json file.

I also was able to change other parameters successfully in other categories.
echo '{"o3d3xx":{"Net":{"UseDHCP":"true"}}}' | o3d3xx-config

But anything I try in the Device category fails.
$ echo '{"o3d3xx":{"Device":{"Name":"TEST"}}}' | o3d3xx-config
Failed to configure camera from JSON config '-':
XMLRPC value out of range
$ echo '{"o3d3xx":{"Device":{"Description":"TEST"}}}' | o3d3xx-config
Failed to configure camera from JSON config '-':
XMLRPC value out of range

tpanzarella, 2015-11-05:
It looks like you are running a version of the firmware that I have not tested against yet. Our latest supported version is: 0.06.39. It is available in the `firmware` directory of our source distribution. You can flash your camera with that using IFM's Vision Assistant.

Likely the issues you are seeing will require us to update the library for the 1.2.x firmware release (I'm guessing as we have seen similar issues before across firmware versions). You can consult our README file (viewable from Github as well) for a software compatibility matrix that maps `libo3d3xx` library versions to IFM firmware versions that are known to work.

I plan to release a new library version in the next week or so, version 0.2.0, that is planned to be compatible with 1.2.x.

Also, `o3d3xx-config` digesting a full JSON should not have the sensor reboot unless the network configuration was changed. So, I'll look into that as well.

graugans, 2015-11-05:
@tpanzarella @bigcmos is using version ``1.1.288`` which can be officially downloaded from our [website](http://www.ifm.com/ifmus/web/dualis-download.htm) I am back in office on Monday, where i can check what the problem is.  I can not remember that we changed something related to activating applications. 
There had been changes related to application import

tpanzarella, 2015-11-05:
Thanks Christian. Downloading now. I'll flash one of my cameras with this version tomorrow and investigate on my end as well. I'll get this fix in 0.2.0 and in the meantime as I am working on it, I'll push my dev branch out to github so @bigcmos and others can work with it prior to the official release.

I'll update this issue with anything (if anything) I discover while looking at this tomorrow.

tpanzarella, 2015-11-06:
I have pushed by development branch `dev-0.2.0` to github and this issue is now fixed. @bigcmos In the near term, you have two options: 1) use my dev branch until I release 0.2.0 officially or 2) do imports of a complete JSON dump that you modify as you note above. I think the last time that it rebooted for you was due to you changing the default network configuration (i.e., to DHCP) in which case a reboot is required. It should not reboot for you otherwise.

As noted above, I am hoping to release 0.2.0 soon and it will officially support the `1.1.288` firmware. But there is more work to do on the release.

Now the details on what happened. The short version is that the default (max?) parameters for the `ServiceReportPassedBuffer` and `ServiceReportFailedBuffer` have changed from `15` to `5`. It also appears that now `15` is out-of-range for those values. I discovered this by duplicating @bigcmos issue of trying to change the active application. By sniffing the wire with `tcpdump`, I saw this:

```
10:39:49.269636 IP jelly.local.55186 > O3D3XX-40-14-35.local.http: Flags [.], ack 3558, win 299, options [nop,nop,TS val 1816361 ecr 77216], length 0
E..4..@.@......h...E...P....4......+K......
...)..-.
10:39:49.270876 IP jelly.local.55186 > O3D3XX-40-14-35.local.http: Flags [P.], seq 1450:1905, ack 3558, win 299, options [nop,nop,TS val 1816361 ecr 77216], length 455
E.....@.@../...h...E...P....4......+.......
...)..-.POST /api/rpc/v1/com.ifm.efector/session_b2ecb331fb1071cde671219ce0c7e2f0/edit/device/ HTTP/1.1
Host: 192.168.0.69
Accept: */*
Content-Type: text/xml
User-Agent: Xmlrpc-c/1.33.6 Curl/7.35.0
Content-Length: 237

<?xml version="1.0" encoding="UTF-8"?>
<methodCall>
<methodName>setParameter</methodName>
<params>
<param><value><string>ActiveApplication</string></value></param>
<param><value><i4>2</i4></value></param>
</params>
</methodCall>

10:39:49.272067 IP O3D3XX-40-14-35.local.http > jelly.local.55186: Flags [.], ack 1905, win 537, options [nop,nop,TS val 77217 ecr 1816361], length 0
E..4..@.@......E...h.P..4...........H......
..-....)
10:39:49.280117 IP O3D3XX-40-14-35.local.http > jelly.local.55186: Flags [P.], seq 3558:3812, ack 1905, win 537, options [nop,nop,TS val 77218 ecr 1816361], length 254
E..2..@.@......E...h.P..4..................
..-....)HTTP/1.1 200 OK
Status: 200 OK
Content-Length: 112
Content-Type: text/xml
Date: Wed, 01 Jul 2015 14:58:45 GMT
Server: lighttpd/1.4.31

<?xml version="1.0" encoding="UTF-8"?><methodResponse><params><param><value/></param></params></methodResponse>

10:39:49.280571 IP jelly.local.55186 > O3D3XX-40-14-35.local.http: Flags [P.], seq 1905:2369, ack 3812, win 322, options [nop,nop,TS val 1816364 ecr 77218], length 464
E.....@.@..%...h...E...P....4......BAj.....
...,..-.POST /api/rpc/v1/com.ifm.efector/session_b2ecb331fb1071cde671219ce0c7e2f0/edit/device/ HTTP/1.1
Host: 192.168.0.69
Accept: */*
Content-Type: text/xml
User-Agent: Xmlrpc-c/1.33.6 Curl/7.35.0
Content-Length: 246

<?xml version="1.0" encoding="UTF-8"?>
<methodCall>
<methodName>setParameter</methodName>
<params>
<param><value><string>ServiceReportPassedBuffer</string></value></param>
<param><value><i4>15</i4></value></param>
</params>
</methodCall>

10:39:49.290024 IP O3D3XX-40-14-35.local.http > jelly.local.55186: Flags [P.], seq 3812:4235, ack 2369, win 553, options [nop,nop,TS val 77218 ecr 1816364], length 423
E.....@.@..$...E...h.P..4..........)\9.....
..-....,HTTP/1.1 200 OK
Status: 200 OK
Content-Length: 281
Content-Type: text/xml
Date: Wed, 01 Jul 2015 14:58:45 GMT
Server: lighttpd/1.4.31

<?xml version="1.0" encoding="UTF-8"?><methodResponse><fault><value><struct><member><name>faultCode</name><value><int>101002</int></value></member><member><name>faultString</name><value><string>value out of range</string></value></member></struct></value></fault></methodResponse>
```

So, we see from above that the call to actually set the `ActiveApplication` succeeded, but the call to set the `ServiceReportPassedBuffer` failed. The obvious natural question to ask next is: "why was it even trying to change the `ServiceReportPassedBuffer`?". This is an implementation detail in the C++ code. Classes like the `DeviceConfig` are value objects and when they get instantiated the instance variables are initialized to default values. I try to keep them consistent with default values from the sensor. In this version of the code, the default for `ServiceReportPassedBuffer` was `15`. So, when trying to set the `ActiveApplication` which is also an instance variable on the `DeviceConfig` the entire `DeviceConfig` was compared against what was active on the camera and it saw that not only was the `ActiveApplication` different but so was the `ServiceReportPassedBuffer`. It then assumes that you want to change that too. Now, since `15` appears to be out-of-range, we got the error. Perhaps @graugans and I can put our heads together a bit to see how this could be avoided in the future.

bigcmos, 2015-11-06:
Great.  I'll test out your dev branch today.

UseDHCP isn't being changed as it was already set to True.  I am setting it to True again just to see if the command succeeds and I wouldn't expect a reboot if the value doesn't change.  The reboot behavior happens for full JSON and partial JSON configs.  Would you like me to file another bug?  Would you like me to attach any logs or tcp dumps along with it?

tpanzarella, 2015-11-06:
Hmmm. By default I thought these cameras shipped with static IPs and specifically 192.168.0.69. After you update to the development branch, git it another shot. If the problem persists, please open it as a new bug and give me step-by-step instructions on what you are doing. Including "pre" and "post" output from `o3d3xx-dump` will also be very helpful.

bigcmos, 2015-11-07:
The new build fixes this bug.  The device is still rebooting for UseDHCP.  I'll file a new issue for that.

closed 2015-11-07 by tpanzarella