ifm/libo3d3xx #23: o3d303 reboot when setting UseDHCP

bigcmos, 2015-11-07:
When setting UseDHCP to true the device reboots, even if UseDHCP value is not changed.  For the output below I'm built off of origin/dev-0.2.0 @b7f26fa.  UseDHCP is always true and it still reboots.  I noticed that between before and after, origin/dev-0.2.0 resets my StaticIPv4Address to match the IP address that DHCP is assigning.  I updated UseDHCP again to verify that it still reboots as you can see between after and after2 where the StaticIPv4Address stays the same.

After setting UseDHCP I am not able to ping the device until it finishes rebooting.

Updating UseDHCP from origin/master does not modify StaticIPv4Address.  It is very surprising that StaticIPv4Address is modified as a result of updating UseDHCP in origin/dev-0.2.0.  Let me know if you would like me to file another issue around the StaticIPv4Address changing.

See the snippits below where I'm built off of origin/dev-0.2.0 @b7f26fa.

```
$ echo $O3D3XX_IP 
172.16.78.70
$ echo '{"o3d3xx":{"Net":{"UseDHCP":"true"}}}' | ./src/bin/o3d3xx-config
$ ./src/bin/o3d3xx-dump | grep IFM_Software
            "IFM_Software": "1.1.288",
```

Here is the diff:
```
$ diff /tmp/o3d303-02_before.json /tmp/o3d303-02_after.json 
5c5
<         "Date": "Fri Nov  6 16:08:40 2015",
---
>         "Date": "Fri Nov  6 16:42:33 2015",
53,54c53,54
<             "ImageTimestampReference": "10619648",
<             "TemperatureFront1": "5.246921823481704e-317",
---
>             "ImageTimestampReference": "13781808",
>             "TemperatureFront1": "6.80923644613506e-317",
56,57c56,57
<             "TemperatureIMX6": "5.247040399236706e-317",
<             "TemperatureIllu": "5.247158974991708e-317"
---
>             "TemperatureIMX6": "6.809355021890062e-317",
>             "TemperatureIllu": "6.809473597645064e-317"
63c63
<             "StaticIPv4Address": "192.168.0.69",
---
>             "StaticIPv4Address": "172.16.78.70",
```

before
```
$ cat /tmp/o3d303-02_before.json 
{
    "o3d3xx":
    {
        "libo3d3xx": "200",
        "Date": "Fri Nov  6 16:08:40 2015",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02001855_008022954",
            "Illumination": "#!02_I300_001_02025830_008001175",
            "MACAddress": "00:02:01:40:21:02",
            "Frontend": "#!02_F341_002_15_00195_008022593",
            "Diagnose": "#!02_D322_C25_01968001_008022488",
            "Connector": "#!02_A300_001_01867216_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.1.288",
            "Linux": "Linux version 3.11.0-00001-g4e630e2 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Mon Jun 8 10:25:20 CEST 2015",
            "Diagnostic_Controller": "0.3.9",
            "ELDK": "GOLDENEYE_ELDK-2015-07-01_16-26-37-cc5673110f1bff06b66149c8320f6c1823d87891",
            "Calibration_Version": "0.6.6",
            "Calibration_Device": "00:02:01:40:21:02",
            "Main_Application": "0.4.34",
            "Algorithm_Version": "0.6.19"
        },
        "Device":
[... 346 lines cut ...]
                    "MinimumAmplitude": "42",
                    "ReduceMotionArtifacts": "false",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_moderate",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}
```

tpanzarella, 2015-11-07:
OK. Thanks for this. I'll look into it on Monday when I am back in the office. I have an idea of what is going on here -- I think it is similar to your issue with the `ActiveApplication`. Out of curiosity, when you re-import from a full `o3d3xx-dump` does it do the reboot? I'm thinking that if the camera (for whatever reason) when in DHCP mode is filling in the `StaticIPv4Address` with the address acquired from DHCP then when you send in a partial JSON, we once again instantiate a `NetworkConfig` object whose instance variable is initialized to `192.168.0.69` detects this as a difference in network configuration from the sensor, even though DHCP is consistent, and hence forces a reboot of the sensor. If this is the case, it sounds like I need to implement special logic on the `NetworkConfig` vs. the standard logic used on all `XXXConfig` objects.

bigcmos, 2015-11-09:
The device did not reboot if I do a import from full and haven't changed anything in the file.

I thought the case of being initialized to 192.168.0.69 might have been the cause, but if you look at after and after2 the StaticIPv4Address doesn't change and the device still reboots.

Let me know if you need any more information to diagnose.

tpanzarella, 2015-11-10:
I have just committed some code on the `dev-0.2.0` branch that tries to handle network changes in such a way as to avoid unnecessary reboots: d0a81e420820afb1931027f6c8f45fd77723f9fa

If you have the chance to test on your end, I'd appreciate it.

bigcmos, 2015-11-11:
Verified that sending `UseDHCP` with no change does not reboot the device.

closed 2015-11-11 by tpanzarella