ifm/libo3d3xx #24: How to create a custom application to import on the o3d303?

bigcmos, 2015-11-12:
@graugans, @tpanzarella.  Continuing from: https://github.com/lovepark/libo3d3xx/issues/22#issuecomment-156240287

What is the difference between the Default Sample App and 100k.o3d3xxapp? I diffed the o3d3xx-dump of the sample app and 100k and the only difference in the config is the name and ID. I would like to create my own apps 1k, 10k, 100k, etc. I'm familiar with PCL and point cloud decimation, but I would rather the sensor to do the initial work of limiting the data before putting it on the network.

graugans, 2015-11-13:
@bigcmos, As @tpanzarella already explained at the moment we only officially support 23k as a resolution. And a preview for 100k. To enable 100k there is a Output100k option within the parameters for the device. Not application related.

graugans, 2015-11-13:
@bigcmos @tpanzarella in the new Firmware the Output100k is removed and replaced by a resolution parameter. With this you can select weather to use 23k or 100k (96k). The maximum resolution is limited by the imager Chip used (352 x 288) we have to reduce this resolution a little bit due to some optical effects which results in the 23k instead of 25k. The 23k resolution is done by a  2x2 Binning. Nevertheless we had already some requests for lower resolutions. @bigcmos I recommend that you conntact your local ifm sales. An file a request for resolutions smaller than the 23k. At the moment we think reducing the resolutions by powers of 2 would will hopefully enough.

closed 2015-12-05 by tpanzarella