ifm/libo3d3xx #25: Most black objects don't produce good depth data and are invisible.

bigcmos, 2015-11-13:
What are the best settings to use to detect black objects?  I've tried modifying all of the knobs I could think of (e.g. exposure time, number of frequencies, filters, channel) and I can't seem to get good depth data off of many black objects.

TL;DR;
I have a shiny black plastic knight and a matte black computer case which are almost invisible to the sensor.  Some black shirts also exhibit this pattern.  Changing the pose of the objects will change how invisible the object is, but with some objects I found no good poses for capture.

In the data below you will notice that the dark night casts a shadow and has a few depth points.  The computer case is almost invisible and the points I get off of it are random and sporadic.

Please find data below of the issue with the settings I used.  I noticed in the depth image that the dark night is visible in the top right frame, so maybe there is a way to pull this data out?

**invisiblack-o3d303-23k-cloud-20151112.png**
[image]

**invisiblack-o3d303-23k-depthimg-20151112.png**
[image]

**invisiblack-o3d303-100k-cloud-20151112.png**
[image]

**invisiblack-o3d303-rgb-calibration-20151112.png**
[image]

```
{
    "o3d3xx":
    {
        "libo3d3xx": "111",
        "Date": "Thu Nov 12 16:17:30 2015",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02001855_008022954",
            "Illumination": "#!02_I300_001_02025830_008001175",
            "MACAddress": "00:02:01:40:21:02",
            "Frontend": "#!02_F341_002_15_00195_008022593",
            "Diagnose": "#!02_D322_C25_01968001_008022488",
            "Connector": "#!02_A300_001_01867216_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.1.288",
            "Linux": "Linux version 3.11.0-00001-g4e630e2 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Mon Jun 8 10:25:20 CEST 2015",
            "Diagnostic_Controller": "0.3.9",
            "ELDK": "GOLDENEYE_ELDK-2015-07-01_16-26-37-cc5673110f1bff06b66149c8320f6c1823d87891",
            "Calibration_Version": "0.6.6",
            "Calibration_Device": "00:02:01:40:21:02",
            "Main_Application": "0.4.34",
            "Algorithm_Version": "0.6.19"
        },
        "Device":
        {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "2",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "5",
            "ServiceReportFailedBuffer": "5",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
[... 146 lines cut ...]
                    "ReduceMotionArtifacts": "false",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "upto30m_low",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}

```

tpanzarella, 2015-11-13:
In the __invisiblack-o3d303-23k-depthimg-20151112.png__ image you pasted in, the top-right image is a binary encoding of the confidence image that only consults the "valid pixel" bit. In this case, good pixels are green and bad pixels are yellow. You could process the confidence image in your code to get more information on why the given pixel was flagged as invalid. Specifically, in the confidence image bits 0 - 3 (inclusive) give information regarding pixel validity:

| Bit        | Value           |
| ------------- |:-------------:|
| 0  | 1 = pixel invalid |
| 1  | 1 = pixel saturated|
| 2 |1 = bad A-B symmetry|
| 3 | 1 = amplitude below minimum amplitude threshold|

Full description of how to interpret the confidence image is available in the IFM O3D Operations Manual. For the black matte objects that are coming back as bad pixels, I would guess that potentially the amplitude is below the minimum threshold. The Imager configuration of a given application let's you tweak this value via the `MinimumAmplitude` directive. Something to look into I suppose.

NOTE: Right now we are returning the *normalized amplitude* image as the `AmplitudeImage()` on the `ImageBuffer`. In the upcoming `2.0.2` release, I expect to expose both the normalized and raw amplitude.

graugans, 2015-11-13:
@bigcmos the camera is based on a time of flight imager. We are using infra-red LEDs to measure the distance between the camera and the object. The first I learned at ifm during developing software for this kind of devices, objects have different reflectivity in the infra-red spectrum  than in the visible spectrum. So it is always a good idea to check the amplitude for the pixel you are interested in. The more amplitude you have the less distance noise you get. I guess you get ``amplitude below minimum amplitude threshold``for those pixels. One solution is to increase the illumination time or to reduce the distance.
Pixls with ``bad A-B symmetry`` are most probably pixels on a object which was moved during the image was taken. BTW is is the same effect when you try to take photos with your mobile during the night.

graugans, 2015-11-13:
``Changing the pose of the objects will change how invisible the object is, but with some objects I found no good poses for capture.``

You mentioned that the black night is shiny. This may lead to some reflection of the infra-red light in a direction the camera can not catch them any more. This is like pointing a laser pointer to a mirror.

tpanzarella, 2015-12-05:
Hey @bigcmos how are things going on this issue? Do you have further updates? Should we keep it open or close it?

tpanzarella, 2015-12-15:
Closing for now as this is a question and has been inactive for over a month.

closed 2015-12-15 by tpanzarella

bigcmos, 2015-12-16:
Thanks for all of the info.  It's still an issue, but I've put this on the back burner for now.  I will be exploring more object materials next year and will follow up on this after more testing.