ifm/libo3d3xx #26: since update, can't seem to compile

wegunterjr-Dematic, 2015-11-19:
I have been able to compile up to this point, but I decided to retrace my steps and make it repeatable for testing purposes and sharing, and am currently unable to compile. I installed all the pre-reqs, recreating the step where I use full ros install to help, and am now getting this error:
What am I overlooking?
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
 Please set them or make sure they are set and tested correctly in the CMake files:
LIB_glog  linked by target "o3d3xx_shared" in directory /var/jenkins/workspace/AGV_pocketHole/libo3d3xx/src/libo3d3xx
linked by target "o3d3xx_static" in directory /var/jenkins/workspace/AGV_pocketHole/libo3d3xx/src/libo3d3xx

tpanzarella, 2015-11-19:
Seems that CMake can't find libglog.so. Do you have it installed in a standard location? If you are on Ubuntu, you can just:

```
$ sudo apt-get install libgoogle-glog0
```

If you have an older version of `libo3d3xx` already installed, check to see what `libglog` you linked against there. You can do that pretty easily with `ldd`.

wegunterjr-Dematic, 2015-11-19:
ah..i mis-spoke. I didn't install glog prior to cmake.  I will add this step now. Not sure how i missed that. maybe because I was planning on running gtest inside of a cmake configure and overlooked the glog.

wegunterjr-Dematic, 2015-11-19:
Okay...installed glog in the new environment. 
I have version
ldd (Ubuntu EGLIBC 2.19-0ubuntu6.6) 2.19

 Am still getting this errror:
```CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_glog
    linked by target "o3d3xx_shared" in directory /home/wegunterjr/libo3d3xx/src/libo3d3xx
    linked by target "o3d3xx_static" in directory /home/wegunterjr/libo3d3xx/src/libo3d3xx```

tpanzarella, 2015-11-19:
OK. So, where did you put it? Do you install via `apt-get` or did you manually install? If you manually installed, did you put it where your linker can find it? For reference, the official debs put it in: `/usr/lib/x86_64-linux-gnu/`. 

Also, are you building according to the instructions in the README?

wegunterjr-Dematic, 2015-11-19:
Currently building according to the instructions in the README. I installed glog using ```apt-get```

tpanzarella, 2015-11-19:
Can you paste in the output of:

```
$ ls -l /usr/lib/x86_64-linux-gnu/libglog.*
```

wegunterjr-Dematic, 2015-11-19:
```lrwxrwxrwx 1 root root     16 May  1  2013 /usr/lib/x86_64-linux-gnu/libglog.so.0 -> libglog.so.0.0.0
-rw-r--r-- 1 root root 163048 May  1  2013 /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0```

tpanzarella, 2015-11-19:
OK. So, in a fresh check out of the source. Do this:

```
$ mkdir build
$ cd build
$ cmake ..
```

Then immediately after that, paste in the output of:

```
$ cmake -L
```

wegunterjr-Dematic, 2015-11-19:
```CMake Error: The source directory "/home/wegunterjr/libo3d3xx/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
-- Cache values
BOOST_THREAD_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpthread.so
BUILD_EXE_CONFIG:BOOL=ON
BUILD_EXE_DUMP:BOOL=ON
BUILD_EXE_HZ:BOOL=ON
BUILD_EXE_IFM_EXPORT:BOOL=ON
BUILD_EXE_IFM_IMPORT:BOOL=ON
BUILD_EXE_IMAGER_TYPES:BOOL=ON
BUILD_EXE_LS:BOOL=ON
BUILD_EXE_REBOOT:BOOL=ON
BUILD_EXE_RESET:BOOL=ON
BUILD_EXE_RM:BOOL=ON
BUILD_EXE_VERSION:BOOL=ON
BUILD_EXE_VIEWER:BOOL=ON
BUILD_SHARED_LIBS:BOOL=ON
BUILD_STATIC_LIBS:BOOL=ON
BUILD_TESTS:BOOL=ON
CMAKE_BUILD_TYPE:STRING=
CMAKE_INSTALL_PREFIX:PATH=/usr/local
DPKG_CMD:FILEPATH=/usr/bin/dpkg
EIGEN_INCLUDE_DIRS:PATH=/usr/include/eigen3
FLANN_INCLUDE_DIRS:PATH=/usr/include
FLANN_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a
FLANN_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a
LIBUSB_1_INCLUDE_DIR:PATH=/usr/include
LIBUSB_1_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libusb-1.0.so
LIB_boost_program_options:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so
LIB_boost_system:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so
LIB_glog:FILEPATH=LIB_glog-NOTFOUND
LIB_opencv_core:FILEPATH=/usr/lib/x86_64-linux-gnu/libopencv_core.so
LIB_opencv_imgproc:FILEPATH=/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so
LIB_xmlrpc_clientxx:FILEPATH=/usr/local/lib/libxmlrpc_client++.so
LIB_xmlrpcxx:FILEPATH=/usr/local/lib/libxmlrpc++.so
OPENNI2_INCLUDE_DIRS:PATH=/usr/include/openni2
OPENNI2_LIBRARY:FILEPATH=/usr/lib/libOpenNI2.so
OPENNI_INCLUDE_DIRS:PATH=/usr/include/ni
OPENNI_LIBRARY:FILEPATH=/usr/lib/libOpenNI.so
OpenCV_DIR:PATH=/usr/share/OpenCV
PCL_COMMON_INCLUDE_DIR:PATH=/usr/include/pcl-1.7
PCL_DIR:PATH=/usr/share/pcl-1.7
PCL_GEOMETRY_INCLUDE_DIR:PATH=/usr/include/pcl-1.7
PCL_IO_INCLUDE_DIR:PATH=/usr/include/pcl-1.7
PCL_KDTREE_INCLUDE_DIR:PATH=/usr/include/pcl-1.7
PCL_OCTREE_INCLUDE_DIR:PATH=/usr/include/pcl-1.7
PCL_SEARCH_INCLUDE_DIR:PATH=/usr/include/pcl-1.7
PCL_VISUALIZATION_INCLUDE_DIR:PATH=/usr/include/pcl-1.7
XMLRPC_C_CONFIG:FILEPATH=/usr/local/bin/xmlrpc-c-config
gtest_build_samples:BOOL=OFF
gtest_build_tests:BOOL=OFF
gtest_disable_pthreads:BOOL=OFF
gtest_force_shared_crt:BOOL=OFF```

tpanzarella, 2015-11-19:
Now:

```
$ make clean
$ make rebuild_cache
```
Then again, paste in the output of:

```
$ cmake -L
```

wegunterjr-Dematic, 2015-11-19:
I get this:
```make: *** No rule to make target `clean'.  Stop.```

tpanzarella, 2015-11-19:
OK.  Something is wrong. Let's start from the top. Step-by-step. If I can follow your environment set up properly. Let's do this:

```
$ cd /home/wegunterjr/libo3d3xx
$ rm -rf build
$ mkdir build
$ cd build
```

Now, paste in the output of:

```
$ cmake ..
```

wegunterjr-Dematic, 2015-11-19:
```-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   program_options
--   system
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Found eigen: /usr/include/eigen3  
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
-- looking for PCL_COMMON
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/libpcl_common.so;debug;/usr/lib/libpcl_common.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libpthread.so (Required is at least version "1.7.1") 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
-- checking for module 'libopenni'
--   found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- checking for module 'libopenni2'
--   found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- Found libusb-1.0: /usr/include  
-- checking for module 'flann'
--   found flann, version 1.8.4
-- Found Flann: /usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
** WARNING ** visualization features related to pcap will be disabled
** WARNING ** visualization features related to png will be disabled
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- looking for PCL_IO
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- looking for PCL_KDTREE
-- Found PCL_KDTREE: /usr/lib/libpcl_kdtree.so  
-- looking for PCL_GEOMETRY
-- Found PCL_GEOMETRY: /usr/include/pcl-1.7  
-- looking for PCL_SEARCH
-- Found PCL_SEARCH: /usr/lib/libpcl_search.so  
-- looking for PCL_VISUALIZATION
-- Found PCL_VISUALIZATION: /usr/lib/libpcl_visualization.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/libpcl_common.so;debug;/usr/lib/libpcl_common.so;optimized;/usr/lib/libpcl_octree.so;debug;/usr/lib/libpcl_octree.so;/usr/lib/libOpenNI.so;/usr/lib/libOpenNI2.so;vtkCommon;vtkFiltering;vtkImaging;vtkGraphics;vtkGenericFiltering;vtkIO;vtkRendering;vtkVolumeRendering;vtkHybrid;vtkWidgets;vtkParallel;vtkInfovis;vtkGeovis;vtkViews;vtkCharts;optimized;/usr/lib/libpcl_io.so;debug;/usr/lib/libpcl_io.so;optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;optimized;/usr/lib/libpcl_kdtree.so;debug;/usr/lib/libpcl_kdtree.so;optimized;/usr/lib/libpcl_search.so;debug;/usr/lib/libpcl_search.so;optimized;/usr/lib/libpcl_visualization.so;debug;/usr/lib/libpcl_visualization.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/libOpenNI.so;/usr/lib/libOpenNI2.so;optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;vtkCommon;vtkFiltering;vtkImaging;vtkGraphics;vtkGenericFiltering;vtkIO;vtkRendering;vtkVolumeRendering;vtkHybrid;vtkWidgets;vtkParallel;vtkInfovis;vtkGeovis;vtkViews;vtkCharts (Required is at least version "1.7.1") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_glog
    linked by target "o3d3xx_shared" in directory /home/wegunterjr/libo3d3xx/src/libo3d3xx
    linked by target "o3d3xx_static" in directory /home/wegunterjr/libo3d3xx/src/libo3d3xx

-- Configuring incomplete, errors occurred!
See also "/home/wegunterjr/libo3d3xx/build/CMakeFiles/CMakeOutput.log".
See also "/home/wegunterjr/libo3d3xx/build/CMakeFiles/CMakeError.log".
```

tpanzarella, 2015-11-19:
I see the problem (I think). Do this:

```
$ cd /usr/lib/x86_64-linux-gnu/
$ ln -s libglog.so.0 libglog.so
```

Then, restart from the top:

```
$ cd /home/wegunterjr/libo3d3xx
$ rm -rf build
$ mkdir build
$ cd build
$ cmake ..
```

wegunterjr-Dematic, 2015-11-19:
Okay. cmake finished with no errors this time. 
I tried to run make just to see what happens, and I get this:
```Scanning dependencies of target o3d3xx
[  2%] Building CXX object src/libo3d3xx/CMakeFiles/o3d3xx.dir/cmdline_opts.cpp.o
In file included from /home/wegunterjr/libo3d3xx/src/libo3d3xx/cmdline_opts.cpp:23:0:
/home/wegunterjr/libo3d3xx/build/include/o3d3xx/camera.hpp:28:26: fatal error: glog/logging.h: No such file or directory
 #include <glog/logging.h>
                          ^
compilation terminated.
make[2]: *** [src/libo3d3xx/CMakeFiles/o3d3xx.dir/cmdline_opts.cpp.o] Error 1
make[1]: *** [src/libo3d3xx/CMakeFiles/o3d3xx.dir/all] Error 2
make: *** [all] Error 2
```

tpanzarella, 2015-11-19:
Do this:

```
$ sudo apt-get install libgoogle-glog-dev
```
Then try again.

wegunterjr-Dematic, 2015-11-19:
Would that have fixed the linker you suggested above?

wegunterjr-Dematic, 2015-11-19:
Also, I expect I am going to fail with the google tests.  I was planning on installing it during a cmake config, but stopped that to get this working again.   
I found   ```sudo apt-get install libgtest0 libgtest-dev``` , but google suggests downloading it or building it with your project.  Thoughts? Suggestions?

tpanzarella, 2015-11-19:
I suspect it would have fixed the linker issue above, yes. Also, for gtest, just install `libgtest-dev` and our cmake build files will build it for you in the tests directory.

wegunterjr-Dematic, 2015-11-19:
aha...excellent.  Thanks!

wegunterjr-Dematic, 2015-11-19:
Quick review, that linker we did earlier, will it be fixed by installing
``` sudo apt-get install libgoogle-glog-dev```

closed 2015-11-19 by tpanzarella

nathangeorge1, 2017-10-17:
Thanks for your help here @tpanzarella; it might be beneficial to put this in the installation instructions; as I also had to follow this thread to get this package built.