ifm/libo3d3xx #27: Is it possible to disable unit testing if camera not connected

wegunterjr, 2015-11-19:
I am weighing the option of 
1. Downloading from this repo each time I run a build in Jenkins (many times a day - at least as many times as I push changes to my repo)
2.  Downloading the current version and maintaining that until I need to update sometime in the future for new updates, etc
Do you foresee a need to disable unit tests, or tests requiring a camera be connected?
Mostly because it is running full bore while connected (haven't figured out how to turn it off in code just yet)

tpanzarella, 2015-11-19:
Starting with `0.2.0` you can:

```
$ mkdir build
$ cd build
$ cmake -DBUILD_TESTS=OFF ..
$ make
```

Full docs on modularity of build are [here](https://github.com/lovepark/libo3d3xx/blob/master/doc/custom_builds.md)

wegunterjr, 2015-11-19:
copy. That makes sense. I was able to do that, but I plan to include this as a library in a project, so I need to figure out how to just disable the unit tests specific to the camera. Or, at least a way to have the unit tests pass without the camera connected.

tpanzarella, 2015-11-19:
They pretty much all use the camera (mostly). Regardless, if there are specific ones you want on/off, the unit testing framework used is gtest. So, you can enable/disable tests via the `GTEST_FILTER` Environment variable. For example, to skip the `DenseImageTest` unit tests you can:

```
$ GTEST_FILTER=-DenseImageTest.* make check
```

Or, to only run the `DenseImageTest`, just get rid of the `-`.

wegunterjr, 2015-11-19:
Is DBUILD_TESTS just specific to this library? ie, it's not a CMAKE or GoogleTest specific option?
In that case, it could be called BUILD_libo3d3xx_TESTS.
```cmake -DBUILD_TESTS=OFF _MYPROJECT_TESTS=OFF```

tpanzarella, 2015-11-19:
Yes, `BUILD_TESTS

wegunterjr, 2015-11-19:
ah.  I wrote before I read your newest post.  The suggestion to use the GTEST_FILTER should work.

closed 2015-12-05 by tpanzarella