ifm/libo3d3xx #28: make check fails on O3D300 IFM camera

tomatac, 2015-12-05:
I am trying to use your libo3d3xx on github to extract the point cloud from an O3D300 IFM camera.
Following the testing process ($ make check) I am getting some errors:

I am running Ubuntu 10.04 LTS 64 bit.

toma@ubuntu:~/git/libo3d3xx/build$ ping 192.168.0.69
PING 192.168.0.69 (192.168.0.69) 56(84) bytes of data.
64 bytes from 192.168.0.69: icmp_seq=1 ttl=64 time=0.523 ms
64 bytes from 192.168.0.69: icmp_seq=2 ttl=64 time=0.614 ms

toma@ubuntu:~/git/libo3d3xx/build/src/bin$ ./o3d3xx-dump
{
    "o3d3xx":
    {
        "libo3d3xx": "111",
        "Date": "Fri Dec  4 21:41:40 2015",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_002_12345678_008022169",
            "Illumination": "#!02_I300_001_01867212_008001175",
            "MACAddress": "00:02:01:40:1C:71",
            "Frontend": "#!02_F341_002_12345678_008022593",
            "Diagnose": "#!02_D312_C27_12345678_008022683",
            "Connector": "#!02_A310_001_12345678_008022060"
        },
        "SWVersion":
        {
            "IFM_Software": "1.1.288",
            "Linux": "Linux version 3.11.0-00001-g4e630e2 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Mon Jun 8 10:25:20 CEST 2015",
            "Diagnostic_Controller": "0.3.9",
            "ELDK": "GOLDENEYE_ELDK-2015-07-01_16-26-37-cc5673110f1bff06b66149c8320f6c1823d87891",
            "Calibration_Version": "0.6.4",
            "Calibration_Device": "00:02:01:40:1c:71",
            "Main_Application": "0.4.34",
            "Algorithm_Version": "0.6.19"
        },
        "Device":
        {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "0",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "5",
            "ServiceReportFailedBuffer": "5",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
            "DeviceType": "",
            "ArticleNumber": "",
            "ArticleStatus": "",
            "UpTime": "6.95301075392642e-310",
            "ImageTimestampReference": "32232032",
            "TemperatureFront1": "1.592485828261073e-316",
            "TemperatureFront2": "6.952977167208689e-310",
            "TemperatureIMX6": "1.592497685836573e-316",
            "TemperatureIllu": "1.592509543412073e-316"
        },
        "Net":
        {
            "MACAddress": "",
            "NetworkSpeed": "0",
            "StaticIPv4Address": "192.168.0.69",
            "StaticIPv4SubNetMask": "255.255.255.0",
            "StaticIPv4Gateway": "192.168.1.1",
            "UseDHCP": "false"
        },
        "Apps": ""
    }
}

toma@ubuntu:~/git/libo3d3xx/build$ make check
[  4%] Built target gtest
[ 56%] Built target o3d3xx
[ 60%] Built target gtest_main
[100%] Built target o3d3xx-tests
[==========] Running 55 tests from 9 test cases.
[----------] Global test environment set-up.
[----------] 4 tests from TemporalFilterTest
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_General
[       OK ] TemporalFilterTest.TemporalFilterConfig_General (237 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameters
unknown file: Failure
C++ exception with description "XMLRPC invalid application index" thrown in the test body.
[  FAILED  ] TemporalFilterTest.GetTemporalFilterParameters (982 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameterLimits
unknown file: Failure
C++ exception with description "XMLRPC invalid application index" thrown in the test body.
[  FAILED  ] TemporalFilterTest.GetTemporalFilterParameterLimits (1000 ms)
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_JSON
unknown file: Failure
C++ exception with description "XMLRPC invalid application index" thrown in the test body.
[  FAILED  ] TemporalFilterTest.TemporalFilterConfig_JSON (1009 ms)
[----------] 4 tests from TemporalFilterTest (3229 ms total)

[----------] 1 test from ImportExport_Tests
[ RUN      ] ImportExport_Tests.ImportExportApp
/home/toma/git/libo3d3xx/test/o3d3xx-import-export-tests.cpp:22: Failure
Expected: bytes = cam->ExportIFMApp(active_idx) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] ImportExport_Tests.ImportExportApp (508 ms)
[----------] 1 test from ImportExport_Tests (508 ms total)

[----------] 8 tests from ImageBuffers_Tests
[ RUN      ] ImageBuffers_Tests.Cloud
/home/toma/git/libo3d3xx/test/o3d3xx-image-tests.cpp:28: Failure
Value of: fg->WaitForFrame(img.get(), 1000)
  Actual: false
Expected: true
/home/toma/git/libo3d3xx/test/o3d3xx-image-tests.cpp:41: Failure
Value of: fg->WaitForFrame(img2.get(), 1000)
  Actual: false
Expected: true
[  FAILED  ] ImageBuffers_Tests.Cloud (2002 ms)
[ RUN      ] ImageBuffers_Tests.XYZImage
/home/toma/git/libo3d3xx/test/o3d3xx-image-tests.cpp:65: Failure
Value of: fg->WaitForFrame(buff.get(), 1000)
  Actual: false
Expected: true
[  FAILED  ] ImageBuffers_Tests.XYZImage (1002 ms)
[ RUN      ] ImageBuffers_Tests.DepthImage
/home/toma/git/libo3d3xx/test/o3d3xx-image-tests.cpp:120: Failure
Value of: fg->WaitForFrame(img.get(), 1000)
  Actual: false
Expected: true
Floating point exception (core dumped)
make[3]: *** [test/CMakeFiles/check] Error 136
make[2]: *** [test/CMakeFiles/check.dir/all] Error 2
make[1]: *** [test/CMakeFiles/check.dir/rule] Error 2
make: *** [check] Error 2

tpanzarella, 2015-12-05:
It appears as though there are no applications loaded on the camera. The best thing to do here would be to set it back to factory defaults and rerun the test suite. You can do this with the `o3d3xx-reset` command line tool.

graugans, 2015-12-05:
@tomatac @tpanzarella the O3D300 is a sensor not a camera. You can easily
check this by counting the pins on the connector.

5 PIN connector -> Camera
8 PIN connector -> Sensor

The Sensor has two apps pre installed one for dimensioning and the other
one for completeness. But the sensor does not have an application pre
installed. You can try to create one with the command line tools.

tomatac, 2015-12-05:
I ran o3d3xx-reset and "make check" after that. "make check" reported the same errors.
I do have the sensor but IFM said that it should work.
How do I create and install an app on the sensor?
The sensor is running a version of Linux. How do I login into the sensor, SSH? What is the default password?

graugans, 2015-12-05:
@tomatac I'll check this in the office on monday. At the moment I have no 03D300 available. O3D300 is compatible with O3D303. The main difference is the missing default camera application.   

Sure, the device is running an embedded linux. The ssh access is only available for ifm maintenance purposes.

tpanzarella, 2015-12-05:
Attached is a JSON dump of the camera created via `o3d3xx-dump`. This is running IFM firmware `1.1.288` and was created with `libo3d3xx` version `0.2.0`. You should ensure that is what your s/w stack looks like. This has one application installed. The default as shipped with the **camera** product and not the **sensor** product. I have not used the **sensor** at all and @graugans would be the authority on noted the differences. However, in theory, you should be able to take the attached `camera.json` file and load it on your **camera** like:

```
$ o3d3xx-config < camera.json
```

After you do that, please post the output of `o3d3xx-dump`.

NOTE: The attached file has the `.txt` extension so it is compatible with github attachments.

[camera.json.txt](https://github.com/lovepark/libo3d3xx/files/52886/camera.json.txt)

graugans, 2015-12-05:
@tomatac @tpanzarella you may have to edit the JSON and change O3D303 -> O3D300

tpanzarella, 2015-12-05:
I suppose an alternative to having to edit the JSON (not sure where it would need to be edited as this is the `libo3d3xx` JSON serialization which basically gets turned into a bunch of XMLRPC calls internally by the library), you could use the command line tools to simply add an application. You can do that with the following command line:

```
$ echo '{"o3d3xx":{"Apps":[{}]}}' | o3d3xx-config
```

This will create the application with all of the camera default settings applied. Most notably an imager configuration of `under5m_moderate` running at 5Hz.

tomatac, 2015-12-05:
Thank you very much for your feedback!
I think that the first step I should do is to back up the state of the camera. How do I do that?
For me it is very important to be able to use the sensor applications also.
This is my test camera and I use it to evaluate application. Some applications are simple enough and the sensor applications are good enough, other are more complex and I am looking to use OpenCV.
Is there anywhere I can find documentation about the software architecture of this smart camera?
I would like to understand better how things work. Thanks! Toma

tpanzarella, 2015-12-05:
@tomatac I think @graugans could advise you on how to _offically_ back up the **sensor** product. Likely through a function available in IFM Vision Assistant (Windows GUI software). But I defer to him.

With respect to your efforts in building an OpenCV-based application utilizing `libo3d3xx` as the driver, I can likely answer any specific questions you may have. In short, the high-level concept of the library is to do all the configuration and manipulation utilizing the command line tools. You can pretty much do anything with them. Then once you have things set up the way you like, you then write custom code utilizing the library's frame grabbing API. A quick and dirty example of using the frame grabbing API is available [here](https://github.com/lovepark/libo3d3xx/blob/master/doc/using.md). A more thorough example can be gleaned from the [camera node](https://github.com/lovepark/o3d3xx-ros/blob/master/src/o3d3xx_node.cpp) of the ROS bindings. Additionally, through [my company](http://loveparkrobotics.com) we can offer you commercial support or build custom applications to your specification.

graugans, 2015-12-05:
@tomatac maybe the term **application** is a bit misleading  those are build into the device firmware. Whenever you create or delete one this is only a JSON file which selects the particular functionality and algorithm on the camera. 

You can create a full device backup of the camera with the help of the ifm Vision Assitant this will download all applications and the device configuration. You can re-apply this backup again with the help of the vision assistant. 

I am part of the embedded core team for the O3D3xx. Most of the support I do here is in my spare time. If you need more application related support do not hesitate to call our application support team. 

I am an expert on the camera interface not on how to solve applications.

graugans, 2015-12-07:
I've tested the command line from @tpanzarella on my **O3D302** (60° opening angle variant) and added activation of the first application. Otherwise the tests do fail again because the newly created application is not active

    echo '{"o3d3xx":{"Apps":[{}],"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config

You can check for an application with the command 

    o3d3xx-ls
      [1] id=829551123, name=New application, description=
    * [2] id=1788193191, name=New application, description=
      [3] id=44362294, name=New application, description=
The asterisk marks the active application
If you want to delete a application you can use the commandline tool ``o3d3xx-rm``

    o3d3xx-rm -i 2
This removes the application on index 2.
When I run the ``make check`` command I still have some issues but those look like minor issues.
    
    [ RUN      ] DenseImageTest.ActiveApplication
    unknown file: Failure
    C++ exception with description "XMLRPC finished but the called failed" thrown in SetUp().
    unknown file: Failure
    C++ exception with description "XMLRPC invalid application index" thrown in TearDown().
    [  FAILED  ] DenseImageTest.ActiveApplication (2009 ms)
    [ RUN      ] DenseImageTest.ImageSize
    unknown file: Failure
    C++ exception with description "XMLRPC finished but the called failed" thrown in SetUp().
    unknown file: Failure
    C++ exception with description "XMLRPC invalid application index" thrown in TearDown().
    [  FAILED  ] DenseImageTest.ImageSize (2008 ms)
    [----------] 2 tests from DenseImageTest (4017 ms total)
    ...
    [----------] Global test environment tear-down
    [==========] 57 tests from 9 test cases ran. (107863 ms total)
    [  PASSED  ] 55 tests.
    [  FAILED  ] 2 tests, listed below:
    [  FAILED  ] DenseImageTest.ActiveApplication
    [  FAILED  ] DenseImageTest.ImageSize

graugans, 2015-12-07:
@tpanzarella am I right the DenseImageTest is from the 100K test suit? So those may fail with any device other than the O3D303 because for the activation of the 100K feature an binary application is uploaded to the device.

tpanzarella, 2015-12-07:
@graugans That is correct. Particularly for the `0.2.0` release it uploads the binary application. This has changed on the dev branch since the 100K is directly exposed via XMLRPC.

tomatac, 2015-12-07:
Hi. First I would like to thank you for your support. I also do this evaluation also in my spare time. 
I think that this is a great technology and I hope that will pay off in the future.
I am trying to understand how does the camera works and based on your feedback I think that I am getting there.
The ifmVisionAssistant has 3 options "Export Settings", "Import Settings" and "Reset to Factory Defaults". I think that the "Export Settings" is equivalent to a backup and the "Reset to Factory Defaults" + "Import Settings" is equivalent to a restore. @graugans could you please confirm or redirect me to the right person that can confirm?
Now, for adding the application that allows me to capture the cloud point, I will summarize what I understood I should do:
1. Take the output of o3d3xx-dump from my camera and add in the applications section (the "sample application") from Tom's dump (that way I can retain the settings that are very specific to my camera like the MAC address)
2. write the new file to the camera: o3d3xx-config < camera.json
3. restart the camera
4. check if the application is registered ok: o3d3xx-ls
5. make it the active application: echo '{"o3d3xx":{"Apps":[{}],"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config
6. run the "make check"
Are these the right steps?

tpanzarella, 2015-12-07:
@tomatac A few notes... You don't have to worry about things like the MAC address, etc. `libo3d3xx` does not try to overwrite anything in the `HWInfo` or `SWVersion` sections of the JSON. They are treated as read-only. They are reported upon by `o3d3xx-dump` as they contain useful information for debugging and troubleshooting.

In terms of setting an application active, you can just use this:

```
$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"2"}}}' | o3d3xx-config
```
The above assumes you want to mark the application at index number `2` active. Adjust as necessary for your environment.

Basically, the `o3d3xx-config` application takes in a JSON file in the format output by `o3d3xx-dump`. However, it can also take just JSON snippets so long as they are *fully qualified*.

graugans, 2015-12-07:
@tomatac 
To create a new default application the recommended way is to use 

    echo '{"o3d3xx":{"Apps":[{}]}' | o3d3xx-config    # create a default camera application
    echo '{"o3d3xx":{"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config # activate application on index 1
Or use it in one single command:

    echo '{"o3d3xx":{"Apps":[{}],"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config

The reuse of the JSON dump is not needed. Because the camera uses its default settings.

```
The ifmVisionAssistant has 3 options "Export Settings", "Import Settings" and "Reset to Factory Defaults". I think that the "Export Settings" is equivalent to a backup and the "Reset to Factory Defaults" + "Import Settings" is equivalent to a restore. @graugans could you please confirm or redirect me to the right person that can confirm?
```
Yes you are right.
 
The MAC address of the device is burned into the CPU there is no way to change it through the camera software. So you are not able to alter it. ``HWInfo`` and ``SWVersion`` and stuff like this is as well read-only from on board camera software.

tpanzarella, 2015-12-07:
@tomatac @graugans One other thing to note... as @graugans [points out above](https://github.com/lovepark/libo3d3xx/issues/28#issuecomment-162433786), there still may be issues with the 100K imager tests for the specific sensor that you have. As `libo3d3xx` utilizes [gtest](https://github.com/google/googletest) as its unit testing framework, you can run the unit tests but disable the 100K tests since we know they are *expected to fail* in your particular instance. The easiest way to do this would be to invoke the `make check` part of the build like:

```
$ GTEST_FILTER=-DenseImageTest.* make check
```

tomatac, 2015-12-08:
Creating the default application worked fine:
echo '{"o3d3xx":{"Apps":[{}]}}' | ./o3d3xx-config

Activation of the application at index 1 did not work. I got this error:
“Failed to configure camera from JSON config '-': XMLRPC value out of range”

 I used a variation recommended earlier by @tpanzarella: 
1. I used ./o3d3xx-dump to dump my camera settings into a file (default_app.json)
2. I edited the file and changed "ActiveApplication": "1"
3. Applied the settings to the camera: o3d3xx-config < default_app.json
4. Checked if the settings were applied successfully: ./o3d3xx-dump 

“make check” passed all the tests, with the 2 exceptions mentioned above that are related to the high resolution option.
Also ./o3d3xx-viewer worked well.
Thank you for your help!

closed 2015-12-08 by tpanzarella