ifm/libo3d3xx #29: Question for the firmware team: Active Application at index "0"

tpanzarella, 2015-12-07:
@graugans I'm experiencing a behavior that I am wondering if it should be considered *normal* or if there is something I can do to prevent it from occurring. I believe this is a question for you and your team.

**NOTE:** All file redirections in the below shell examples are attached to this issue for review.

Let's assume I have the camera set at factory defaults:

```
$ o3d3xx-reset
$ o3d3xx-dump > /tmp/camera_baseline.json.txt
$ o3d3xx-ls
* [1] id=559769693, name=Sample Application, description=
```
[camera_baseline.json.txt](https://github.com/lovepark/libo3d3xx/files/54482/camera_baseline.json.txt)

So, we know by default the imager will be set to single frequency dual exposure (aka `under5m_moderate`). Now, I want to change the imager for my application to, for example, double frequency single exposure (aka `upto30m_low`):

```
$ echo '{"o3d3xx":{"Apps":[{"Index":"1","Imager":{"Type":"upto30m_low"}}]}}' | o3d3xx-config
$ o3d3xx-dump > /tmp/camera2.json.txt
$ o3d3xx-ls
* [1] id=559769693, name=Sample Application, description=
```
[camera2.json.txt](https://github.com/lovepark/libo3d3xx/files/54496/camera2.json.txt)

So far so good. Now, I want to increase the frame rate. Let's set it to 10 Hz:

```
$ echo '{"o3d3xx":{"Apps":[{"Index":"1","Imager":{"FrameRate":"10"}}]}}' | o3d3xx-config
$ o3d3xx-dump > /tmp/camera_10Hz.json.txt
$ o3d3xx-ls
* [1] id=559769693, name=Sample Application, description=
```
[camera_10Hz.json.txt](https://github.com/lovepark/libo3d3xx/files/54501/camera_10Hz.json.txt)

We are still good. Now, suppose I want to set the frame rate to 15Hz:

```
$ echo '{"o3d3xx":{"Apps":[{"Index":"1","Imager":{"FrameRate":"15"}}]}}' | o3d3xx-config
$ o3d3xx-dump > /tmp/camera_15Hz.json.txt
$ o3d3xx-ls
  [1] id=559769693, name=Sample Application, description=
```
[camera_15Hz.json.txt](https://github.com/lovepark/libo3d3xx/files/54504/camera_15Hz.json.txt)

**BOOM!** Here is the problem. We can see that we still have the application on the camera but it is no longer marked as active. In fact, via inspection, we can see that the `ActiveApplication` is actually `0`:

```
$ o3d3xx-dump | grep -i 'ActiveApplication'
            "ActiveApplication": "0",
```

So, if we try to set it back to `1`, we can see that we get no complaints, yet, the setting does not stick.

```
$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config
$ o3d3xx-ls
  [1] id=559769693, name=Sample Application, description=
$ o3d3xx-dump | grep -i 'ActiveApplication'
            "ActiveApplication": "0",
```

Now, if we step the frame rate back down to 10Hz and reactivate the application, everything is OK again:

```
$ echo '{"o3d3xx":{"Apps":[{"Index":"1","Imager":{"FrameRate":"10"}}],"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config
$ o3d3xx-ls
* [1] id=559769693, name=Sample Application, description=
```

Is this the intended behavior? Are there known `FrameRate` settings that will cause this for a given imager type? 

As can be gleaned from the attached JSON dumps, I'm running the development version of the `libo3d3xx` code (`0.3.0`) and the `1.2.533` firmware on the camera.

graugans, 2015-12-07:
@tpanzarella the whole team is on a **C++11** training until Thursday. I'll cross check with a *1.3.x* pre release and the *0.3.0* dev branch. When I find some time between the trainings.

tpanzarella, 2015-12-07:
Thanks Christian!

@graugans  As an FYI... you may want to flash the test camera with the `1.2.533` firmware as I have not had time yet to work with the `1.3.x` stuff. For example, I think I recall you mentioning that the `Output100K` parameter has changed and likely a few other things too. So, other additional/unexpected breakages may also occur between `libo3d3xx` and that version of the camera-side code.

BTW, **C++11** training sounds fun. Enjoy!

cfreundl, 2015-12-12:
It seems that the application fails to validate due to a too high frame rate. There is a constraint with respect to the maximum allowed LED duty cycle of 50 %  which limits the frame rate in free-run trigger mode depending on the chosen exposure mode and exposure time.

You can manually perform application validation on the application object during editing (URL `http://<sensor-ip>/api/rpc/v1/com.ifm.efector/session_<session-id>/edit/application/`, RPC `validate` with no input parameters). A valid application would return an empty error list.

In case that one of the returned issues is in fact the LED duty cycle violation (error code 101055) the error message shows the maximum allowed frame rate for the current exposure settings.

tpanzarella, 2015-12-12:
@cfreundl This makes sense. Thanks for the explanation. However, for the steps that I have outlined above, an XMLRPC error was not returned. It just *silently* sets the `FrameRate` to the level I asked for (15Hz in this case) but moves the `ActiveApplication` to `0`. I think it would be much better to disallow the framerate to be set (leave it unmutated) and to fail with an error giving an indication as to the problem as you allude to above. I was not aware of the `validate` function and I will look into that to see if that would have returned the error to me. I do not have easy access to a camera right now, so, I cannot perform any tests, but I can verify next week.

cfreundl, 2015-12-13:
We deliberately chose that approach because we have more constraints like that where the individual parameters are within their hard limits but the combination is invalid (e.g. there are also restrictions on the allowed filter settings depending on the trigger mode). If we immediately forbid the setting of the first parameter, it would be difficult to perform a sequence of parameter changes which results in a valid application but goes over some invalid intermediate steps. Also, an application which is valid on one device may be invalid on an other device, e.g. if the application uses IOs but the second device is globally configured to application switching while the first device is not - but that is more an issue of sensor applications rather than camera ones.

tpanzarella, 2015-12-13:
So, is it fair to consider an `ActiveApplication` equal to `0` as a sentinel that denotes *something* in the configuration is invalid? But from the camera's perspective, it makes this assumption that you wanted to "save your work" because you are in the process of crafting a valid configuration. Something like that??

graugans, 2015-12-13:
@tpanzarella yes that's the idea. As Christoph already stated. In some situations it is useful to have an *invalid state* for applications. Another usecase we had in mind was upgrading from one firmware version to the other. Which may transform an application into *invalid state*

tpanzarella, 2015-12-13:
Understood. Thanks guys. I think having this issue available on github to point to will be useful.

closed 2015-12-13 by tpanzarella