ifm/libo3d3xx #31: Error starting o3d3xx-viewer

Obbart, 2015-12-09:
As soon as I start the program I get this error message and the application crashes.
 error: (-218) Library was built without OpenGL support in function cvNamedWindow

I built opencv 3.0 using this command that seems to include OpenGL support
cmake -D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_INSTALL_PREFIX=/usr/local -D BUILD_NEW_PYTHON_SUPPORT=ON -D WITH_V4L=ON -D WITH_OPENGL=ON ..

Probably I did something wrong, any ideas?
Thanks,
          Emanuele

tpanzarella, 2015-12-09:
We currently only support OpenCV 2. More info when I am at a computer.
(Sorry).

tpanzarella, 2015-12-10:
Guidance from the following closed issues may also be useful:

https://github.com/lovepark/libo3d3xx/issues/18
https://github.com/lovepark/o3d3xx-ros/issues/8
https://github.com/lovepark/libo3d3xx/issues/19

Bottom-line: Just install the binaries for OpenCV 2.4 via `apt-get` from the official Ubuntu repos and you should be fine.

Obbart, 2015-12-10:
ok, now it works just fine, thanks

closed 2015-12-10 by tpanzarella