ifm/libo3d3xx #33: Cannot get viewer working for O3D303

bigcmos, 2015-12-11:
I have a few working o3d303 sensors and a new one that I just received which isn't displaying anything in the viewer, just a blank screen, see below.  The config dump below looks similar to my other sensors.

[image]

```
$ ./o3d3xx-ls 
* [1] id=1186278907, name=Sample Application, description=
$ ./o3d3xx-dump > /tmp/o3d303-dev.json
$ cat /tmp/o3d303-dev.json
{
    "o3d3xx":
    {
        "libo3d3xx": "200",
        "Date": "Fri Dec 11 14:10:25 2015",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02065512_008022954",
            "Illumination": "#!02_I300_001_02085611_008001175",
            "MACAddress": "00:02:01:40:25:5D",
            "Frontend": "#!02_F341_002_15_00238_802259343",
            "Diagnose": "#!02_D322_C30_02061433_008023093",
            "Connector": "#!02_A300_001_02064651_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.1.288",
            "Linux": "Linux version 3.11.0-00001-g4e630e2 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Mon Jun 8 10:25:20 CEST 2015",
            "Diagnostic_Controller": "0.3.9",
            "ELDK": "GOLDENEYE_ELDK-2015-07-01_16-26-37-cc5673110f1bff06b66149c8320f6c1823d87891",
            "Calibration_Version": "0.7.1",
            "Calibration_Device": "00:02:01:40:25:5d",
            "Main_Application": "0.4.34",
            "Algorithm_Version": "0.6.19"
        },
        "Device":
        {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "1",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "5",
            "ServiceReportFailedBuffer": "5",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
            "DeviceType": "",
            "ArticleNumber": "",
            "ArticleStatus": "",
            "UpTime": "6.95301075392642e-310",
            "ImageTimestampReference": "31590384",
            "TemperatureFront1": "1.560784204908798e-316",
            "TemperatureFront2": "6.952977167208689e-310",
            "TemperatureIMX6": "1.560796062484299e-316",
            "TemperatureIllu": "1.560807920059799e-316"
        },
        "Net":
        {
            "MACAddress": "",
            "NetworkSpeed": "0",
            "StaticIPv4Address": "192.168.0.69",
            "StaticIPv4SubNetMask": "255.255.255.0",
            "StaticIPv4Gateway": "192.168.0.201",
            "UseDHCP": "false"
        },
        "Apps":
        [
            {
                "Name": "Sample Application",
                "Description": "",
                "TriggerMode": "1",
                "PcicTcpResultOutputEnabled": "true",
                "PcicTcpResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"ascii\" }, \"elements\": [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"blob\", \"id\": \"normalized_amplitude_image\" }, { \"type\": \"blob\", \"id\": \"distance_image\" }, { \"type\": \"blob\", \"id\": \"x_image\" }, { \"type\": \"blob\", \"id\": \"y_image\" }, { \"type\": \"blob\", \"id\": \"z_image\" }, { \"type\": \"blob\", \"id\": \"confidence_image\" }, { \"type\": \"blob\", \"id\": \"diagnostic_data\" }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
                "PcicEipResultSchema": "{ \"layouter\": \"flexible\", \"format\": { \"dataencoding\": \"binary\", \"order\": \"big\" }, \"elements\" : [ { \"type\": \"string\", \"value\": \"star\", \"id\": \"start_string\" }, { \"type\": \"records\", \"id\": \"models\", \"elements\": [ { \"type\": \"int16\", \"id\": \"boxFound\" }, { \"type\": \"int16\", \"id\": \"width\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"height\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"length\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"xMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"zMidTop\", \"format\": { \"scale\": 1000 } }, { \"type\": \"int16\", \"id\": \"yawAngle\" }, { \"type\": \"int16\", \"id\": \"qualityLength\" }, { \"type\": \"int16\", \"id\": \"qualityWidth\" }, { \"type\": \"int16\", \"id\": \"qualityHeight\" } ] }, { \"type\": \"string\", \"value\": \"stop\", \"id\": \"end_string\" } ] }",
                "LogicGraph": "{\"IOMap\": {\"OUT1\": \"RFT\",\"OUT2\": \"AQUFIN\"},\"blocks\": {\"B00001\": {\"pos\": {\"x\": 200,\"y\": 200},\"properties\": {},\"type\": \"PIN_EVENT_IMAGE_ACQUISITION_FINISHED\"},\"B00002\": {\"pos\": {\"x\": 200,\"y\": 75},\"properties\": {},\"type\": \"PIN_EVENT_READY_FOR_TRIGGER\"},\"B00003\": {\"pos\": {\"x\": 600,\"y\": 75},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT1\"},\"B00005\": {\"pos\": {\"x\": 600,\"y\": 200},\"properties\": {\"pulse_duration\": 0},\"type\": \"DIGITAL_OUT2\"}},\"connectors\": {\"C00000\": {\"dst\": \"B00003\",\"dstEP\": 0,\"src\": \"B00002\",\"srcEP\": 0},\"C00001\": {\"dst\": \"B00005\",\"dstEP\": 0,\"src\": \"B00001\",\"srcEP\": 0}}}",
                "Type": "Camera",
                "Index": "1",
                "Id": "1186278907",
                "Imager":
                {
                    "Channel": "0",
                    "ClippingBottom": "131",
                    "ClippingLeft": "0",
                    "ClippingRight": "175",
                    "ClippingTop": "0",
                    "ContinuousAutoExposure": "false",
                    "EnableAmplitudeCorrection": "true",
                    "EnableFilterAmplitudeImage": "true",
                    "EnableFilterDistanceImage": "true",
                    "EnableRectificationAmplitudeImage": "false",
                    "EnableRectificationDistanceImage": "false",
                    "ExposureTime": "5000",
                    "ExposureTimeList": "125;5000",
                    "ExposureTimeRatio": "40",
                    "FrameRate": "5",
                    "MinimumAmplitude": "42",
                    "ReduceMotionArtifacts": "false",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_moderate",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}
```

bigcmos, 2015-12-11:
helpful to note I am building from v0.2.0

* 437e48f 2015-11-16 (HEAD, tag: v0.2.0, origin/dev-0.2.0) | fixed another typo

tpanzarella, 2015-12-11:
Are the 2D images showing? Or are both the point cloud and 2D images
blacked out?

bigcmos, 2015-12-11:
Both are blacked out.

tpanzarella, 2015-12-11:
Did you spot check against IFM Vision Assistant?

bigcmos, 2015-12-11:
Yes.  IFM Vision Assistant seems to output good data.

tpanzarella, 2015-12-11:
Can you run o3d3xx-hz and let me know what it reports?

bigcmos, 2015-12-11:
$ ./o3d3xx-hz
FrameGrabber running at: 4.89804 Hz
10 frames captured; averaged over 1 runs

tpanzarella, 2015-12-12:
Hmm. So, we are getting data. Is there anything unique about your
rendering/graphics environment? Because, from everything you told me, we
are getting data fine from the camera but the rendering is not working. The
viewer application is really just a very thin wrapper around both OpenCV
and PCL's native viewer applications.

Are you running in a ROS environment by chance? I'm curious to see if rviz
can render the data.

bigcmos, 2015-12-12:
No same system I've been using.  rviz works fine.  When I get back to my desk in about 30 minutes I'll test out one of the other sensors to make sure they can still visualize, but they were working this morning.

bigcmos, 2015-12-12:
I was able to verify that my other sensors are working and can visualize on that computer.  I can dig into the source more on Monday, but let me know if you have any other tips for debugging.

bigcmos, 2015-12-12:
Sorry, I haven't tried to visualize the sensor in rviz.  I can give that a shot on Monday.

tpanzarella, 2015-12-12:
This is not something I have seen before. However, here is a few things I'm thinking about as I read this issue:

1. If you can, I'd like to know if you can visualize the data in rviz. From what I read so far, this sounds like a rendering issue. But, I'm not entirely convinced just yet.

2. Getting the specific versions of PCL and OpenCV you have on your system may also be helpful.

3. If you decide to instrument the code, perhaps simply writing the point cloud and other images to disk would be good. That way we can manually inspect the pixel data. You can do this by instrumenting the `o3d3xx-viewer` code directly, or if you work within a ROS framework, you can use the `o3d3xx_file_writer_node`. Instructions to do that can be gleaned from the first part of [this article](https://github.com/lovepark/o3d3xx-ros/blob/master/doc/matlab_tutorial.md). Directly instrumenting `o3d3xx-viewer` would probably be better as this is would keep the debugging very close to where we are seeing the problem. However, using the established ROS tools would also give us some useful insight and be less effort on your part.

4. Now, this one, I'm just thinking out loud... For the point cloud the pixels are colored with the amplitude data and, of course, the amplitude images are a specifically the amplitude data. If these levels are low, the images may not appear. However, the colormaps that are being used would render them visual. This then leads me to the confidence data. Is there any chance all pixels are being flagged as bad? In that case the cloud would be full of `NaN` and the the 2D images would be `0`. But, even in this case, the binary rendering of the confidence image would show up as all *yellow*.

5. Logs are written by default to `/tmp`. If you do a debug build (just edit the top-level `CMakeLists.txt`) the logging will get really verbose. However, most of the debugging info will be coming from the `FrameGrabber` as it parses through the bytes being returned from the camera. By running `o3d3xx-hz` we sort of already concluded that we are getting the data from the camera just fine. But, it may give us a clue.

6. If you remain stuck, and you are willing, you can ship your camera to me and I will work with it directly to see if I can get to the bottom of this. It has me puzzled at the moment. Please keep me informed.

bigcmos, 2015-12-14:
Started digging into this today... and the sensor is reporting valid depth images and points.  I haven't changed anything since Friday (came back to the same terminal I had open yesterday).  I'm able to view the data on rviz and o3d303-viewer... So let's close this out until I can repro in the future.

bigcmos, 2015-12-16:
Had the issue happen again today.  Same issue with the viewer, but o3d3xx-hz was also timing out.  Was able to resolve the issue by rebooting the sensor.  Will keep an eye out, but a bit worried about what is causing the sensor to get into this state.  Would love some guidance on how to debug if this happens again.

Symptoms
```
$ /opt/libo3d3xx/bin/o3d3xx-viewer 
^C
$ /opt/libo3d3xx/bin/o3d3xx-ls
* [1] id=456425781, name=Sample Application, description=
$ /opt/libo3d3xx/bin/o3d3xx-hz
Timeout waiting for camera!
```

Work Around
```
$ /opt/libo3d3xx/bin/o3d3xx-reboot
```

bigcmos, 2015-12-16:
Oh, and it looks like we had a power failure over night which would explain why the sensor might have rebooted the first time I was having trouble.

graugans, 2015-12-17:
@bigcmos In case of an under/over voltage the camera is going into an error state. You can check the LEDs on the sensor when they are blinking this means an error condition occurred. You can check the [Operation instructions on page 16](http://www.ifm.com/mounting/706398UK.pdf) for the blink codes.

[image]

For debugging purposes you can request the camera trace logs by XML-RPC the method is called ``getTraceLogs`` you have to provide one integer parameter ``amount of log lines`` when providing ``-1`` you'll get all logs in the buffer 

You can request the logs with curl

```
curl --data "method=getTraceLogs&params=[-1]" http://192.168.0.69/api/rpc/v1/com.ifm.efector/
````
The result looks like this

```
$ curl --data "method=getTraceLogs&params=[-1]" http://192.168.0.69/api/rpc/v1/com.ifm.efector/
<div style="border: 1px solid; padding: 2px"><b>Response</b><br>{
    "error": null,
    "id": null,
    "result": [
        "T: 8.2632, P: 715,  C: WARN,  S: iom,  M: Program starting ...",
        "T: 8.2631, P: 711,  C: WARN,  S: qt_dbus_web_bridge,  M: Starting ...",
        "T: 8.3676, P: 715,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_VFE_ON_ALARM",
        "T: 8.3702, P: 715,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_VFE_ON_ALARM",
        "T: 9.0392, P: 737,  C: WARN,  S: gecco,  M: Starting ...",
        "T: 9.1180, P: 732,  C: WARN,  S: imager,  M: Starting ...",
        "T: 15.0267, P: 737,  C: WARN,  S: gecco,  M: Imager is available again",
        "T: 15.0291, P: 732,  C: WARN,  S: imager,  M: LRAE is available again",
        "T: 15.0338, P: 732,  C: WARN,  S: imager,  M: void ifm::ImageAcquisition::requestConfigurationChange()",
        "T: 15.0710, P: 732,  C: WARN,  S: imager,  M: void ifm::ImageAcquisition::requestConfigurationChange()",
        "T: 15.0852, P: 732,  C: WARN,  S: imager,  M: void ifm::ImageAcquisition::requestConfigurationChange()",
        "T: 15.4165, P: 737,  C: WARN,  S: gecco,  M: Finished imager configuration"
    ]
}
</div><br><form method='post'>id: <input name='id' type='text' value='1'><br><i>(JSON-RPC id, will directly be copied to answer-msg)</i><br>
method: <input name='method' type='text'><br><i>(e.g. org.freedesktop.DBus.Introspectable.Introspect, org.freedesktop.DBus.ListNames)</i><br>params: <input name='params' type='text' value='[]'><br><i>(The bridge now support minimal type-convertion ... //FIXME: atm there are still some parts missing for non-primitiv data ...)</i><br>
<input type='submit'><br></form>
<div>methods:<ul><li>com.ifm.efector.getParameter(s parameterName) -> (is) status, s parameterValue</li><li>com.ifm.efector.getAllParameters() -> (is) status, a{sv} parameterMap</li><li>com.ifm.efector.getSWVersion() -> (is) status, a{sv} versionMap</li><li>com.ifm.efector.getHWInfo() -> (is) status, a{sv} versionMap</li><li>com.ifm.efector.getClientCompatibilityList() -> (is) status, a{s} compatibilityList</li><li>com.ifm.efector.getApplicationList() -> (is) status, aa{sv} applicationList</li><li>com.ifm.efector.requestSession(s password, s requestedSessionID) -> (is) status, s sessionID</li><li>com.ifm.efector.reboot(i rebootType) -> (is) status</li><li>com.ifm.efector.getActiveDataConnections() -> (is) status, aa{sv} connectionList</li><li>com.ifm.efector.systemCommand(s command, s input) -> (is) status, s output</li><li>com.ifm.efector.switchApplication(i applicationIndex) -> (is) status</li><li>com.ifm.efector.ioSwitchApplication(i applicationIndex) -> (is) status</li><li>com.ifm.efector.ioTeachFirstModel() -> (is) status</li><li>com.ifm.efector.getTraceLogs(i nLogs) -> (is) status, as logs</li><li>com.ifm.efector.doPing() -> (is) status, s runlevel</li><li>com.ifm.efector.startServiceReportReadOut(i timeout) -> (is) status</li><li>com.ifm.efector.stopServiceReportReadOut() -> (is) status</li><li>com.ifm.efector.getServiceReportPassedSlots() -> (is) status, a{s} passedSlots</li><li>com.ifm.efector.getServiceReportFailedSlots() -> (is) status, a{s} failedSlots</li><li>com.ifm.efector.getServiceReportData(u id, s slotName) -> (is) status, ay resultData</li><li>com.ifm.efector.getApplicationStatisticData(u id) -> (is) status, s resultData</li><li>org.freedesktop.DBus.Properties.Get(s interface_name, s property_name) -> v value</li><li>org.freedesktop.DBus.Properties.Set(s interface_name, s property_name, v value) -> </li><li>org.freedesktop.DBus.Properties.GetAll(s interface_name) -> a{sv} values</li><li>org.freedesktop.DBus.Introspectable.Introspect() -> s xml_data</li><li>org.freedesktop.DBus.Peer.Ping() -> </li><li>org.freedesktop.DBus.Peer.GetMachineId() -> s machine_uuid</li></ul></div>
```

bigcmos, 2016-02-10:
Thanks for the great info.  Haven't had an issue since then.  Closing this out.

closed 2016-02-10 by bigcmos