ifm/libo3d3xx #34: Support for Firmware Update by a command line utility

graugans, 2015-12-12:
This is a discussion about an enhancement. Is there any interest of an command line utility which provides support for firmware updates? At the moment the only way to update the O3D3xx devices is by running the official **ifm Vision Assistance** Due its platform support limitations a Windows host is needed for this.

If there is any interest I can implement a command line utility to support this.

Any suggestions on naming and command line options are welcome.

tpanzarella, 2015-12-12:
In the `dev-0.3.0` branch, I have added the `-r` flag to `o3d3xx-reboot` for rebooting the camera into recovery mode. This then allows for updating the firmware via the camera's internal HTTP server. This is my current approach.

I think your suggestion would be nice-to-have. Particularly for updating a fleet of cameras all at once, having a command-line tool would be nice for wrapping in a shell script.

In terms of naming, I'd suggest `o3d3xx-flash`.

tpanzarella, 2016-02-20:
Christian, if you are working on this, I'd suggest holding off until `0.4.0` is released. This is a major reorganization of library in support of the modularization that Ville had requested. I do have a `dev-0.4.0` branch pushed up to github now, so, you can work off of that, but it is probably quite volatile at this point. I expect to have it all wrapped up sometime next week.

graugans, 2016-03-30:
I've pushed part of the internal ``ruby`` classes. This contains a basic [firmware update utility](https://github.com/graugans/ruby-o3d3xx/blob/master/bin/o3d3xx-fwupdate.rb).

bigcmos, 2016-04-06:
@graugans Is this ready for use?  I'm currently deploying to many remote locations and hoping to use this tool today if possible.

graugans, 2016-04-06:
I used it several times to update my devices. But never tested on a slow or buggy network connection.

closed 2020-07-27 by theseankelly