ifm/libo3d3xx #35: Compability with O3D3xx "Sensor"

graugans, 2015-12-14:
During the discussion in #29 and #28 we discovered that the so called **sensor** is not working out of the box with this lib. In #28 we created a command line to create a camera application and activate this.

    echo '{"o3d3xx":{"Apps":[{}],"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config

After this the built in checks are mostly working. Except the 100K Tests, for more details check #28.

During issue #29 I took a deeper look into the difference between the default camera application and the default sensor application and discovered some other differences.

```
                 "Imager":
                 {
                     "Channel": "0",
@@ -91,11 +91,11 @@
                     "EnableFastFrequency": "false",
                     "EnableFilterAmplitudeImage": "true",
                     "EnableFilterDistanceImage": "true",
-                    "EnableRectificationAmplitudeImage": "false",
-                    "EnableRectificationDistanceImage": "false",
-                    "ExposureTime": "5000",
-                    "ExposureTimeList": "5000",
-                    "FrameRate": "10",
+                    "EnableRectificationAmplitudeImage": "true",
+                    "EnableRectificationDistanceImage": "true",
+                    "ExposureTime": "1000",
+                    "ExposureTimeList": "1000",
+                    "FrameRate": "5",
                     "MinimumAmplitude": "42",
                     "Output100K": "false",
                     "ReduceMotionArtifacts": "false",
@@ -105,7 +105,7 @@
                     "ThreeFreqMax2FLineDistPercentage": "80",
                     "ThreeFreqMax3FLineDistPercentage": "80",
                     "TwoFreqMaxLineDistPercentage": "80",
-                    "Type": "upto30m_low",
+                    "Type": "under5m_low",
                     "SpatialFilter":
```

For example the camera application sets ``EnableRectificationAmplitudeImage`` and ``EnableRectificationDistanceImage``  parameters to **disabled** by default. Beside this the ``ExposureTime`` is default at  **5000** and Exposure mode is set to ``upto30m_low``.

So the correct configuration should look like this to use a **O3D3xx Sensor** as a camera
```
o3d3xx-reset
echo '{"o3d3xx":{"Apps":[{"Index":"1","Imager":{ "EnableRectificationAmplitudeImage": "false","EnableRectificationDistanceImage": "false","Type":"upto30m_low","ExposureTime":"5000","ExposureTimeList":"5000","FrameRate": "10"}}],"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config
```

The question is where is the right place to put such documentation on?

tpanzarella, 2015-12-14:
Maybe we codify it. For example, if we add a switch to o3d3xx-reset that
does the standard reset plus the other necessary modifications. If we did
this, I think it should be an explicit switch (as opposed to some directive
in the JSON) as this would ensure the user is explicitly enabling this
behavior as opposed to it occurring implicitly.

graugans, 2015-12-14:
Thats a  good idea. Any suggestions for the switch? 
- ``--create-camera-app`` 
- ``--create-and-activate-camera-app``
- ``--sensor``

tpanzarella, 2015-12-14:
I like --sensor

closed 2020-07-27 by theseankelly