ifm/libo3d3xx #36: issue with PCL from Ubuntu repos

wegunterjr-Dematic, 2015-12-22:
I know this is know, but I plan to use the PCL library for the time being.  I plan to just build PCL 1.7.2 from source in the morning, but wanted to verify if that would be okay?  Does that mess with other code? 
I had another ticket that dealt with that when I first started a month ago, but forgot the details behind it.

wegunterjr-Dematic, 2015-12-23:
I pulled PCL 1.8 and built from source using the c++ 11 compile flags for cmake. I find that some libraries are missing.   I am going through the list now and adding the missing ones. 
Also, i remembered that the issue I was reference above wasn't related to PCL, it was using OpenCV 3.0 and not OpenCV 2.4.    I imagine there isn't an issue using PCL 1.8?

wegunterjr-Dematic, 2016-01-04:
still getting an error with the math library when trying to use the PCL.  I will try to build PCL again for C++ 11 from source.

graugans, 2016-01-04:
What version of Ubuntu are using? I had no issues to compile on 14.04.

wegunterjr, 2016-01-04:
I am using Ubuntu 14.04 LTS.   Did you compile from source and use:
``` cmake -std=c++11 ```

graugans, 2016-01-04:
I am not a PCL nor a CMake Pro but I guess to override the CXXFlags you have to call CMake like this

```
 CXXFLAGS=-std=c++11  cmake .. && make
```

wegunterjr, 2016-01-04:
copy that. I am trying it now with the command line flag
``` cmake -std=c++11 ../pcl ```    (I do an out of source build, hence the call to the pcl source folder)

wegunterjr, 2016-01-04:
@graugans compile worked, but need to test and see if I can use it in my project.

wegunterjr-Dematic, 2016-01-13:
alright, the compiling in C++11 definitely worked.

closed 2016-01-13 by wegunterjr-Dematic