ifm/libo3d3xx #38: setting 2 ifm O3D303 sensor

metallo25, 2016-01-19:
I am trying to set two O3D03 sensor , but there is interference between the two. Even thought I set them at different frequency using the IFM software, it still affect each other data. Is there an option in ros driver where I can change the frequency, so that both of them can coexist?.

tpanzarella, 2016-01-19:
You can use `o3d3xx-dump`, modify the `Channel` in the JSON config, then reload that configuration via `o3d3xx-config`. Just put each camera on its own channel. Be mindful of how many frequencies each camera is using based on your imager configuration.

graugans, 2016-01-19:
@metallo25 depending on your application you may can cascade the two cameras. The first camera is in free-running trigger mode the second one is configured as hardware trigger. Now you have to connect the **aquisition finished** PIN to the trigger input PIN of the second camera.

zfxw0206, 2016-08-01:
Now I have 4 cameras, 2 on the top of my machine, 2 on the bottom. My question is how many channels do we have? because now I set the channel of top left camera and bottom right camera be 0 while the channel of other 2 cameras be 1, they still interference with each other.

What can I do with 4 cameras?

Thank you!

graugans, 2016-08-02:
@zfxw0206 have you already checked #46 There are some details how to synchronize multiple cameras. When you are talking about trigger modes what trigger mode are you using?
- software
- hardware (I/O)

The recommendation for multi camera set-ups is to use hardware trigger and the same illumination time for the cameras. I'll add @garrettplace so he can keep track on the multi camera stuff.

zfxw0206, 2016-08-02:
We are using software trigger mode. I have another question, how many channels do we have when we configure the camera? Is it possible to set the 4 cameras to 4 different channels?

graugans, 2016-08-02:
Sorry, there are only 3 Channels (Frequencies) available at the moment. There are some ideas about reduction of multi camera disturbances but I am not aware about the time schedules and success rate. To lower the disturbance hardware trigger is recommended because of lower latencies. This is related to the underlying PMD ( Photonic Mixer Device) technology.

closed 2020-07-27 by theseankelly