ifm/libo3d3xx #39: send software trigger?

Postea-CS, 2016-02-02:
Hi,
Looking at the api. I can't find any information about sending software triggers when the camera is in process trigger mode. Wonder if i am missing something here. Thanks for any information.

tpanzarella, 2016-02-02:
I am not sure that I understand what you are looking to do. The library's framegrabber continuously streams data from the camera in a separate thread from your `main`. It double buffers the data and allows you to fetch the latest verified frame in a zero-copy manner. You can reach in and get the data at whichever rate your main loop is running, you will always get the latest frame. A good example of writing a loop that grabs the data and does something with it would be the main ROS publisher node from `o3d3xx-ros`. That is here: 

https://github.com/lovepark/o3d3xx-ros/blob/master/src/o3d3xx_node.cpp#L117

I hope that helps.

Postea-CS, 2016-02-02:
The method you describe would work if the camera is in a free run mode. In the process trigger mode. We will need to send a software trigger to the camera before the image is available. I can't find the function to send a software trigger in the library.

tpanzarella, 2016-02-02:
OK. Yeah, we would have to build in that support. The only use case we handle today is where you want to stream data in from the camera and then asynchronously grab the latest data. I'll mark this as an enhancement that we can build into the library.

Postea-CS, 2016-02-02:
Ok thanks. In that case. Any chance you have a sample of command line to send process interface commands? I know the xmlrpc commands to configure the camera. Is it possible to send process interface thru something command line? thanks

tpanzarella, 2016-02-02:
@graugans may have something you can use.

Postea-CS, 2016-02-02:
i am looking for some information about the process interface and how to communicate and send commands thru the process interface. Any references ( code or doc) as to how to do this? hence what port, sample example would be great. thanks!

tpanzarella, 2016-02-02:
The Operations Manual in the `doc` directory (Ch. 8) explains the PCIC interface. 

You can look at the ctor of the `frame_grabber.cpp` file for an example of crafting a PCIC command in C++ to be sent to the camera. In this case it is the `c` command (to set the result schema). Then in `Run()` you will see how to send and receive the PCIC request/reply.

In terms of the TCP port, you can use `o3d3xx-dump` to see what yours is set to. By default it is 50010. 

Also note, that it is very doable (albeit tedious) to telnet to the TCP port and manually send the commands -- I have done that for purposes of debugging in the past.

tpanzarella, 2016-02-02:
In follow up to the above, once you have the protocol spec in hand per the operations manual, it is probably much easier to write PCIC scripts in something like Python as opposed to C++, particularly if you are still in discovery mode.

graugans, 2016-02-02:
The process interface is multi-connection capable. So you can use the libo3d3xx to grab images and other connection to trigger the images. I suggest to use the asynchronous trigger (t) command and for simplification you have to send a p0 to stop sending any asynchronous data over the trigger connection.

I'll provide some ruby sample code demonstrating how the communication works.

graugans, 2016-02-02:
The interesting parts are the functions ``recv_v3()`` and  ``send_v3()`` I do not recommend to use any protocol other than **v3** Because it is the most machine readable one.

``` ruby
ticket = "%04d" % [1000+rand(8999)]
length = "%09d" % [data.length+6]
sendbuf = "#{ticket}L#{length}\r\n#{ticket}#{data}\r\n"
```

The ticket is needed to distinguish between asynchronous replies from the device ``<1000`` and replies to commands you have send to the device. The length information contains the length of the data to transfer and the ticket length (4 bytes) + 2 bytes (\r\n) So the sendbuf looks like this for the p<0> (disable asynchronous output) command:

``
6666L000000008t\r\n6666p0\r\n
``

The trigger command looks like this:

``
6666L000000007t\r\n6666t\r\n
``

This is completely untested due to lack of camera in reach :)

Postea-CS, 2016-02-02:
Ok thanks for your help. the > 1000 ticket number was tripping me up.  

I have tested the trigger command
1010L000000007\r\n1010t\r\n
works when sending it over telnet port 50010. 

many thanks!

closed 2016-02-02 by Postea-CS