ifm/libo3d3xx #41: Linux is showing less FPS than windows for the same config

bigcmos, 2016-02-10:
I have been trying to squeeze as much performance out of the sensor and increase the output rate as I can.  I configured the sensor on Windows using the IFM Vision Assistant with the selected the application I named: lessthan5_low26.  IFM Vision Assistant on Windows claims I am getting a solid 26 Hz from the camera.  For the same configuration on Linux, o3d3xx-hz returns about 18 Hz.

Am I doing something wrong on Linux or is this an expected difference between the Windows app and Linux app?  See configuration below.

```
$ /opt/libo3d3xx/bin/o3d3xx-ls
  [1] id=1186278907, name=Sample Application, description=
  [2] id=1222655088, name=100K Pixel Sample Application, description=
  [3] id=631185153, name=upto30_moderate, description=High Fidelity @ 5Hz
  [4] id=297319475, name=lessthan5_low20, description=Low Fidelity @ 20Hz
* [5] id=869942976, name=lessthan5_low26, description=Low Fidelity @ 26Hz
$ /opt/libo3d3xx/bin/o3d3xx-hz -f 50 -r 5
FrameGrabber running at: 17.6827 Hz
50 frames captured; averaged over 5 runs
```

**Configuration**
```
{
    "o3d3xx":
    {
        "libo3d3xx": "200",
        "Date": "Tue Feb  2 15:15:51 2016",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02001855_008022954",
            "Illumination": "#!02_I300_001_02025830_008001175",
            "MACAddress": "00:02:01:40:1F:59",
            "Frontend": "#!02_F341_002_15_00201_008022593",
            "Diagnose": "#!02_D322_C25_01977637_008022488",
            "Connector": "#!02_A300_001_01867216_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.1.288",
            "Linux": "Linux version 3.11.0-00001-g4e630e2 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Mon Jun 8 10:25:20 CEST 2015",
            "Diagnostic_Controller": "0.3.9",
            "ELDK": "GOLDENEYE_ELDK-2015-07-01_16-26-37-cc5673110f1bff06b66149c8320f6c1823d87891",
            "Calibration_Version": "0.6.6",
            "Calibration_Device": "00:02:01:40:1f:59",
            "Main_Application": "0.4.34",
            "Algorithm_Version": "0.6.19"
        },
        "Device":
        {
            "Name": "o3d3xx-dev",
            "Description": "",
            "ActiveApplication": "5",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "5",
            "ServiceReportFailedBuffer": "5",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
            "DeviceType": "",
            "ArticleNumber": "",
            "ArticleStatus": "",
            "UpTime": "6.953222953505516e-310",
[... 239 lines cut ...]
                    "MinimumAmplitude": "42",
                    "ReduceMotionArtifacts": "false",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_low",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}
```

graugans, 2016-02-10:
@bigcmos may I ask what is your target frame rate?

When you need a higher frame rate it may worth to check with the latest firmware 1.3.xxx. We did some performance optimizations in this release. You can download form the [ifm website](https://www.ifm.com/ifmus/web/dualis-download.htm) But I am not sure if the libo3d3xx is compatible with this release. Maybe @tpanzarella does know this.

Regarding the higher frame rates in maybe @cfreundl can shed some light on this he did all the performance measurements.

bigcmos, 2016-02-10:
@graugans Targeting minimum 20Hz or higher.  I realize it's a trade off of frequency, distance, number of exposures.  Any tips you have would be appreciated.

I was able to get 26 Hz on Windows by using `under5m_low`.  I am not super happy with the quality of under5m_low as it creates artifacts near the sensor along the view frustum (see issue https://github.com/lovepark/libo3d3xx/issues/22).  I also decreased the exposure time to `5181` which seems to create more noise in the signal (flat surfaces are less flat and more a cluster of points around the surface).

Ideally, I would love to be able to get the great signal (low noise, ideal distance, no frustum artifacts) I see using `upto30m_moderate` but with 20-30Hz output.  I would be ok sacrificing the number of points produced to achieve this rate.

cfreundl, 2016-02-10:
There are a few things to consider:
* The frame rate indicated by the VisionAssistant is usually higher than the achievable frame rate. That is because the application edit screen puts the camera into "Live Image" video mode (selected by the "Play" button) where the camera is triggered with only 5 fps. In that case the overall system load is relatively low and the processing time per frame is much lower than when the system load is high. The VisionAssistant just converts the reported processing time to fps. In that screen you can get more realistic results if you change the video mode to "React to trigger", there the camera is triggered just as it would be in run mode.
* I second @graugans suggestion to try the latest firmware. With the latest 1.3 version that is not yet available at the website (it is just more stable, performance-wise there should be no difference) I was able to get 19.5 fps, measured with a Python PCIC client running on Linux.
* There is a limit on the frame rate imposed by the maximum duty cycle of the illumination, which is currently 50%. In your sample application (under5m_low, exposure time 5181 µs) the overall exposure time of a frame is a little more than 21.7 ms (in that exposure mode a frame consists of 4 phases with the defined exposure time plus a warmup phase), so you can trigger a frame at most every 43.3 ms which is equivalent to a frame rate of 23 fps. The maximum frame rate goes even more down if you increase the exposure time or switch to other exposure modes with more exposures of used frequencies.

tpanzarella, 2016-02-10:
@bigcmos @graugans In terms of support for the 1.3.x firmware, the xml-rpc interface will almost certainly be incompatible, but I would expect the framegrabber to work just fine. In our upcoming release we will be up-to-date with the latest firmware posted on the IFM website. Stay tuned.

@bigcmos Another factor that may be at play in terms of performance is, for the currently released version of libo3d3xx, under the hood, the framegrabber sets a known PCIC output schema on the camera before streaming back the data. This output schema contains, in addition to the images your above output schema reports, the raw amplitude image. So, for every "frame" you are getting an additional image sent over the wire with libo3d3xx. There may also be additional on-camera overhead incurred to support this as well (not sure). Again, in the upcoming release, the library will support user-tuneable schemas so the library will only stream back specifically what you want it to deliver. Implementation-wise, the framegrabber code uses Boost ASIO which is just a thin wrapper over Linux epoll using edge-triggered I/O. Once the data are received by the framegrabber it is presented to the client in a no-copy manner.

graugans, 2016-02-10:
@bigcmos Just a question. What data do you need from the camera Amplitude + [X,Y,Z] or just [X,Y,Z]? Does your client (PC) has enough horsepower to calculate the [X,Y,Z] with a camera model? So this may could lower the bandwidth used on the Ethernet an increase general throughput as suggested by @tpanzarella already. So we could skip calculation of [X,Y,Z] in the camera and save some CPU Cycles. But the upper limitation at the moment is the allowed LED duty cycle as @cfreundl already said. The LEDs will degenerate over the time when going over the 50% duty cycle limit.

tpanzarella, 2016-02-11:
@bigcmos If you want to play with selectively streaming back data from the camera (i.e., to compute the Cartesian data off-board) I just committed to my dev branch (`dev-0.3.0`) the code that will allow you to do that. Use at your own risk :-) ... but let me know how it goes! For a cookbook-style tutorial on how to do this Cartesian computation, you can look at the unit test code for: `ImageBuffers_Tests.ComputeCartesian` in `test/o3d3xx-image-tests.cpp`.

I'm going to work on getting this code base compatible with the latest IFM firmware too -- @graugans and @cfreundl note that there are performance enhancements in there too. I'll let you know when that is committed to `dev-0.3.0` as well.

Please note, I have not exposed this stuff through the ROS interface yet, that will come after this all gets released into `libo3d3xx`. I just wanted to let you know b/c I know you are actively looking at performance and this may give you another few knobs to turn.

tpanzarella, 2016-02-11:
@bigcmos The `dev-0.3.0` branch is updated to be compatible with the `1.3.x` firmware. In this context, the definition of "compatible" is that all the unit tests are passing. If you decide to try this stuff, let me know if you run into any issues. But at this point, you now have access to IFM's latest firmware which has performance enhancements in addition to the performance knobs mentioned in my comment above.

closed 2020-07-27 by theseankelly