ifm/libo3d3xx #42: Writing pcd to file causing sample program to stop seeing the camera...

wegunterjr, 2016-02-11:
Using the sample code provided here:
https://github.com/lovepark/libo3d3xx/blob/master/doc/using.md

I added some code to write pcd to file, but for some reason, when i do that, I am unable to find the camera.  So this is my code, super simple:  

`// now stream the data
while (true)
{
  // block for .5 sec waiting for data from the camera
  if (! fg->WaitForFrame(buff.get(), 500))
    {
      continue;
    }

  // Now access to the current frame is available as:
 // Object for storing the point cloud.
	pcl::PointCloud<pcl::PointXYZ>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZ>);
 
	// Read a PCD file from disk.
	if (pcl::io::loadPCDFile<pcl::PointXYZ>(argv[1], *cloud) != 0)
	{
		return -1;
	}
 
	// Write it back to disk under a different name.
	// Another possibility would be "savePCDFileBinary()".
	pcl::io::savePCDFileASCII("output.pcd", *cloud);
}`

tpanzarella, 2016-02-11:
I don't think I understand what you are trying to do. The code you posted above is not clear to me. A few comments none-the-less:

1. The point type used in the cloud by libo3d3xx is: `pcl::PointXYZI` which you really should access as `o3d3xx::PointT`. We write the amplitude data to the cloud as the intensity channel.

2. You can test your ability to stream data from the camera with the `o3d3xx-hz` application. If this works, take a close look at what you are doing in your code to be sure you are not shutting down the frame grabber or doing something else that would cause the above timeouts. A side effect of running `o3d3xx-hz`, it will also let you know the health of your link to the camera. Perhaps you are timing out because 500 ms is not enough time to get a frame (shouldn't be). Are you on a noisy network? Streaming over WiFi?

wegunterjr, 2016-02-11:
I wanted to see if I could capture a frame and then write it to a file.  I imagine I will do most of my work with the data from the camera and interpret it without actually writing to a file, but what if I wanted to...

Here is the code..just added the write to file:

`#include <cstdint>
#include <fstream>
#include <functional>
#include <memory>
#include <mutex>
#include <stdexcept>
#include <sstream>
#include <string>
#include <vector>
#include <complex>
#include <iostream>
#include <boost/filesystem.hpp>
#include <boost/thread/thread.hpp>
#include <pcl/io/pcd_io.h>
#include "o3d3xx.h"
#include "o3d3xx/image.h"
#include "o3d3xx/Dump.h"

int
main(int argc, char** argv)
{
	// Object for storing the point cloud.
	pcl::PointCloud<pcl::PointXYZI>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZI>);

// instantiate the camera with default ip, xmlrpc port, and password
o3d3xx::Camera::Ptr cam = std::make_shared<o3d3xx::Camera>();

// associate the camera with a frame grabber
o3d3xx::FrameGrabber::Ptr frameGrabber = std::make_shared<o3d3xx::FrameGrabber>(cam);

// create an buffer for holding the image data from the camera
o3d3xx::ImageBuffer::Ptr buff = std::make_shared<o3d3xx::ImageBuffer>();
while(true)
{
if (! frameGrabber->WaitForFrame(buff.get(), 1500))
	        {
	          continue;
	        }

	cloud = buff->Cloud();

	// Write it back to disk under a different name.
	// Another possibility would be "savePCDFileBinary()".
	pcl::io::savePCDFileASCII("output.pcd", *cloud);

}
}
`

tpanzarella, 2016-02-11:
Your code is not coming through clear as pasted in above. However, I can glean that you are including `o3d3xx/Dump.h` which is one of the headers generated from the ROS IDL compiler to expose the `/Dump` service. So again, it is not clear, is this a standalone program using `libo3d3xx` or is this a ROS node? If the latter, and you are calling the `/Dump` service while also streaming data from the camera, the reason for the timeouts is because to do this you need to keep switching the camera between "run" mode and "edit" mode. Every time you switch to "edit" mode, the TCP connection with the camera's process interface is broken. To that end, you need to re-init the framegrabber. You can use the same shared pointer to do so, just rewrap a new raw pointer. Look at the code for the `o3d3xx_node` in the ROS repository to see how it is handled there. Also, if you want to write files to disk, you can always use the `file_writer_node` or look at that code to serve as your working example as to how to write PCD files to disk.

wegunterjr, 2016-02-11:
ah dang. Totally missed that bad code paste.    Not trying to use ROS, just `libo3d3xx` and PCL.

tpanzarella, 2016-02-23:
With the release of `0.4.0`, a new `examples` module has been added to the repository. There is an example called `ex-file_io.cpp` that shows to write the image data out to a file. Of course, this is based upon the new code structure, but, even if you are on an older version of the `libo3d3xx` code, once you have your `ImageBuffer` populated, the I/O bits are the same. I'm going to close this for now.

closed 2016-02-23 by tpanzarella