ifm/libo3d3xx #43: Cannot run custom applications under the new firmware

bigcmos, 2016-02-22:
After upgrading to the new firmware I can no longer run custom applications on the o3d303.  None of my old applications that I had created worked (they just fail to run).  I factory reset the sensor and the only modification I've made is using DHCP.  The Sample Application runs as expected.  If I duplicate the sample application it runs as expected.  As soon as I make a modification to the cloned sample application (switch to upto30m) it no longer runs.

In IFM assistant I can launch the applications and the data looks good after changing the settings.  As soon as I switch from applications screen to monitor screen the sensor stops outputting and has a frozen in time point cloud from whatever settings I selected in the application.  From the command line I cannot set the active application.

See configuration below. Sample Application is selectable, Sample Application 2 is not selectable.  Sample Application 2 was created using IFM Vision Assistant on Windows by changing the max background distance to upto30m.

**Versions**
Firmware: _1.3.1001_
IFM Vision Assistant: _1.2.11.0_
libo3d3xx: _9e58393 2016-02-15 (HEAD, tag: v0.3.0, origin/master, origin/HEAD, bigcmos/master, master) | updated 100k docs_

**Configuration**
```
export O3D3XX_IP=172.16.78.46
$ o3d3xx-ls
  [1] id=1285362191, name=Sample Application, description=
  [2] id=487861873, name=Sample Application 2, description=
$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"1"}}}' | o3d3xx-config
$ o3d3xx-ls
* [1] id=1285362191, name=Sample Application, description=
  [2] id=487861873, name=Sample Application 2, description=
$ o3d3xx-hz
FrameGrabber running at: 4.78241 Hz
10 frames captured; averaged over 1 runs
$ echo '{"o3d3xx":{"Device":{"ActiveApplication":"2"}}}' | o3d3xx-config
$ o3d3xx-ls
  [1] id=1285362191, name=Sample Application, description=
  [2] id=487861873, name=Sample Application 2, description=
$ o3d3xx-hz
Timeout waiting for camera!
$ o3d3xx-dump
{
    "o3d3xx":
    {
        "libo3d3xx": "300",
        "Date": "Mon Feb 22 14:29:12 2016",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_002_12345678_008022169",
            "Illumination": "#!02_I300_001_01867212_008001175",
            "MiraSerial": "2830-45c0-0c05-0820",
            "MACAddress": "00:02:01:40:1C:A1",
            "Frontend": "#!02_F341_002_12345678_008022593",
            "Diagnose": "#!02_D322_C25_12345678_008022488",
            "Connector": "#!02_A300_001_01867216_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.3.1001",
            "Linux": "Linux version 3.11.0-00005-g2394d14 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Wed Nov 25 16:10:41 CET 2015",
            "Diagnostic_Controller": "0.3.87",
            "ELDK": "GOLDENEYE_ELDK-2015-11-26_16-11-02-5321dac5357219d1408dbf7f53f164e8c138302a",
            "Calibration_Version": "0.6.3",
            "Calibration_Device": "00:02:01:40:1c:a1",
            "Main_Application": "0.4.100",
            "Algorithm_Version": "1.2.6"
        },
        "Device":
        {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "0",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "15",
[... 116 lines cut ...]
                    "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
                    "ReduceMotionArtifacts": "false",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "upto30m_moderate",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}
```

tpanzarella, 2016-02-22:
Take a look at lovepark/libo3d3xx#29 It may apply. I will try to take a look at your configuration when I have the opportunity as well.

bigcmos, 2016-02-22:
Thanks!  Looks like the same issue I am having, see error logs below.

I used to be able to specify 30Hz as the requested frame rate and then modify the settings without an issue.

Is there a way to specify the requested frame rate to be infinite (aka as fast as you can send)?  I'm tweaking the settings to get something close to 30Hz, and would rather not have to set this value.

```
$ curl --data "method=getTraceLogs&params=[-1]" http://172.16.78.46/api/rpc/v1/com.ifm.efector/
<div style="border: 1px solid; padding: 2px"><b>Response</b><br>{
    "error": null,
    "id": null,
    "result": [
        "T: 11.4015, P: 722,  C: WARN,  S: qt_dbus_web_bridge,  M: Starting ...",
        "T: 11.4024, P: 726,  C: WARN,  S: iom,  M: Program starting ...",
        "T: 11.5095, P: 726,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_VFE_ON_ALARM",
        "T: 11.5119, P: 726,  C: WARN,  S: diagCtrl,  M: src/lib/uc/drivers/config/drv_cfg.c[457]: calib_dac failed",
        "T: 11.5127, P: 726,  C: WARN,  S: iom,  M: DIAGC_DBUS_EVENT_VFE_ON_ALARM",
        "T: 12.0382, P: 757,  C: WARN,  S: suprge,  M: suprge: supervision activity starting, watchdog unfused ...",
        "T: 12.6025, P: 749,  C: WARN,  S: gecco,  M: Starting ...",
        "T: 12.6557, P: 743,  C: WARN,  S: imager,  M: Starting ...",
        "T: 18.2048, P: 749,  C: WARN,  S: gecco,  M: Imager is available again",
        "T: 18.2065, P: 743,  C: WARN,  S: imager,  M: LRAE is available again",
        "T: 18.2943, P: 749,  C: WARN,  S: gecco,  M: Finished imager configuration",
        "T: 55.3146, P: 749,  C: ERROR,  S: gecco,  M: bool ifm::SmartCamera::activateApplication() faults.size() > 0 ",
        "T: 55.3261, P: 749,  C: ERROR,  S: gecco,  M: bool ifm::SmartCamera::activateApplication()  ID:  QVariant(int, 101055) ",
        "T: 55.3264, P: 749,  C: ERROR,  S: gecco,  M: bool ifm::SmartCamera::activateApplication()  Text:  QVariant(QString, \"LED duty cycle check: maximum allowed frame rate for these exposure settings: 20 fps\") ",
        "T: 66.8545, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property ContinuousAutoExposure from object imagerconfig_109695398_1: error 1007\" ",
        "T: 66.8609, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property IOTeachLight from object application_109695398: error 1007\" ",
        "T: 67.0243, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 203.8655, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property ContinuousAutoExposure from object imagerconfig_109695398_1: error 1007\" ",
        "T: 203.8725, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property IOTeachLight from object application_109695398: error 1007\" ",
        "T: 204.0022, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 277.7372, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property ContinuousAutoExposure from object imagerconfig_2041161455_1: error 1007\" ",
        "T: 277.7411, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property IOTeachLight from object application_2041161455: error 1007\" ",
        "T: 277.8384, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 293.1502, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property IOTeachLight from object application_2041161455: error 1007\" ",
        "T: 293.2574, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 320.6488, P: 749,  C: WARN,  S: gecco,  M: \"Could not retrieve property IOTeachLight from object application_2041161455: error 1007\" ",
        "T: 320.7454, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 324.4513, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 324.7300, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 324.8340, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 346.8935, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 347.2148, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 353.0202, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 357.7178, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 362.0018, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 372.6825, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 375.7587, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
        "T: 378.6412, P: 743,  C: WARN,  S: imager,  M: void ifm::ModelHandler::initialize(uint32_t) : could not retrieve TemplateInfo ",
[... cut ...] u id) -> (is) status, s resultData</li><li>org.freedesktop.DBus.Properties.Get(s interface_name, s property_name) -> v value</li><li>org.freedesktop.DBus.Properties.Set(s interface_name, s property_name, v value) -> </li><li>org.freedesktop.DBus.Properties.GetAll(s interface_name) -> a{sv} values</li><li>org.freedesktop.DBus.Introspectable.Introspect() -> s xml_data</li><li>org.freedesktop.DBus.Peer.Ping() -> </li><li>org.freedesktop.DBus.Peer.GetMachineId() -> s machine_uuid</li></ul></div>
```

cfreundl, 2016-02-23:
We are aware that the way the LED duty cycle check is performed is quite annoying, both when you configure an application in the VisionAssistant and also when you use the XMLRPC interface and an invalid application just does not start to run because of that.
We changed the behavior there, for free-run applications a too high frame rate will no longer make the application invalid, instead the device takes care itself to limit the frame rate internally. This behavior will be available in the upcoming 1.4.x firmware release.

bigcmos, 2016-02-23:
Looking forward to it.  Thanks.

closed 2016-02-23 by bigcmos