ifm/libo3d3xx #44: Intrinsic calibration parameters of o3d303 Sensor.

bigcmos, 2016-02-24:
How do you calibrate the o3d303 sensors and are the parameters available?

I have been trying to calibrate the camera using the raw amplitude image and a checkerboard pattern with little success.

I need to find the camera matrix including fx, fy, cx, cy as per:  http://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html

Does the camera have a calibration file I can use?  Do these parameters change if I am in 23k mode (binning) vs 100k mode?

graugans, 2016-02-24:
Are you aware of the imager parameters: ``EnableRectificationAmplitudeImage`` and ``"EnableRectificationDistanceImage`` those will perform such an translation internally. Both are disabled by default for the camera version

```
http://<IP-of-the-camera>/api/rpc/v1/com.ifm.efector/session_<session id>/edit/application/imager_001/
getAllParameters()
```

```json
{
    "Channel": "0",
    "ClippingBottom": "131",
    "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
    "ClippingLeft": "0",
    "ClippingRight": "175",
    "ClippingTop": "0",
    "ContinuousAutoExposure": "false",
    "EnableAmplitudeCorrection": "true",
    "EnableFastFrequency": "false",
    "EnableFilterAmplitudeImage": "true",
    "EnableFilterDistanceImage": "true",
    "EnableRectificationAmplitudeImage": "false",
    "EnableRectificationDistanceImage": "false",
    "ExposureTime": "169",
    "ExposureTimeList": "169",
    "FrameRate": "5",
    "MaxAllowedLEDFrameRate": "298.3",
    "MinimumAmplitude": "42",
    "Resolution": "0",
    "SpatialFilterType": "0",
    "SymmetryThreshold": "0.4",
    "TemporalFilterType": "0",
    "ThreeFreqMax2FLineDistPercentage": "80",
    "ThreeFreqMax3FLineDistPercentage": "80",
    "TwoFreqMaxLineDistPercentage": "80",
    "Type": "under5m_low"
}
```
At the moment we do not provide a interface to retrieve the internal calibration data. If the above solution is not feasible. Please feel free to contact your local ifm sales contact to  start an discussion about this feature.

bigcmos, 2016-02-24:
yes, I am aware of those and while useful the camera matrix is needed for my work.  I'll contact IFM.

BTW, what is the easiest way to get a new session ID?
```
$ curl --data "method=requestSession&params=[]"  http://${O3D3XX_IP}/api/rpc/v1/com.ifm.efector/
<div style="border: 1px solid; padding: 2px"><b>Error</b><br>{
    "error": "No such method 'requestSession' in any interface at object path '/' (signature '')",
    "id": null,
    "result": null
}
```

GarrettPlace, 2016-02-24:
@bigcmos - consider your ifm contact notified.

cfreundl, 2016-02-24:
@bigcmos The RPC `requestSession` requires two string parameters: a desired session ID and the device password. If neither is required you can leave the strings empty but they need to be present, i.e. try `method=requestSession&params=["",""]` as GET request.

graugans, 2016-02-24:
@GarrettPlace :thumbsup:
@bigcmos During design of the XML-RPC interface one requirement from my side was using predefined session IDs to simplyfy commandline interfacing. So you can create an session ID offline  like this:

```bash
SESSION_ID=`echo "O3D300" | md5sum | cut -d ' ' -f1`
```
create a file called **requestSession.xml** with this content:
```xml
<?xml version="1.0"?>
<methodCall>
  <methodName>requestSession</methodName>
  <params>
    <param>
      <value>
        <string/>
      </value>
    </param>
    <param>
      <value>
        <string>2a92506f4de075c9318cf91585080672</string>
      </value>
    </param>
  </params>
</methodCall>
```
After this call curl with the following command line:
```
curl -H "Content-Type: application/xml" -X POST --data-binary @requestSession.xml  http://192.168.0.70/api/rpc/v1/com.ifm.efector/
```

 :gem: But if you need a simple tool to play around just point your web browser of choice to the URL:
```
http://<camera-ip>/api/rpc/v1/com.ifm.efector/
```

bigcmos, 2016-02-24:
@cfreundl No dice.

```
$ curl --data "method=requestSession&params=["",""]"  http://92.168.0.69/api/rpc/v1/com.ifm.efector/
<div style="border: 1px solid; padding: 2px"><b>Error</b><br>{
    "error": "No such method 'requestSession' in any interface at object path '/' (signature '')",
    "id": null,
    "result": null
}
```

also tried the below with the same result.

```
$ curl --data "method=requestSession&params=['','']"  http://92.168.0.69/api/rpc/v1/com.ifm.efector/
$ curl --data "method=requestSession&params=['','2a92506f4de075c9318cf91585080672']"  http://${O3D3XX_IP}/api/rpc/v1/com.ifm.efector/
```

bigcmos, 2016-02-24:
@graugans 
Tried the session ID thing which seems to work (though the session timeout is only 10 or 30 seconds. Can I increase this?).
```
$ curl -H "Content-Type: application/xml" -X POST --data-binary @requestSession.xml  http://92.168.0.69/api/rpc/v1/com.ifm.efector/
<?xml version="1.0" encoding="UTF-8"?><methodResponse><params><param><value><string>2a92506f4de075c9318cf91585080672</string></value></param></params></methodResponse>
```

When I hit the web page I get a list of methods I can run:

```
http://192.168.0.69/api/rpc/v1/com.ifm.efector/session_2a92506f4de075c9318cf91585080672
```

From there I visit the webpage for the application and try to getParameters and I get the error:

```
http://192.168.0.69/api/rpc/v1/com.ifm.efector/session_2a92506f4de075c9318cf91585080672/edit/application/imager_001

id: 1
method: getAllParameters
params: []

{ "error": "No such object path '/session_2a92506f4de075c9318cf91585080672/edit/application/imager_001", "id": "1", "result": null }
```

Do I need to do something before editing the application?

tpanzarella, 2016-02-24:
@bigcmos You can set the session timeout using the usual `o3d3xx-dump` then `o3d3xx-config` (or just pipe the proper json into `o3d3xx-config`). You want to edit the `SessionTimeout` parameter on the `Device

graugans, 2016-02-24:
@bigcmos I am not having a sensor by hand. You can change the timeout with the method ``heartbeat`` on the session object. The parameter is in seconds and maximum value is ``300``. But as @tpanzarella said better use the commandline tools to tweak stuff.

tpanzarella, 2016-02-24:
@bigcmos Seems like we are pretty far off the *intrinsics reservation* at this point, however, for what it is worth, if you are looking to poke at the XML-RPC server on the camera interactively, you could also use an `ipython` shell. For example:

```
In [1]: from xmlrpclib import ServerProxy

In [2]: URL = "http://192.168.0.69/api/rpc/v1/com.ifm.efector"

In [3]: main = ServerProxy(URL)

In [4]: id = main.requestSession("","")

In [5]: id
Out[5]: 'fb812dfbef54f5df71e106bf79563c47'

In [6]: sess = ServerProxy("%s/session_%s" % (URL, id))

In [7]: sess.heartbeat(300)
Out[7]: 300

In [8]: sess.cancelSession()
Out[8]: ''

```

jessie1992, 2020-01-02:
I have the same issue as @bigcmos. How do I get the intrinsic calibration parameters now?

graugans, 2020-01-07:
Hey @jessie1992 my recommendation is to use the new [ifm3d](https://github.com/ifm/ifm3d) library which comes with support for retrieving the intrinsic parameters which is shown in this Jupyter Notebook: https://github.com/ifm/ifm3d-examples/blob/master/image_rectification/python/Image_Rectification.ipynb

closed 2020-01-07 by graugans