ifm/libo3d3xx #45: Resuming frame grabbing after the camera is power cycled

cmdunkers, 2016-02-24:
More of a question...Using the examples I am able to get frames from the camera, however if I power cycle the camera without restarting my code the frame grabber always times out when I try to get a frame. I looked through the documentation and saw that the framegrabber has to be re instantiated after being stopped, but I have not stopped the frame grabber. Is there a way to handle this edge case and if so what needs to happen in the code?

tpanzarella, 2016-02-24:
If you send me a concrete code example, I could speak specifically to it, however, in general one approach you could take is what we do in the ROS wrapper's [main publishing loop](https://github.com/lovepark/o3d3xx-ros/blob/master/src/o3d3xx_node.cpp#L142) The basic idea being, if you timeout `x` number of times or don't get a frame for `y` number of seconds, you explicitly restart the framegrabber. The trick is that `o3d3xx::FrameGrabber::Ptr` is a `std::shared_ptr` wrapping an `o3d3xx::FrameGrabber` pointer. So, you call `fg.reset(new o3d3xx::FrameGrabber(cam))` on the shared pointer reference. This implicitly runs the the originally wrapped pointer's dtor and constructs a new one. By constructing  a new `FrameGrabber` the TCP connection is reestablished with your camera once the PCIC daemon is back up and running.

Hope this helps.

tpanzarella, 2016-02-24:
Since this will likely be generally useful, I just added a new example: `modules/examples/ex-framegrabber_recycling.cpp` that demonstrates the technique above in a more isolated fashion (i.e., not clouded by all the ROS specifics.

cmdunkers, 2016-02-24:
I implemented what you suggested and it worked, thanks for the quick response

closed 2016-02-24 by tpanzarella