ifm/libo3d3xx #46: Syncronizing Multiple o3d303 sensors over 6Hz

bigcmos, 2016-02-24:
Is there a way to run these sensors together faster than 6Hz without interference on the same frequency channel?  If I add more sensors am I constrained by some global maximum frame rate?  I realize that I can use the group and frequency settings to get up to 3 different channels, but I want to understand the limitations per channel of how many I can deploy into a small space.

I have a setup of 4x o3d303 pointing at a table and I'm trying to collect data from all of them as fast as possible.  They are chained together 01 -> 02 -> 03 -> 04 where 01 is generating my trigger signal and is  running in continuous mode and the others trigger on a positive edge from the output of the previous.  While the config I am using allows the sensors to run at over 20Hz, you can see from the images and video below the aberrations that occur when the sensors run at 7Hz.

**Images**
o3d3xx 01 @ 5Hz
[image]

o3d3xx 01 @ 7Hz
[image]

**Oscilloscope on trigger lines for 03 and 04 running @ 7Hz**
**Red** is the trigger signal output from sensor 02 into sensor 03.
**Yellow** is the trigger signal output from sensor 03 into sensor 04.
[image]

**Video**
[Triggered o3d3xx 5Hz](https://www.youtube.com/watch?v=NayL6q2G0N0&feature=youtu.be)
[Triggered o3d3xx 7Hz](https://www.youtube.com/watch?v=-iaPn1N5Da8&feature=youtu.be)

**Configurations for 7 Hz**
[o3d3xx-172.16.78.46.json.txt](https://github.com/lovepark/libo3d3xx/files/145442/o3d3xx-172.16.78.46.json.txt)
[o3d3xx-172.16.78.47.json.txt](https://github.com/lovepark/libo3d3xx/files/145441/o3d3xx-172.16.78.47.json.txt)
[o3d3xx-172.16.78.48.json.txt](https://github.com/lovepark/libo3d3xx/files/145440/o3d3xx-172.16.78.48.json.txt)
[o3d3xx-172.16.78.49.json.txt](https://github.com/lovepark/libo3d3xx/files/145443/o3d3xx-172.16.78.49.json.txt)

manujnaman, 2016-02-25:
We were running two sensors chained at 15Hz pointed in the same direction with average scene depth of ~5m. Sensors would start fairly independent but in around 30min or so started interfering with each other. If one sensor was switched off interference would go away. At this point we are using sensors pointing in different directions.

bigcmos, 2016-02-25:
I have also tried with the setting upto30m with four sensors and can only get 2Hz before the sensors start interfering.  I would love some advice on how to improve the capture rate for multiple sensors.

bigcmos, 2016-02-26:
Here is some output from the upto30m test I ran today.  You can see the sensors are roughly registered running at 2Hz.  The 4th camera, in _yellow_, shows temporal aberrations which are caused by interference from the other sensors.  About 25s into the video when we turn off the visualization for 04 you can see the great data we are getting from the other sensors.

If I increase the frame rate to more than 2Hz all of the sensors demonstrate similar aberrations.  I've also had limited success with upto30m using different frequency channels for group use.

Any tips on increasing the frame rate for multiple sensors while maintaining clean data?

**Camera Colors**
01 _Blue_
02 _Red_
03 _Green_
04 _Yellow_

https://www.youtube.com/watch?v=Ctit2ek9NH0

[image]
[image]

**Sensor Configs**
[o3d3xx-172.16.78.46.json.txt](https://github.com/lovepark/libo3d3xx/files/147554/o3d3xx-172.16.78.46.json.txt)
[o3d3xx-172.16.78.47.json.txt](https://github.com/lovepark/libo3d3xx/files/147553/o3d3xx-172.16.78.47.json.txt)
[o3d3xx-172.16.78.48.json.txt](https://github.com/lovepark/libo3d3xx/files/147555/o3d3xx-172.16.78.48.json.txt)
[o3d3xx-172.16.78.49.json.txt](https://github.com/lovepark/libo3d3xx/files/147556/o3d3xx-172.16.78.49.json.txt)

graugans, 2016-02-26:
Have you chained the sensor the way I've described in issue #38

bigcmos, 2016-02-26:
Yes, it's connected exactly like that.  Camera 01 is in continuous mode, 02, 03, 04 are triggered on positive edge hardware trigger.

graugans, 2016-02-26:
I have a suspicion, in continuous mode we do image acquisition and calculation in parallel for some modes. To achieve higher frame rates. This may cause the  interference maybe one option is to run all the sensors in hardware trigger mode and initially kick-start the chain by a push button connected by a pull-down resistor to GND. Assuming the ``cascading pin`` is ``24V`` in idle mode. 
[image]

bigcmos, 2016-02-26:
This will solve the temporal aberrations seen in the 4th camera, but this will slow down the frame rate of the system even more...

I'll give it a try and let you know, but would love a solution to increase the frame rate.  I need to hit a frame rate between 10-20Hz per camera minimum.

graugans, 2016-02-27:
The H/W trigger set-up should achieve the highest possible frame rate because the cascading pin is connected to the state when the image acquisition is finished. It is not when image calculation is done. So in theory with small gaps one camera is always tacking images. 

[image]

cfreundl, 2016-02-28:
Before I have a chance to test this or a similar setup at work, here are some thoughts in advance.

I took the configuration of the applications named "lessthan5_low26" ( @bigcmos These should be the correct ones, as this is the only one where the first camera is set to continuous and the others are set to hardware trigger), and computed the overall illumination time. The first three cameras are set to 5000 µs per phase and the last one to 5181 µs. Again, we need to keep in mind that an exposure in the mode "under5m_low" consists of four phases plus dummy phases with 1000 µs exposure time altogether. This sums up to 21 ms for the first three cameras and 21.7 ms for the fourth camera, and 84.7 ms in total. So, if we would just use the total requred time for the illumination itself, we get an upper bound of 11.8 Hz for the achievable frame rate.

However, this still does not consider the readout time per phase into account, I need to check that with our system engineers. Additionally, another delay comes from the fact that we detect the completion of the acquisition when the last phase of sequence is received and verified by the process on the main processor. I will see how much this contributes to the total time required for a single overall result but I guess even in the optimal case this will limit the frame rate to less than 10 Hz in that setup.

@bigcmos Making use of the group/channel settings is nothing that you absolutely cannot do, right? Currently there is probably no other way to achieve higher frame rates.

graugans, 2016-03-22:
Our system engineers made some tests and we have a set-up working with 4 cameras. Two of the four cameras are triggered at the same time. All off the cameras are configured with the same settings. This is needed to have no overlap. When we trigger the two cameras at the same time the interference is minimized. Because of this we need a hardware trigger set-up like you see in this [video](https://www.youtube.com/watch?v=o3qImD9baTQ&feature=youtu.be)

bigcmos, 2016-04-08:
Thanks for replicating the setup and giving more insight into how to achieve the frame-rate.  I will be able to use different frequencies for some installs, but not always.

I've rewired to make the sensors to replicate your circular chain so that both sensors are in Trigger mode 3 (positive edge trigger).  To simplify my setup I'm only using Sensors 1 & 3 which are catty corner pointed at the center of a conference table.  I'm running using application lessthan5_low26 which is set to 5000 µs per phase which should be about 21 ms per camera and 42 ms after both cameras are done.  From your notes above this should theoretically yield about 23 Hz per sensor.  After a small loss factor I should expect to get near 20 Hz which is acceptable for our use case.

When running in this mode though I'm only getting about 12 Hz which is about 1/2 of what I would expect is possible.  Am I missing something or did I do the math wrong?  I would love to be able to synchronize 2 sensors on the same frequency with each sensor running at 20 Hz.

```
acme@dj:~$ o3d3xx-hz --ip 172.16.78.46
FrameGrabber running at: 11.5969 Hz
10 frames captured; averaged over 1 runs
acme@dj:~$ o3d3xx-hz --ip 172.16.78.48
FrameGrabber running at: 11.6334 Hz
10 frames captured; averaged over 1 runs
```

json dumps from the two sensors below.

[cage-03.json.txt](https://github.com/lovepark/libo3d3xx/files/210976/cage-03.json.txt)
[cage-01.json.txt](https://github.com/lovepark/libo3d3xx/files/210977/cage-01.json.txt)

bigcmos, 2016-04-12:
Also, I have the sensor 1 trigger out connected to sensor 3 trigger in and 3 out to 1 in so they will round robin trigger forever once started.  I was able to start the trigger from the Windows app, but I wanted to be able to trigger from linux.

I tried the trick from https://github.com/lovepark/libo3d3xx/issues/39 by sending a command over telnet but it didn't seem to start the triggering.  Should I expect a software trigger to work in this case?

```
$ telnet 172.16.78.46 50010
Trying 172.16.78.46...
Connected to 172.16.78.46.
Escape character is '^]'.
1010L000000007\r\n1010t\r\n
0000L000000007
0000?
^]

telnet> quit
Connection closed.
acme@dj:~$ o3d3xx-hz --ip 172.16.78.46
Timeout waiting for camera!
```

graugans, 2016-04-12:
A combination of multiple trigger sources is not supported. For this case I recommend the hardware trigger and trigger the first in the chain by mapping one of the other Sensors output to the ``PCIC O command`` with this setup you can use this PIN to start the trigger chain.

You set-up the PIN in the GUI like this:
[image]

In the JSON the parameter is called LogicGraph
```
  "LogicGraph": "{
      "IOMap": {
          "OUT1": "PCIC_OUT",
          "OUT2": "AQUFIN"
      },
      "blocks": {
          "B00001": {
              "pos": {
                  "x": 200,
                  "y": 200
              },
              "properties": {
              },
              "type": "PIN_EVENT_IMAGE_ACQUISITION_FINISHED"
          },
          "B00002": {
              "pos": {
                  "x": 198,
                  "y": 79
              },
              "properties": {
              },
              "type": "PIN_EVENT_PCIC_O_CMD"
          },
          "B00003": {
              "pos": {
                  "x": 600,
                  "y": 75
              },
              "properties": {
                  "pulse_duration": 0
              },
              "type": "DIGITAL_OUT1"
          },
          "B00005": {
              "pos": {
                  "x": 600,
                  "y": 200
              },
              "properties": {
                  "pulse_duration": 0
              },
              "type": "DIGITAL_OUT2"
          }
      },
      "connectors": {
          "C00000": {
              "dst": "B00005",
              "dstEP": 0,
              "src": "B00001",
              "srcEP": 0
          },
          "C00001": {
              "dst": "B00003",
              "dstEP": 0,
              "src": "B00002",
              "srcEP": 0
          }
      }
      }
  "
```

To toggle the PIN I do use the following commands in a ruby (irb) shell:
```ruby
require 'o3d3xx'
pcic = O3D3XX::PCIC.new()
pcic.connect('192.168.0.70','50010')
pcic.transfer('p0')

# set PIN 1 to low
irb(main):053:0> pcic.transfer("o010")
# set PIN 1 to high
irb(main):054:0> pcic.transfer("o011")
```

Or you can use this python code:

```python
import socket

IP_ADDR = "192.168.0.69"
PCIC_PORT = 50010
PIN = 1 #1=IO1, 2=IO2, 3=IO3

class PCIC:
    def __init__(self, ipaddr, port=50010):
        #open PCIC socket
        self.pcicSocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.pcicSocket.connect((ipaddr, port))
        self.ticket_no=1000

    def incTicketNo(self):
        self.ticket_no+=1
        if (self.ticket_no>9999):
            self.ticket_no=1000

    def setPin(self, pin, pinState):
        #create PCIC V3 header
        cmd = "o%02d%01d"%(pin, pinState)
        resp = self.send_v3(cmd)
        if (resp != b"*"):
            raise RuntimeError("error setting pin, PCIC answer was " + resp.decode())

    def send_v3(self, cmd):
        #create PCIC V3 header
        data = "%04dL%09d\r\n%04d%s\r\n"%(self.ticket_no, len(cmd)+6, self.ticket_no, cmd)
        self.pcicSocket.send(data.encode())
        #receive PCIC V3 header all asynchronous messages are dropped away
        resp = self.rcv_v3()
        self.incTicketNo()
        return resp

    def rcv_v3(self):
        while (1):
            #receive ticket + length + CRLF
            rcv = self.sock_rcv(16)
            ticket_rcv = int(rcv[:4])
            length = int(rcv[5:-2])
            #receive ticket + cmd + CRLF
            rcv = self.sock_rcv(length)
            if self.ticket_no == ticket_rcv:
                break
        return rcv[4:-2]

    def sock_rcv(self, numBytes):
        buf = b''
        while numBytes != 0:
            rcv = self.pcicSocket.recv(numBytes)
            numBytes -= len(rcv)
            buf = buf + rcv
        return buf

p = PCIC(IP_ADDR, PCIC_PORT)
#create a fast LOW->HIGH->LOW transition at pin
#p.setPin(PIN, 0)
p.setPin(PIN, 1)
#p.setPin(PIN, 0)
```

graugans, 2016-04-12:
The tricky part with this set-up will be the network delay time + OS latencies which maybe longer than one cycle. So you'll have to select pulsed output for the PIN configuration:

[image]

The current firmware may suffers from a Bug for the pulsed outputs.

bigcmos, 2016-04-12:
I'm trying to increase performance and I don't think I want another computer in the loop.  I would like to be able to have two sensors paired up with hardware triggers to get the fastest possible frame rate but  use the software trigger to boot strap the signal.  Should I start a new issue for a feature request to allow a software trigger without using up the digital in?

Also, any thoughts on the performance issues?  I would like to get to 20Hz for each sensors with two of these sensors chained in series.  Is that possible?

graugans, 2016-04-13:
I do not understand your concerns. In both cases you have to execute a ``PCIC`` call for the software trigger it is ``t`` for the mapped ``I/O`` pin it is ``o0<PIN><state>`` You can control both use cases with the host you are grabbing frames. With the next firmware release there is support for the pulsed output so no time constraints in this case.

Maybe we can flatter @tpanzarella to integrate the I/O support into the libO3D3xx and o3d3xx-ros what cloud be interesting for a user of the lib to have an actuator  to control a lamp, pump or horn.

graugans, 2016-04-14:
Regarding Performance, what are you achieve at the moment? Are the configs above still valid? I will ask the system engineering if they can reassemble the set-up so we can check with an oscilloscope.

bigcmos, 2016-05-02:
12Hz for each sensor in a 2 sensor chain.  See https://github.com/lovepark/libo3d3xx/issues/46#issuecomment-207643253

The configs are good from that post.  I can pull oscilloscope readings and share.

I still have two issues:
1) How do I improve performance so that I can run 2 sensors at 20Hz each on the same frequency channel without interference.
2) How do I start the chain of two sensors without having an external button I have to press.

bigcmos, 2016-05-02:
Oscilloscope readings below.  Key takeaway is the capture finished trigger is firing for each sensor between 40-60ms, and it takes about 80-100ms for both sensors to complete.  This yields a capture frequency between 10-12Hz for each sensor.  I am trying to get to at least 24Hz per sensor in this configuration which would mean ~20ms trigger and ~40ms round robin.  From reading [the comment above](https://github.com/lovepark/libo3d3xx/issues/46#issuecomment-189883516)

> "under5m_low" consists of four phases plus dummy phases with 1000 µs exposure time altogether. This sums up to 21 ms

 by @cfreundl it sounded like this was doable.  Is this possible?

[image]

[image]

**Wiring diagram**
[image]

**o3d3xx-hz**
I have been having a lot of trouble getting reliable data out of o3d3xx-hz.  I don't really trust these numbers because I've been getting some wild results (anything between 1hz and 10Hz) and more often than not I get the error message `Timeout waiting for camera!`.

```
$ o3d3xx-hz --ip 172.16.78.46
FrameGrabber running at: 10.7113 Hz
10 frames captured; averaged over 1 runs
$ o3d3xx-hz --ip 172.16.78.48
FrameGrabber running at: 6.15649 Hz
10 frames captured; averaged over 1 runs
```

**o3d3xx-dump**
Including the configurations of the two sensors for completeness.

[o3d-trigger-01.json.txt](https://github.com/lovepark/libo3d3xx/files/245930/o3d-trigger-01.json.txt)
[o3d-trigger-03.json.txt](https://github.com/lovepark/libo3d3xx/files/245931/o3d-trigger-03.json.txt)

tpanzarella, 2016-05-02:
> I have been having a lot of trouble getting reliable data out of o3d3xx-hz. I don't really trust these numbers because I've been getting some wild results (anything between 1hz and 10Hz) and more often than not I get the error message Timeout waiting for camera!.

The code for `o3d3xx-hz` is readily available for you to view. If you believe it is reporting unreliable information, then, we welcome your PR. I have never had reason to believe it misreports information. Frankly, averaging data over a single run like you are doing above is typically a bad idea. The `-r` parameter will let you increase this number of runs.

bigcmos, 2016-05-02:
I haven't had issues in the past, but recently I can't get reliable readings.  I normally use `-r` and sometimes `-f`, but I am frequently getting `Timeout waiting for camera!` and `-r` or `-f` exacerbates the issue.

It's unclear what the root cause is but it started happening more now that I'm running in this new triggering mode.  It's not clear that `o3d3xx-hz` is the problem, and more likely the sensor is refusing to communicate.  I am having similar disconnects from the windows client as well.  Also, after running the sensors for a prolonged period of time (>30 min) in this configuration they will stop triggering.

**example**
```
$ o3d3xx-hz --ip 172.16.78.46 -r 5
Timeout waiting for camera!
```

tpanzarella, 2016-05-02:
`o3d3xx-hz` will "timeout" if it does not receive a frame from the camera w/in 1 second. And this is what you are experiencing in your example above. I suppose a reasonable patch to this code would be to make this timeout configurable. At the time of writing, for my purposes, if it took longer than 1 second to get a frame, this was indicative of a bigger issue (IP routing issue or something like that) as I was not interested in measuring frame rates that slow. I suspect you are in a similar situation -- but I could be mistaken.

Also note, `o3d3xx-hz` takes into consideration how long it takes for network transmission of the frame as well as parsing and validating the byte buffer from the camera -- in this version of the code it does *not* include constructing the images from the raw bytes. So, I would expect inconsistencies between what `o3d3xx-hz` reports and what you are measuring with your scope.

Finally, the PCIC schema also comes into play here. Depending on what images you are streaming from the camera, the statistics may vary. Is it possible that in your 6-camera configuration that you are running into network congestion issues? I have not done the math, but 6 cameras streaming the default schema at a high frame rate on the same subnetwork may be contributing to your issues (just a thought).

bigcmos, 2016-05-03:
Good thoughts, but I don't think this issue is network related.  I'm starting to think this is sensor related.  I'm only running 4 cameras in this setup.  Looking at iftop the cameras are using 40Mb/s each on a GB switch so it's not close to saturating the link.  I am able to run these cameras in a non triggering mode at 20Hz and they will all report back ~20Hz from `o3d3xx-hz`.

Also, when running the o3d3xx ROS node I can pull the topic frequency for the generated point cloud.  For sensors 2 & 4 I'm getting ~12Hz which is what I expect by looking at the trigger output on the oscilloscope.  For sensors 1 & 3 I was getting ~5Hz and then the sensor timed out and are not responding.  I'm going to try replacing 1 & 3 and see if that fixes the issue.

bigcmos, 2016-05-03:
Also, I'm still looking for an answer on how to get 24Hz with this triggered setup.

graugans, 2016-05-03:
Sorry for the lag, I had been busy for the last few weeks. I guess running 4 Cameras in 24Hz is not possible at the moment. In the above set-up due to the chaining the frame rate is divided by factor two. You can run up to 3 cameras in parallel, each camera set to an individual channel and configured to hardware trigger.

But in the proposed set-up with 3 cameras all of them have to run in parallel. And this will be the hard part to achieve. At the moment the only chance I see for the moment is to do some AND logic with the ready for trigger signalling. Because if we trigger the sensors when they are not ready for trigger the trigger is lost, we do not queue the triggers, and the cameras will go out of sync. So with 3 cameras we always have to wait for the slowest camera.

graugans, 2016-05-03:
After checking your configs with @cfreundl  we recommend to update to firmware version 1.4.x there are at-least two critical bugs in the 1.3.1001 firmware. I've attached a small sketch to show the idea about synchronizing 3 cameras. 
[image]
The downside in this set-up you'll need a PLC or something capable of fast switching. A [Relay](https://en.wikipedia.org/wiki/Relay) has typical turn on times of 10ms and will not provide 24Hz longer than a few hours.

bigcmos, 2016-05-03:
@graugans Yup, I already have a setup like this.

To be clear I want to run 6 cameras at 24Hz looking at the same scene without interference.  Using the 3 frequency channels I would need to be able to run 2 cameras in serial each at 24Hz which is what I'm trying to reproduce.  Do you think that it is possible to have two cameras running at 24Hz with a future firmware upgrade or is there a hardware limitation?  The 21ms number seemed to indicate it was possible...

graugans, 2016-05-04:
You are right when we manage to run 2 Cameras interleaved in 24Hz up to 6 cameras could be possible. @cfreundl and I talked about this yesterday and we are planning to set-up some testing to check what is the limiting factor of 12Hz at the moment.

cfreundl, 2016-05-06:
@bigcmos I have built a 2-camera setup at my place and can reproduce the measurements, with the current firmware 1.4.1639 I get about 12.7 fps with the same application running on both cameras (`under5m_low` with 5000 µs exposure time).

You can increase the frame rate a little bit, if you set the device parameter `IODebouncing` to `false` on both cameras, which for each camera lowers the reaction time to the trigger coming from the other camera, resulting in an overall frame rate of about 13.7 fps (in my setup).

I am afraid that this is probably the current upper limit for the overall frame rate that can be achieved for the 2-camera cascaded setup. I mentioned the 21 ms overall exposure time in my [previous comment](https://github.com/lovepark/libo3d3xx/issues/46#issuecomment-189883516) but also said that the readout time needs to be added to the acquisition time as well. I tried to measure the readout time with the oscilloscope as good as possible and got about 2.5 ms per phase. As we have 6 phases in the exposure mode `under5m_low`, 15 ms have to be added to the 21 ms of exposure, resulting in 36 ms altogether per camera, and finally 72 ms (or about 13.9 Hz) for the cascaded setup. Which means that with the debouncing turned off we are pretty close to that upper limit.

There are currently investigations on our side to improve the phase sequence in terms of reducing the number of phases to 4. In case this is possible, the overall exposure time would be reduced to 20 ms and the readout time to 10 ms, resulting in an overall frame rate of about 16.7 fps for the cascaded setup but the reduction of the number of phases will certainly add another cost per phase so the upper limit will be a little bit lower.

bigcmos, 2016-05-11:
Thanks for investigating.  Is there any way to do the acquisitions or readouts in parallel?

Other time of flight sensors I have can run at 30Hz, so I don't understand what is fundamentally different about this sensor.

graugans, 2016-05-13:
I suggest  we set-up some telephone conference after the 22.05.2016 when the team is back from vacation.

bayaliev, 2019-05-15:
> The tricky part with this set-up will be the network delay time + OS latencies which maybe longer than one cycle. So you'll have to select pulsed output for the PIN configuration:
> 
> [image]
> 
> The current firmware ~may~ suffers from a Bug for the pulsed outputs.

this is really helpful @graugans, thanks for taking time and posting it!

closed 2020-07-27 by theseankelly