ifm/libo3d3xx #47: Trigger o3d303 by software?

zfxw0206, 2016-03-16:
Now I am using ROS to run 2 ifm cameras and want to look for a way to trigger them.
I found a lot of people trigger ifm by using PIN 2 of the power port. Except for this, I am wondering if there is another software way to trigger it?
Thank you!

graugans, 2016-03-17:
The trigger mode 2 is for triggering the device over Ethernet, the ifm proprietary Process Interface (PCIC) is a TCP/IP connection listening on port ``50010``. This interface is for example used to transmit the result data. You can establish up to five simultaneous connections to the camera. In your case you can start the ``libo3d3xx-framegrabber`` in listening mode and send the trigger command over another connection. For more details on this please check my comments on the issue #39

zfxw0206, 2016-03-17:
Thank you very much for you reply. After reading your comments, I learnt some concepts about the trigger, but I am still confused what shall I do with your code in  "issue #39" and how to send the trigger command over another connection in Ubuntu.

Another problem is when I testing process ($make check), I got the following errors:
[==========] 43 tests from 6 test cases ran. (89106 ms total)
[  PASSED  ] 4 tests.
[  FAILED  ] 39 tests, listed below:
[  FAILED  ] ImportExport_Tests.ImportExportApp
[  FAILED  ] SpatialFilterTest.SpatialFilterConfig_General
[  FAILED  ] SpatialFilterTest.GetSpatialFilterParameters
[  FAILED  ] SpatialFilterTest.GetSpatialFilterParameterLimits
[  FAILED  ] SpatialFilterTest.SpatialFilterConfig_JSON
[  FAILED  ] Camera_Tests.GetAllParameters
[  FAILED  ] Camera_Tests.GetParameter
[  FAILED  ] Camera_Tests.GetSWVersion
[  FAILED  ] Camera_Tests.GetApplicationList
[  FAILED  ] Camera_Tests.RequestSession
[  FAILED  ] Camera_Tests.CancelSession
[  FAILED  ] Camera_Tests.Heartbeat
[  FAILED  ] Camera_Tests.SetOperatingMode
[  FAILED  ] Camera_Tests.GetDeviceConfig
[  FAILED  ] Camera_Tests.ActivateDisablePassword
[  FAILED  ] Camera_Tests.SetDeviceConfig
[  FAILED  ] Camera_Tests.DeviceConfig_JSON
[  FAILED  ] Camera_Tests.GetNetParameters
[  FAILED  ] Camera_Tests.NetConfig
[  FAILED  ] Camera_Tests.NetConfig_JSON
[  FAILED  ] TemporalFilterTest.TemporalFilterConfig_General
[  FAILED  ] TemporalFilterTest.GetTemporalFilterParameters
[  FAILED  ] TemporalFilterTest.GetTemporalFilterParameterLimits
[  FAILED  ] TemporalFilterTest.TemporalFilterConfig_JSON
[  FAILED  ] AppImagerTest.CopyDeleteApplication
[  FAILED  ] AppImagerTest.CreateApplication
[  FAILED  ] AppImagerTest.ChangeAppNameAndDescription
[  FAILED  ] AppImagerTest.EditApplication
[  FAILED  ] AppImagerTest.GetAppParameters
[  FAILED  ] AppImagerTest.AppConfig
[  FAILED  ] AppImagerTest.AppConfig_JSON
[  FAILED  ] AppImagerTest.GetAvailableImagerTypes
[  FAILED  ] AppImagerTest.ChangeImagerType
[  FAILED  ] AppImagerTest.GetImagerParameters
[  FAILED  ] AppImagerTest.GetImagerParameterLimits
[  FAILED  ] AppImagerTest.ImagerConfig
[  FAILED  ] AppImagerTest.ImagerConfigValueOutOfRange
[  FAILED  ] AppImagerTest.ImagerConfig_JSON
[  FAILED  ] AppImagerTest.Exposure
39 FAILED TESTS

Most of them are because of "XMLRPC call timed out", how could I solve the problem? Thank you!

tpanzarella, 2016-03-17:
Your network is not setup correctly. We need more info to help here.

zfxw0206, 2016-03-17:
Could you please tell me which information do you need? How to setup?  Thank you!

tpanzarella, 2016-03-17:
What is your IP address? What is the camera's IP address? Can you `ping` the camera? What does your routing table look like? For the last piece, pasting in the output of `netstat -r` would help.

zfxw0206, 2016-03-17:
My PC have 2 network ports. I connect one to the Internet, the other one to the camera directly. 
My IP address for camera is 10.10.10.81; The camera's IP is 10.10.10.71. I can ping the camera.
Here is the output of nestat -r:
administrator@administrator-desktop:~/libo3d3xx/build$ netstat -r
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
default         10.5.40.1       0.0.0.0         UG        0 0          0 eth1
10.5.40.0       *               255.255.255.0   U         0 0          0 eth1
10.10.10.0      *               255.255.255.0   U         0 0          0 eth0

eth0 is for the camera, eth1 is for the internet.

tpanzarella, 2016-03-17:
OK. So the problem is the non-standard IP address of the camera. To make the unit tests pass, you should set `O3D3XX_IP=10.10.10.71` in your environment prior to running `make check`.

When using the command-line tools, all of them accept the `--ip` argument. Or, you could `export` the above mentioned environment variable and the library will respect that too.

When running via ROS, the `camera.launch` file also accept an `ip` argument on the command-line.

zfxw0206, 2016-03-17:
[image]
My IP address is 10.10.10.81; my camera IP is 10.10.10.71. I can ping it.

tpanzarella, 2016-03-17:
Run the unit tests like:

```
$ O3D3XX_IP=10.10.10.71 make check
```

zfxw0206, 2016-03-17:
Yes, it works. Thank you.
But I still get 3 failed:
1.
[ RUN      ] Camera_Tests.GetDeviceConfig
/home/administrator/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:182: Failure
Value of: 32
Expected: params.size()
Which is: 30
unknown file: Failure
C++ exception with description "_Map_base::at" thrown in the test body.
[  FAILED  ] Camera_Tests.GetDeviceConfig (666 ms)
2.
[ RUN      ] AppImagerTest.GetImagerParameters
/home/administrator/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:265: Failure
Expected: params.at("EnableFastFrequency") doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.GetImagerParameters (1968 ms)
3.
[ RUN      ] AppImagerTest.ImagerConfig
/home/administrator/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:419: Failure
Expected: cam_->SetImagerConfig(im.get()) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.ImagerConfig (1921 ms)

tpanzarella, 2016-03-17:
This is a result of running some combination of an unsupported software stack. The supported versions are outlined in the libo3d3xx README in the software compatibility matrix. This is rendered nicely on the main github page for convenience. That said, please send the output of: (assuming you are in your `build` directory...)

```
$ src/bin/o3d3xx-dump --ip 10.10.10.71 | grep -i 'ifm_software'
```

and

```
$ src/bin/o3d3xx-version
```

zfxw0206, 2016-03-17:
~/libo3d3xx/modules/camera/build$ src/bin/o3d3xx-dump --ip 10.10.10.71 | grep -i 'ifm_software'
            "IFM_Software": "1.1.288",

~/libo3d3xx/modules/camera/build$ src/bin/o3d3xx-version
libo3d3xx: version=0.4.0

tpanzarella, 2016-03-17:
You have two options:

Option 1: Update your camera firmware to version 1.3.1001. It is available in the `firmware` directory. You can flash the camera with IFM's Vision Assistant or with:

```
 $ src/bin/o3d3xx-reboot --ip 10.10.10.71 -r 
```

And then hit the camera's internal HTTP server at `http://10.10.10.71:8080`.

Option 2: Downgrade your version of libo3d3xx to `0.2.0` which you can get on the "releases" tab of the github page.

Option 1 is the recommended approach.

tpanzarella, 2016-03-17:
Actually, a question for @graugans ... when you reboot into recovery mode, will the camera be configured to use the IP address set via the XML-RPC or will it use the default? @zfxw0206 if it is the latter, and if you are using "Option 1", when you hit the HTTP server it will be at `http://192.168.0.69:8080`.

zfxw0206, 2016-03-17:
I have a old version of libo3d3xx. it is 0.1.11. Just now I use this old version and make check again, all tests passed. Thank you!

tpanzarella, 2016-03-17:
I do not recommend that you use `0.1.11`. That has only been tested with IFM firmware: `0.06.39`. Particularly if you are looking for (free) community support, please use a supported software stack as specified in the software compatibility matrix. Thanks.

zfxw0206, 2016-03-17:
Ok, I update it to 1.3.1001 and now all tests for camera module and  passed. But for image module, I got another 2 failures:
1.
[ RUN      ] ImageBuffers_Tests.Extrinsics
/home/administrator/libo3d3xx/modules/image/test/o3d3xx-image-tests.cpp:385: Failure
The difference between reference_extrinsics[0] + 2. and new_extrinsics[0] is 1.2782776653766632, which exceeds abs_error, where
reference_extrinsics[0] + 2. evaluates to 1.5934296548366547,
new_extrinsics[0] evaluates to 0.31515198945999146, and
abs_error evaluates to 1.
/home/administrator/libo3d3xx/modules/image/test/o3d3xx-image-tests.cpp:386: Failure
The difference between reference_extrinsics[1] + 3. and new_extrinsics[1] is 1.3609009981155396, which exceeds abs_error, where
reference_extrinsics[1] + 3. evaluates to -0.097953319549560547,
new_extrinsics[1] evaluates to 1.262947678565979, and
abs_error evaluates to 1.
[  FAILED  ] ImageBuffers_Tests.Extrinsics (3220 ms)
2.
[ RUN      ] ImageBuffers_Tests.ComputeCartesian
/home/administrator/libo3d3xx/modules/image/test/o3d3xx-image-tests.cpp:570: Failure
Value of: std::equal(x_cam.begin<std::int16_t>(), x_cam.end<std::int16_t>(), x_computed.begin<std::int16_t>(), cmp)
  Actual: false
Expected: true
/home/administrator/libo3d3xx/modules/image/test/o3d3xx-image-tests.cpp:576: Failure
Value of: std::equal(z_cam.begin<std::int16_t>(), z_cam.end<std::int16_t>(), z_computed.begin<std::int16_t>(), cmp)
  Actual: false
Expected: true
[  FAILED  ] ImageBuffers_Tests.ComputeCartesian (392 ms)

tpanzarella, 2016-03-17:
Can you paste in the complete output of:

```
$ src/bin/o3d3xx-dump --ip 10.10.10.71
```

zfxw0206, 2016-03-17:
{
    "o3d3xx":
    {
        "libo3d3xx": "400",
        "Date": "Thu Mar 17 16:21:42 2016",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02129009_008022954",
            "Illumination": "#!02_I300_001_02097990_008001175",
            "MiraSerial": "3610-5090-13e4-044c",
            "MACAddress": "00:02:01:40:27:61",
            "Frontend": "#!02_F341_002_15_00255_802259343",
            "Diagnose": "#!02_D322_C30_02101064_008023093",
            "Connector": "#!02_A300_001_02064651_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.3.1001",
            "Linux": "Linux version 3.11.0-00005-g2394d14 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Wed Nov 25 16:10:41 CET 2015",
            "Diagnostic_Controller": "0.3.87",
            "ELDK": "GOLDENEYE_ELDK-2015-11-26_16-11-02-5321dac5357219d1408dbf7f53f164e8c138302a",
            "Calibration_Version": "0.9.0",
            "Calibration_Device": "00:02:01:40:27:61",
            "Main_Application": "0.4.100",
            "Algorithm_Version": "1.2.6"
        },
        "Device":
        {
            "Name": "New sensor",
            "Description": "",
            "ActiveApplication": "1",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "15",
            "ServiceReportFailedBuffer": "15",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "EvaluationFinishedMinHoldTime": "10",
            "SaveRestoreStatsOnApplSwitch": "true",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
            "DeviceType": "",
            "ArticleNumber": "",
            "ArticleStatus": "",
            "UpTime": "1.37400920916504e-316",
            "ImageTimestampReference": "27810496",
            "TemperatureFront1": "6.952586354768227e-310",
            "TemperatureFront2": "1.374032924316041e-316",
            "TemperatureIMX6": "1.374044781891541e-316",
            "TemperatureIllu": "6.952552768050496e-310"
        },
        "Net":
        {
            "MACAddress": "",
            "NetworkSpeed": "0",
            "StaticIPv4Address": "10.10.10.71",
            "StaticIPv4SubNetMask": "255.255.255.0",
            "StaticIPv4Gateway": "10.10.10.0",
            "UseDHCP": "false"
        },
[... 32 lines cut ...]
                    "Resolution": "0",
                    "ClippingCuboid": "{\"XMin\": -3.402823e+38, \"XMax\": 3.402823e+38, \"YMin\": -3.402823e+38, \"YMax\": 3.402823e+38, \"ZMin\": -3.402823e+38, \"ZMax\": 3.402823e+38}",
                    "ReduceMotionArtifacts": "false",
                    "SpatialFilterType": "0",
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "0",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_moderate",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}

zfxw0206, 2016-03-17:
[camera.json.tar.gz](https://github.com/lovepark/libo3d3xx/files/178720/camera.json.tar.gz)

tpanzarella, 2016-03-17:
I will have to look into this later as I need to leave for the day (pick up kids). Nothing glaring the JSON. For now, you can skip these tests by setting `GTEST_FILTER=-ImageBuffers_Tests.Extrinsics:ImageBuffers_Tests.ComputeCartesian` in your environment before running the tests.

zfxw0206, 2016-03-17:
OK, thank you for your patient.

tpanzarella, 2016-03-17:
@graugans It does not appear that @zfxw0206 has an extrinsic calibration stored on the camera which would typically be the cause for these two tests to fail. That said, do you get these failures when running the tests? I do not.

tpanzarella, 2016-03-17:
@graugans Just thought of something.... @zfxw0206 recently updated his firmware. What he had on there before was pretty old, which means his H/W is probably relatively old too. I know there used to be a bug in the unit vectors that would cause off-board Cartesian computation to fail. Was that bug H/W related at all? Could there be some other legacy issues going on that we are getting bit by here?

graugans, 2016-03-17:
I'll check the details tomorrow. Regarding the question about IP-Settings. When you set an IP config through XML-RPC this is used in all configurations ``recovery`` and ``productive`` mode. Otherwise Update by the ifm GUI will not work for the average user.

graugans, 2016-03-18:
```
@graugans Just thought of something.... @zfxw0206 recently updated his firmware. What he had on there before was pretty old, which means his H/W is probably relatively old too. I know there used to be a bug in the unit vectors that would cause off-board Cartesian computation to fail. Was that bug H/W related at all? Could there be some other legacy issues going on that we are getting bit by here?
```
No, this bug is not related to the hardware. I'll try to find some time today and run those tests by myself.

zfxw0206, 2016-03-18:
Now I want to use ROS indigo to trigger my o3d303. I saw the ruby code in issue #39. Unfortunately, only ROS hydro support ruby, so I don't think I can use it in indigo. 

I saw you use t command to trigger the camera and p0 to stop. I only get "o3d3xx-hz" and "o3d3xx-schema" in module/framegrabber/src/bin and still have no idea how to send XML-RPC command and process interface command in Ubuntu ROS.  

Could you please tell me more details about how to start the libo3d3xx-framegrabber in listening mode and send the trigger command over another connection. Thank you.

graugans, 2016-03-18:
The Ruby code is just to give an idea how to trigger stuff. Maybe @cfreundl can provide his python class which provides similar features.

The ``p0`` is to turn off asynchronous messages. Otherwise you'll have to handle them. When I am back at a PC I'll search for the manuals on PCIC

zfxw0206, 2016-03-18:
Just now, I installed ruby in Ubuntu. Can I use that ruby code directly to trigger my o3d303? If so, how could I do it? Thank you!

zfxw0206, 2016-03-18:
Just now, I installed ruby in my Ubuntu, can I use that ruby code directly to trigger my o3d303? If so, what shall I do? Thank you?

graugans, 2016-03-21:
Your trigger_ifm.rb should look like this:

```
require "./pcic.rb"
pcic = PCIC.new()
pcic.connect('192.168.0.69',50010)
pcic.transfer('p0')
pcic.async_trigger()
```
This assumes you have activated the *PCIC* trigger for your application.

zfxw0206, 2016-03-21:
Yes, it works. Thank you very much!
But I still get 2 failures when I compile the image module. Could you please help me to solve it?
Thank you!

graugans, 2016-03-21:
I'll take a look into the compilation tomorrow when back to the office

tpanzarella, 2016-04-12:
@zfxw0206 The `ComputeCartesian` test has been updated in `0.4.1` to compare pixel-by-pixel accuracy to w/in 1 cm rather than 1 mm. This may have that test pass for you now. I should note, that I did not change the equivalent test in the ROS code (yet?).

s-hemer, 2016-04-13:
I can confirm the fail of the ImageBuffers_Tests.Extrinsics also with 0.4.1 and firmware 1.4.1639 on my O3D301 which originally came with the 1.2 or 1.3 firmware (as far as I remember I updated it once).

`[ RUN      ] ImageBuffers_Tests.Extrinsics
libo3d3xx/modules/image/test/o3d3xx-image-tests.cpp:385: Failure
The difference between reference_extrinsics[0] + 2. and new_extrinsics[0] is 1.5523750782012939, which exceeds abs_error, where
reference_extrinsics[0] + 2. evaluates to 1.2480340003967285,
new_extrinsics[0] evaluates to -0.30434107780456543, and
abs_error evaluates to 1.
libo3d3xx/modules/image/test/o3d3xx-image-tests.cpp:386: Failure
The difference between reference_extrinsics[1] + 3. and new_extrinsics[1] is 1.5755751132965088, which exceeds abs_error, where
reference_extrinsics[1] + 3. evaluates to -0.35268354415893555,
new_extrinsics[1] evaluates to 1.2228915691375732, and
abs_error evaluates to 1.
[  FAILED  ] ImageBuffers_Tests.Extrinsics (3090 ms)
`

tpanzarella, 2016-04-13:
OK, when I have the time, I will have to look into this in a bit more detail. For now, unless you plan to use the extrinsics and unit vectors to compute the cartesian data manually off board the camera, I think you can safely ignore these errors.

adilon, 2016-04-14:
I have the same problem when I compiling the Image Module: 

`[ RUN      ] ImageBuffers_Tests.Extrinsics
/home/adilon/Téléchargements/libo3d3xx-master/modules/image/test/o3d3xx-image-tests.cpp:385: Failure
The difference between reference_extrinsics[0] + 2. and new_extrinsics[0] is 1.4622671008110046, which exceeds abs_error, where
reference_extrinsics[0] + 2. evaluates to 2.4904198050498962,
new_extrinsics[0] evaluates to 1.0281527042388916, and
abs_error evaluates to 1.
/home/adilon/Téléchargements/libo3d3xx-master/modules/image/test/o3d3xx-image-tests.cpp:386: Failure
The difference between reference_extrinsics[1] + 3. and new_extrinsics[1] is 1.3986629247665405, which exceeds abs_error, where
reference_extrinsics[1] + 3. evaluates to 1.8712924718856812,
new_extrinsics[1] evaluates to 3.2699553966522217, and
abs_error evaluates to 1.
[  FAILED  ] ImageBuffers_Tests.Extrinsics (3099 ms)`

adilon, 2016-04-14:
I have the same problem when I compiling the Image Module: 

```
[ RUN      ] ImageBuffers_Tests.Extrinsics
/home/adilon/Téléchargements/libo3d3xx-master/modules/image/test/o3d3xx-image-tests.cpp:385: Failure
The difference between reference_extrinsics[0] + 2. and new_extrinsics[0] is 1.4622671008110046, which exceeds abs_error, where
reference_extrinsics[0] + 2. evaluates to 2.4904198050498962,
new_extrinsics[0] evaluates to 1.0281527042388916, and
abs_error evaluates to 1.
/home/adilon/Téléchargements/libo3d3xx-master/modules/image/test/o3d3xx-image-tests.cpp:386: Failure
The difference between reference_extrinsics[1] + 3. and new_extrinsics[1] is 1.3986629247665405, which exceeds abs_error, where
reference_extrinsics[1] + 3. evaluates to 1.8712924718856812,
new_extrinsics[1] evaluates to 3.2699553966522217, and
abs_error evaluates to 1.
[  FAILED  ] ImageBuffers_Tests.Extrinsics (3099 ms)
```

tpanzarella, 2016-04-14:
OK. This needs to move to a new issue as it is unrelated to software triggering. I will open the issue.

tpanzarella, 2016-05-27:
Version 0.4.4, the current release as of this writing, now has support for software triggering. An example of using this w/in `libo3d3xx` can be found in the `FrameGrabberTest.SoftwareTrigger` unit test (part of the framegrabber module). This has also been exposed in `o3d3xx-ros` via the `/o3d3xx/camera/Trigger` service.

closed 2016-05-27 by tpanzarella