ifm/libo3d3xx #49: camera make-check fails : Problem with XML-RPC requests ?

ankalou, 2016-04-13:
I have downloaded the library source code, and am trying to get the camera module to work. C-make and make run fine, but when make-check gives the following errors :
`$ make check
[  5%] Built target gtest
[ 60%] Built target o3d3xx_camera
[ 60%] Built target o3d3xx_camera_shared
[ 65%] Built target gtest_main
[100%] Built target o3d3xx-camera-tests
[==========] Running 43 tests from 6 test cases.
[----------] Global test environment set-up.
[----------] 17 tests from Camera_Tests
[ RUN      ] Camera_Tests.Ctor
[       OK ] Camera_Tests.Ctor (0 ms)
[ RUN      ] Camera_Tests.GetXMLRPCURLPrefix
[       OK ] Camera_Tests.GetXMLRPCURLPrefix (0 ms)
[ RUN      ] Camera_Tests.GetAllParameters
[       OK ] Camera_Tests.GetAllParameters (380 ms)
[ RUN      ] Camera_Tests.GetParameter
[       OK ] Camera_Tests.GetParameter (1326 ms)
[ RUN      ] Camera_Tests.GetSWVersion
[       OK ] Camera_Tests.GetSWVersion (36 ms)
[ RUN      ] Camera_Tests.GetApplicationList
[       OK ] Camera_Tests.GetApplicationList (25 ms)
[ RUN      ] Camera_Tests.RequestSession
[       OK ] Camera_Tests.RequestSession (60 ms)
[ RUN      ] Camera_Tests.CancelSession
[       OK ] Camera_Tests.CancelSession (74 ms)
[ RUN      ] Camera_Tests.Heartbeat
[       OK ] Camera_Tests.Heartbeat (130 ms)
[ RUN      ] Camera_Tests.SetOperatingMode
[       OK ] Camera_Tests.SetOperatingMode (138 ms)
**[ RUN      ] Camera_Tests.GetDeviceConfig
/home/localuser/ros_ifm/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:182: Failure
Value of: 32
Expected: params.size()
Which is: 30
unknown file: Failure
C++ exception with description "_Map_base::at" thrown in the test body.
[  FAILED  ] Camera_Tests.GetDeviceConfig (703 ms)**
[ RUN      ] Camera_Tests.ActivateDisablePassword
[       OK ] Camera_Tests.ActivateDisablePassword (667 ms)
[ RUN      ] Camera_Tests.SetDeviceConfig
[       OK ] Camera_Tests.SetDeviceConfig (2738 ms)
[ RUN      ] Camera_Tests.DeviceConfig_JSON
[       OK ] Camera_Tests.DeviceConfig_JSON (104 ms)
[ RUN      ] Camera_Tests.GetNetParameters
[       OK ] Camera_Tests.GetNetParameters (577 ms)
[ RUN      ] Camera_Tests.NetConfig
[       OK ] Camera_Tests.NetConfig (577 ms)
[ RUN      ] Camera_Tests.NetConfig_JSON
[       OK ] Camera_Tests.NetConfig_JSON (583 ms)
[----------] 17 tests from Camera_Tests (8121 ms total)

[----------] 4 tests from SpatialFilterTest
[ RUN      ] SpatialFilterTest.SpatialFilterConfig_General
[       OK ] SpatialFilterTest.SpatialFilterConfig_General (702 ms)
[ RUN      ] SpatialFilterTest.GetSpatialFilterParameters
[       OK ] SpatialFilterTest.GetSpatialFilterParameters (2160 ms)
[ RUN      ] SpatialFilterTest.GetSpatialFilterParameterLimits
[       OK ] SpatialFilterTest.GetSpatialFilterParameterLimits (1966 ms)
[ RUN      ] SpatialFilterTest.SpatialFilterConfig_JSON
[       OK ] SpatialFilterTest.SpatialFilterConfig_JSON (1837 ms)
[----------] 4 tests from SpatialFilterTest (6665 ms total)

[----------] 2 tests from General_Tests
[ RUN      ] General_Tests.Exceptions
[       OK ] General_Tests.Exceptions (0 ms)
[ RUN      ] General_Tests.Version
[       OK ] General_Tests.Version (0 ms)
[----------] 2 tests from General_Tests (0 ms total)

[----------] 1 test from ImportExport_Tests
[ RUN      ] ImportExport_Tests.ImportExportApp
[       OK ] ImportExport_Tests.ImportExportApp (1357 ms)
[----------] 1 test from ImportExport_Tests (1357 ms total)

[----------] 4 tests from TemporalFilterTest
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_General
[       OK ] TemporalFilterTest.TemporalFilterConfig_General (695 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameters
[       OK ] TemporalFilterTest.GetTemporalFilterParameters (1977 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameterLimits
[       OK ] TemporalFilterTest.GetTemporalFilterParameterLimits (2032 ms)
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_JSON
[       OK ] TemporalFilterTest.TemporalFilterConfig_JSON (1950 ms)
[----------] 4 tests from TemporalFilterTest (6654 ms total)

[----------] 15 tests from AppImagerTest
[ RUN      ] AppImagerTest.CopyDeleteApplication
[       OK ] AppImagerTest.CopyDeleteApplication (1323 ms)
[ RUN      ] AppImagerTest.CreateApplication
[       OK ] AppImagerTest.CreateApplication (936 ms)
[ RUN      ] AppImagerTest.ChangeAppNameAndDescription
[       OK ] AppImagerTest.ChangeAppNameAndDescription (986 ms)
[ RUN      ] AppImagerTest.EditApplication
[       OK ] AppImagerTest.EditApplication (949 ms)
[ RUN      ] AppImagerTest.GetAppParameters
[       OK ] AppImagerTest.GetAppParameters (1520 ms)
[ RUN      ] AppImagerTest.AppConfig
[       OK ] AppImagerTest.AppConfig (1777 ms)
[ RUN      ] AppImagerTest.AppConfig_JSON
[       OK ] AppImagerTest.AppConfig_JSON (1713 ms)
[ RUN      ] AppImagerTest.GetAvailableImagerTypes
[       OK ] AppImagerTest.GetAvailableImagerTypes (1512 ms)
[ RUN      ] AppImagerTest.ChangeImagerType
[       OK ] AppImagerTest.ChangeImagerType (2811 ms)
**[ RUN      ] AppImagerTest.GetImagerParameters
/home/localuser/ros_ifm/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:265: Failure
Expected: params.at("EnableFastFrequency") doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.GetImagerParameters (1928 ms)**
[ RUN      ] AppImagerTest.GetImagerParameterLimits
[       OK ] AppImagerTest.GetImagerParameterLimits (2830 ms)
**[ RUN      ] AppImagerTest.ImagerConfig
/home/localuser/ros_ifm/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:419: Failure
Expected: cam_->SetImagerConfig(im.get()) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.ImagerConfig (1911 ms)**
[ RUN      ] AppImagerTest.ImagerConfigValueOutOfRange
[       OK ] AppImagerTest.ImagerConfigValueOutOfRange (3448 ms)
[ RUN      ] AppImagerTest.ImagerConfig_JSON
[       OK ] AppImagerTest.ImagerConfig_JSON (1622 ms)
[ RUN      ] AppImagerTest.Exposure
[       OK ] AppImagerTest.Exposure (3885 ms)
[----------] 15 tests from AppImagerTest (29153 ms total)

[----------] Global test environment tear-down
[==========] 43 tests from 6 test cases ran. (51950 ms total)
[  PASSED  ] 40 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] Camera_Tests.GetDeviceConfig
[  FAILED  ] AppImagerTest.GetImagerParameters
[  FAILED  ] AppImagerTest.ImagerConfig

 3 FAILED TESTS
make[3]: *** [test/CMakeFiles/check] Erreur 1
make[2]: *** [test/CMakeFiles/check.dir/all] Erreur 2
make[1]: *** [test/CMakeFiles/check.dir/rule] Erreur 2
make: *** [check] Erreur 2
`
After digging into the code, I guess that the following function calls (in camera.cpp) fail :
cam_-> GetImagerParameters()
cam_-> SetImagerParameters()
But then it goes down to XML-RPC, that I know nothing about.

Any ideas on what could have gone wrong ? I haven't even edited anything yet...

My setup : Ubuntu 14.04, camera O3D303. The camera runs fine with ifmVisionAssistantV1.2.7 (with wine).

tpanzarella, 2016-04-13:
This is almost certainly due to a mismatch in a supported version of `libo3d3xx` and the on-camera firmware. Please see the main [README](https://github.com/lovepark/libo3d3xx/blob/master/README.md) and reference the software compatibility matrix. As of today, I'd suggest using the latest version of libo3d3xx (`0.4.1`) and firmware version `1.4.1639`. 

To update your camera, can use `o3d3xx-reboot -r`. Then hit the camera's internal http server (by default: http://192.168.0.69:8080. From there you will be greeting with a web interface where you can upload the firmware file. You can find `1.4.1639` in the firmware directory of this source distribution.

ankalou, 2016-04-13:
First, thanks a lot for your fast reply !
I downloaded the library with `git clone https://github.com/lovepark/libo3d3xx.git`, so it should be the last version.
`3d3xx-dump` gives `"IFM_Software": "1.1.288"`, so that may be the issue indeed, I'll try updating the firmware.

ankalou, 2016-04-13:
I had to git clone again (I guess my version was outdated already) but now it worked !
Thanks !

closed 2016-04-13 by ankalou

tpanzarella, 2016-04-13:
Stable versions are available [here](https://github.com/lovepark/libo3d3xx/releases). Running from the HEAD will always be considered _living on the edge_. At some point, maybe once we hit `1.0`, we can start releasing binaries.

ankalou, 2016-04-13:
Hmm.. Ok, I followed [these](https://github.com/lovepark/o3d3xx-ros) instructions but you're right, it's safer to keep a stable version.
It would be nice to have test binaries equivalent to ifmVision Assistant...

tpanzarella, 2016-04-13:
Since the releases are just annotated tags, and you have already cloned the repo, you have the stable tag on your machine already. You can `checkout` the particular tag of interest and build from there. I guess I need to update the docs for `o3d3xx-ros` to reflect this. Thanks for pointing it out.