ifm/libo3d3xx #50: ImageBuffers_Tests.Extrinsics failing

tpanzarella, 2016-04-14:
The `ImageBuffers_Tests.Extrinsics` unit test has been reported by numerous users, as of late, to fail. I have been able to verify this on a newer camera that I have. However, the test continues to pass on older cameras running the same version of the firmware. When I have the time, I will take this on and find the root of the issue and commit a fix. In the meantime, there is a workaround to get the tests to pass. Simply set the following environment variable prior to building the code:

```
$ export GTEST_FILTER=-ImageBuffers_Tests.Extrinsics
```

This will have `gtest` skip that particular test. This workaround should be harmless unless you fall into one of two categories:

1. You plan to store an extrinsic calibration on the camera (in this case you have bigger problems)
2. You plan to compute the Cartesian data manually off-board the camera

tpanzarella, 2016-04-15:
This was a legitimate bug in the unit test `ImageBuffers_Tests.Extrinsics`. When I had originally written the test code, I misunderstood the relationship between the XML-RPC values for the extrinsics and how they relate to the data that are returned over PCIC. I took the time to sort it out tonight and I believe I have a properly implemented unit test now. See 6b73a1dc674d266c7fb6f9bbb429e45a62832715. Perhaps @graugans can validate on the IFM side.

I have pushed out a new patch version of the code: `0.4.2`. The above "workaround" is valid for versions prior to `0.4.2` as nothing has changed in the actual runtime code, just the unit test has been updated to properly validate that the runtime code is parsing out the serialized extrinsics properly now.

Sorry for any confusion. Tagging interested parties: @adilon @s-hemer @zfxw0206

adilon, 2016-04-15:
Thanks,
No problem with libo3d3xx-0.4.2 and camera O3D303

closed 2016-04-16 by tpanzarella

s-hemer, 2016-04-19:
I can also report that `0.4.2` fixed the issue on compiling for my O3D301. Thanks