ifm/libo3d3xx #51: make check (AppImagerTest.Get*Parameters) fails on O3D303 Camera

NamanCMU, 2016-04-20:
Hi all,

I am trying to use libo3d3xx with our IFM O3D303 Camera and eventually I want to use o3d3xx-ros to extract the Point Cloud from the camera and use it for navigation of our Robot. I am able to ping the camera but when I run **make** inside the build directory, I get the following output with 2 Failed tests (AppImagerTest.GetAppParameters and AppImagerTest.GetImagerParameters Tests failed) : 

namanpc@namanpc-All-Series:~/libo3d3xx/build$ make
Scanning dependencies of target O3D3XX_CAMERA
[  3%] Creating directories for 'O3D3XX_CAMERA'
[  6%] No download step for 'O3D3XX_CAMERA'
[ 10%] No patch step for 'O3D3XX_CAMERA'
[ 13%] No update step for 'O3D3XX_CAMERA'
[ 16%] Performing configure step for 'O3D3XX_CAMERA'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   program_options
--   system
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/namanpc/libo3d3xx/build/O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-build
[ 20%] Performing build step for 'O3D3XX_CAMERA'
Scanning dependencies of target o3d3xx_camera
[  3%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/net_config.cpp.o
[  6%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/device_config.cpp.o
[ 10%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/spatial_filter_config.cpp.o
[ 13%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/err.cpp.o
[ 16%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/camera.cpp.o
[ 20%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/app_config.cpp.o
[ 23%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/cmdline_opts.cpp.o
[ 26%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/util.cpp.o
[ 30%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/version.cpp.o
[ 33%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/temporal_filter_config.cpp.o
[ 36%] Building CXX object src/libo3d3xx_camera/CMakeFiles/o3d3xx_camera.dir/imager_config.cpp.o
[ 36%] Built target o3d3xx_camera
Scanning dependencies of target o3d3xx_camera_shared
Linking CXX shared library libo3d3xx_camera.so
[ 36%] Built target o3d3xx_camera_shared
Scanning dependencies of target o3d3xx_camera_static
Linking CXX static library libo3d3xx_camera_static.a
[ 36%] Built target o3d3xx_camera_static
Scanning dependencies of target o3d3xx-config
[ 40%] Building CXX object src/bin/CMakeFiles/o3d3xx-config.dir/o3d3xx-config.cpp.o
Linking CXX executable o3d3xx-config
[ 40%] Built target o3d3xx-config
Scanning dependencies of target o3d3xx-dump
[ 43%] Building CXX object src/bin/CMakeFiles/o3d3xx-dump.dir/o3d3xx-dump.cpp.o
Linking CXX executable o3d3xx-dump
[ 43%] Built target o3d3xx-dump
Scanning dependencies of target o3d3xx-ifm-export
[ 46%] Building CXX object src/bin/CMakeFiles/o3d3xx-ifm-export.dir/o3d3xx-ifm-export.cpp.o
Linking CXX executable o3d3xx-ifm-export
[ 46%] Built target o3d3xx-ifm-export
[... 291 lines cut ...]
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_moderate",
                    "MaxAllowedLEDFrameRate": "0",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    }
                }
            }
        ]
    }
}

I have already tried resetting the o3d3xx camera by running the command **o3d3xx-reset** but I still got the same output.
Also, libo3d3xx: version=0.4.2 and "IFM_Software": "1.3.1001".

Any help will be appreciated.
Thanks!

tpanzarella, 2016-04-20:
You should update your camera to firmware version `1.4.1639`. On the main [README](https://github.com/lovepark/libo3d3xx/blob/master/README.md) we maintain a software compatibility matrix that maps compatible versions of `libo3d3xx` to the associated IFM firmware. The firmware is available in the `firmware` directory of this source distribution. The easiest way to do the update is to run `o3d3xx-reboot -r` from your build directory (since you don't have it installed yet). Then hit the sensor's internall http server at: `http://192.168.0.69:8080`. Here you can upload the firmware CPIO archive.

NamanCMU, 2016-04-21:
Thanks tpanzarella! It worked :)

closed 2016-04-21 by NamanCMU