ifm/libo3d3xx #52: Install with Ubuntu 16.04

adilon, 2016-04-21:
Hi,
I have dependency problems with ubuntu 16.04.

```
Run CPack packaging tool...
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: O3D3XX_CAMERA
CPack: - Install project: O3D3XX_CAMERA
CPack: Create package
CPack: - package: /home/adilon/Bureau/libo3d3xx-master/build/O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-build/libo3d3xx-camera_0.4.2_amd64.deb generated.
[ 10%] Performing install step for 'O3D3XX_CAMERA'
Sélection du paquet libo3d3xx-camera précédemment désélectionné.
(Lecture de la base de données... 254336 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de libo3d3xx-camera_0.4.2_amd64.deb ...
Dépaquetage de libo3d3xx-camera (0.4.2) ...
dpkg: des problèmes de dépendances empêchent la configuration de libo3d3xx-camera :
 libo3d3xx-camera dépend de libgoogle-glog0 ; cependant :
  Le paquet libgoogle-glog0 n'est pas installé.
 libo3d3xx-camera dépend de libboost-system1.54.0 ; cependant :
  Le paquet libboost-system1.54.0 n'est pas installé.
 libo3d3xx-camera dépend de libboost-program-options1.54.0 ; cependant :
  Le paquet libboost-program-options1.54.0 n'est pas installé.
 libo3d3xx-camera dépend de libxmlrpc-c++8 ; cependant :
  Le paquet libxmlrpc-c++8 n'est pas installé.

dpkg: erreur de traitement du paquet libo3d3xx-camera (--install) :
 problèmes de dépendances - laissé non configuré
Des erreurs ont été rencontrées pendant l'exécution :
 libo3d3xx-camera
CMakeFiles/O3D3XX_CAMERA.dir/build.make:77 : la recette pour la cible « O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-stamp/O3D3XX_CAMERA-install » a échouée
make[2]: *** [O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-stamp/O3D3XX_CAMERA-install] Erreur 1
CMakeFiles/Makefile2:67 : la recette pour la cible « CMakeFiles/O3D3XX_CAMERA.dir/all » a échouée
make[1]: *** [CMakeFiles/O3D3XX_CAMERA.dir/all] Erreur 2
Makefile:83 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
```
I have libgoogle-glog0v5, libxmlrpc-c++8v5, libboost-system1.58.0 and libboost-program-options1.58.0

```
sudo apt-get install libgoogle-glog0
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Aucune version du paquet libgoogle-glog0 n'est disponible, mais il existe dans la base
de données. Cela signifie en général que le paquet est manquant, qu'il est devenu obsolète
ou qu'il n'est disponible que sur une autre source
Cependant les paquets suivants le remplacent :
  libgoogle-glog0v5:i386 libgoogle-glog0v5

E: Le paquet « libgoogle-glog0 » n'a pas de version susceptible d'être installée
```

```
sudo apt-get install libxmlrpc-c++8
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Aucune version du paquet libxmlrpc-c++8 n'est disponible, mais il existe dans la base
de données. Cela signifie en général que le paquet est manquant, qu'il est devenu obsolète
ou qu'il n'est disponible que sur une autre source
Cependant les paquets suivants le remplacent :
  libxmlrpc-c++8v5:i386 libxmlrpc-c++8v5

E: Le paquet « libxmlrpc-c++8 » n'a pas de version susceptible d'être installée
```

Thanks for your help.

tpanzarella, 2016-04-21:
I haven't tested with that version of Ubuntu yet. However, you should be able to edit the generated Debian control file [here](https://github.com/lovepark/libo3d3xx/blob/master/modules/camera/CMakeLists.txt#L167) to get this install to work for you. Again, untested/unsupported but seems like a low-risk thing for you to try. I currently do not have the time to deal with this myself right now. I'm on 14.04 LTS.

adilon, 2016-04-21:
Thanks, i modified all Debian control file.
I compiled module Camera, Framegrabber OK.
But i have probleme with module Image:

```
~/Bureau/libo3d3xx-master/modules/image/build$ make
Scanning dependencies of target o3d3xx_image
[  6%] Building CXX object src/libo3d3xx_image/CMakeFiles/o3d3xx_image.dir/image.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[  6%] Built target o3d3xx_image
Scanning dependencies of target o3d3xx_image_static
[ 13%] Linking CXX static library libo3d3xx_image_static.a
[ 13%] Built target o3d3xx_image_static
Scanning dependencies of target o3d3xx_image_shared
[ 20%] Linking CXX shared library libo3d3xx_image.so
[ 20%] Built target o3d3xx_image_shared
Scanning dependencies of target o3d3xx-viewer_automoc
[ 26%] Automatic moc for target o3d3xx-viewer
[ 26%] Built target o3d3xx-viewer_automoc
Scanning dependencies of target o3d3xx-viewer
[ 33%] Building CXX object src/bin/CMakeFiles/o3d3xx-viewer.dir/o3d3xx-viewer.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
[ 40%] Building CXX object src/bin/CMakeFiles/o3d3xx-viewer.dir/o3d3xx-viewer_automoc.cpp.o
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
make[2]: ***  Aucune règle pour fabriquer la cible « /usr/lib/x86_64-linux-gnu/libproj.so », nécessaire pour « src/bin/o3d3xx-viewer ». Arrêt.
CMakeFiles/Makefile2:241 : la recette pour la cible « src/bin/CMakeFiles/o3d3xx-viewer.dir/all » a échouée
make[1]: *** [src/bin/CMakeFiles/o3d3xx-viewer.dir/all] Erreur 2
Makefile:149 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
```

adilon, 2016-04-21:
Make OK,

1) Install libproj
2) make 
3) delete all -lvtkproj4 (modules/image/build/src/bin/CmakeFiles/o3d3xx-viewview.dir/link.tkt)
4)make
5) make check
6) make package
7) sudo dpkg -i libo3d3xx-image_0.4.2_amd64.deb

If you have dependency problems in step 7, delete all files and start again step 2 and edit Debian control file.

o3d3xx-viewer OK

tpanzarella, 2016-05-27:
16.04 support is official as of `0.4.3`. Be sure to also read #56

closed 2016-05-27 by tpanzarella