ifm/libo3d3xx #54: unit tests fail on O3D303 camera, then can't install o3d3xx-ros package

HeloiseFFMH, 2016-05-17:
Hi, 

I try to install libo3d3xx and then install and use o3d3xx-ros, but the make fails when running the unit tests. I guess it is because I can't ping the O3D303 camera? How can I find or change the camera IP? I have a computer with windows too and tried to change the IP from the IFM software but it doesn't work. I ran nmap but I the camera does'nt appear in the network map.

Here are my outputs of $ifconfig and $make (after the previous steps working well):

> 
> $ifconfig
> lo        Link encap:Local Loopback  
>           inet addr:127.0.0.1  Mask:255.0.0.0
>           inet6 addr: ::1/128 Scope:Host
>           UP LOOPBACK RUNNING  MTU:65536  Metric:1
>           RX packets:2820 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:2820 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:0 
>           RX bytes:318041 (318.0 KB)  TX bytes:318041 (318.0 KB)
> 
> wlan0     Link encap:Ethernet  HWaddr 60:67:20:66:8a:f8  
>           inet addr:132.68.135.92  Bcast:132.68.135.255  Mask:255.255.254.0
>           inet6 addr: fe80::6267:20ff:fe66:8af8/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:338312 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:66211 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:372039431 (372.0 MB)  TX bytes:21859527 (21.8 MB)

> $ make
> [  3%] Performing test step for 'O3D3XX_CAMERA'
> [  5%] Built target gtest
> [ 10%] Built target gtest_main
> [ 65%] Built target o3d3xx_camera
> [ 65%] Built target o3d3xx_camera_shared
> [100%] Built target o3d3xx-camera-tests
> [==========] Running 43 tests from 6 test cases.
> [----------] Global test environment set-up.
> [----------] 17 tests from Camera_Tests
> [ RUN      ] Camera_Tests.Ctor
> [       OK ] Camera_Tests.Ctor (1 ms)
> [ RUN      ] Camera_Tests.GetXMLRPCURLPrefix
> [       OK ] Camera_Tests.GetXMLRPCURLPrefix (0 ms)
> [ RUN      ] Camera_Tests.GetAllParameters
> /home/heloise/ffmh_ws/src/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:61: Failure
> Expected: all_params = cam->GetAllParameters() doesn't throw an exception.
>   Actual: it throws.
> [  FAILED  ] Camera_Tests.GetAllParameters (3004 ms)
> [ RUN      ] Camera_Tests.GetParameter
> /home/heloise/ffmh_ws/src/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:70: Failure
> Expected: all_params = cam->GetAllParameters() doesn't throw an exception.
>   Actual: it throws.
> [  FAILED  ] Camera_Tests.GetParameter (6006 ms)
> [ RUN      ] Camera_Tests.GetSWVersion
> /home/heloise/ffmh_ws/src/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:90: Failure
> Expected: sw_version = cam->GetSWVersion() doesn't throw an exception.
>   Actual: it throws.
> [  FAILED  ] Camera_Tests.GetSWVersion (3002 ms)
> [ RUN      ] Camera_Tests.GetApplicationList
> /home/heloise/ffmh_ws/src/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:99: Failure
> Expected: apps = cam->GetApplicationList() doesn't throw an exception.
>   Actual: it throws.
> [  FAILED  ] Camera_Tests.GetApplicationList (3003 ms)
> [ RUN      ] Camera_Tests.RequestSession
> /home/heloise/ffmh_ws/src/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:107: Failure
> Expected: cam->RequestSession() doesn't throw an exception.
>   Actual: it throws.
> /home/heloise/ffmh_ws/src/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:108: Failure
> Value of: 32
> Expected: cam->GetSessionID().size()
> Which is: 0
> [  FAILED  ] Camera_Tests.RequestSession (3003 ms)
> [ RUN      ] Camera_Tests.CancelSession
[... 198 lines cut ...]
> [  FAILED  ] AppImagerTest.GetAppParameters
> [  FAILED  ] AppImagerTest.AppConfig
> [  FAILED  ] AppImagerTest.AppConfig_JSON
> [  FAILED  ] AppImagerTest.GetAvailableImagerTypes
> [  FAILED  ] AppImagerTest.ChangeImagerType
> [  FAILED  ] AppImagerTest.GetImagerParameters
> [  FAILED  ] AppImagerTest.GetImagerParameterLimits
> [  FAILED  ] AppImagerTest.ImagerConfig
> [  FAILED  ] AppImagerTest.ImagerConfigValueOutOfRange
> [  FAILED  ] AppImagerTest.ImagerConfig_JSON
> [  FAILED  ] AppImagerTest.Exposure
> [  FAILED  ] ImportExport_Tests.ImportExportApp
> [  FAILED  ] TemporalFilterTest.TemporalFilterConfig_General
> [  FAILED  ] TemporalFilterTest.GetTemporalFilterParameters
> [  FAILED  ] TemporalFilterTest.GetTemporalFilterParameterLimits
> [  FAILED  ] TemporalFilterTest.TemporalFilterConfig_JSON
> 
> 39 FAILED TESTS
> make[6]: *** [test/CMakeFiles/check] Error 1
> make[5]: *** [test/CMakeFiles/check.dir/all] Error 2
> make[4]: *** [test/CMakeFiles/check.dir/rule] Error 2
> make[3]: *** [check] Error 2
> make[2]: *** [O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-stamp/O3D3XX_CAMERA-test] Error 2
> make[1]: *** [CMakeFiles/O3D3XX_CAMERA.dir/all] Error 2
> make: *** [all] Error 2

hris2003, 2016-05-17:
The default IP of the camera is 192.168.0.69.
Then, for the rest, if you can not ping the camera, then it's normal that the test does not pass. In my case when I don't have a connected camera, I just skip the test and compile the driver packages whatsoever.

HeloiseFFMH, 2016-05-17:
Hi hris2003,

Thank you for your answer. With the windows software I get a different IP of the camera and can't define it in Linux. How do you compile without the tests?

hris2003, 2016-05-17:
If you have the camera's IP, you can try this:  
`$ O3D3XX_IP=<your_camera_IP> make`

Otherwise, to skip the test part, I followed this [tutorial](https://github.com/lovepark/libo3d3xx#full-instructions) and skipped the `make check` command.

HeloiseFFMH, 2016-05-17:
OK, I will try step by step,thanks!

HeloiseFFMH, 2016-05-18:
Hi,

I succeeded in installing the ros-driver module by module without the tests, thank you very much! But I still can't run it because I don't find the IP of the camera.
I tried to find it by arping, but it didn't work:
```
$sudo arping 00:02:01:40:30:FA
arping: unknown host 00:02:01:40:30:FA
```
 Any idea? Thank you veru much!

hris2003, 2016-05-18:
Hi,
You did say that ifmVisionAssistant on Window detects and show the camera IP. Can you try to ping that IP in linux?

HeloiseFFMH, 2016-05-18:
I tried but it seems that the IP changes when I disconnect the camera from windows to connect it in Ubuntu

hris2003, 2016-05-18:
That address is a zeroconf address. You should change it to ip available inside you network, then you'd be good to go.

HeloiseFFMH, 2016-05-18:
Yes, I would be glad to change it to an IP in my network, but I don't succeed in

hris2003, 2016-05-18:
Well, can you plug the camera's network cable directly to your computer and retry?

HeloiseFFMH, 2016-05-18:
Hi hris2003, thanks, I don't know any other way to connect to the camera that plug directly to my computer... -?- That is the way I am trying to connect to it, is there any other one?

graugans, 2016-05-18:
The camera has an link-local address and a fixed or dhcp address. When you have ``avahi`` running on your Ubuntu box you can try to ping the camera by its hostname. Followed by ``.local`` The hostname is constructed like this:

```

if test -f /etc/hostname
then
        BASENAME=`cat /etc/hostname`
        EXTENSION=`ifconfig eth0 | awk '/HWaddr/ {print $5}' | cut -d : -f4-6 | sed 's/:/-/g'`
        if [[ X"" != X"$EXTENSION" ]]
        then
            hostname $BASENAME-$EXTENSION
        else
            hostname -F /etc/hostname
        fi
fi
```
Basename is ``O3D3xx`` so the full hostname ist ``O3D3xx-02-04-AF`` when your cameras MAC is: ``00:02:01:02:04:AF``

graugans, 2016-05-18:
@HeloiseFFMH The issue with the temporary IP under  Windows maybe caused by yours Ethernet Adapters configuration. When your PC is configured to receive the IP by DHCP the Windows GUI will prevent you from changing the IP to another address range. But you can check the current ip under device settings.

HeloiseFFMH, 2016-05-24:
Hi @graugans, thanks for your answer. I tried to run avahi-daemon and then to ping the device by the IP and by the hostname but I don't succeed in.
Right, I can see the device IP under device settings in Windows, but then I can't ping it in Ubuntu.

hris2003, 2016-05-24:
@HeloiseFFMH : Hi, I just got the chance to configure my ifm camera recently. Here's how I do it:
Step 0: Define what IP you want to assign for your robot, let's say 192.168.10.10 (or whatever you want)
Step 1: Change my computer's IP to a fixed IP of the same range as of the IP I wanted to affect to the camera, for example: 192.168.10.100
Step 2: In ifmVisionAssistance, detect the camera, then go to device setup (or something like that) then change its IP to the designated IP (defined in step 0) then click apply
Step 3: Now you camera should be pingable with the new IP.

HTH.

HeloiseFFMH, 2016-05-24:
Hi @hris2003 , unfortunately it doesn't work for me, I can't even ping the camera IP. Moreover the camera doesn't appear while running $netstat -r with/without it.

hris2003, 2016-05-24:
But you said that ifmvisionassitant sees your camera. How so? Can you list the steps you did and what you got?

HeloiseFFMH, 2016-05-24:
Yes, but I run ifmvisionassistant from another computer running with windows

hris2003, 2016-05-24:
First of, you need to make sure that from ifmvisionassistant, the camera is detectable and connectable with its newly set ip. Is that the case?

HeloiseFFMH, 2016-05-24:
I get "Configuration of PC and device allows no direct TCP communication"

hris2003, 2016-05-24:
Yeah! That's correct! You need to change the IP of the ifmvisionassitant computer to the same IP range as that of the camera. if not, no access possible.

HeloiseFFMH, 2016-05-24:
From windows ? How do I change IP in windows?

hris2003, 2016-05-24:
Like this: http://www.howtogeek.com/howto/19249/how-to-assign-a-static-ip-address-in-xp-vista-or-windows-7/

HeloiseFFMH, 2016-05-24:
The IP of my windows computer is 192.168.0.7, and the IP I try to set is the original one: 192.168.0.69, is it not in the same range?

hris2003, 2016-05-24:
Is the camera's network cable plugged directly to your windows machine?

HeloiseFFMH, 2016-05-24:
Yes
I tried to set a fixed IP to the computer but I still can't set the IP of the camera

hris2003, 2016-05-24:
And if you change it to something else? Such as 192.168.0.100, or some other net range, like 192.168.1.xxx, and disable all other network connection of your ifmvisionassitant computer so that there's only the camera and the computer interconnected?

cfreundl, 2016-05-24:
I'll try to recap the problem situation so far:

- you can access and configure your camera from a Windows PC using the ifm Vision Assistant
- you are setting the camera's IP address to a static one: 192.168.0.69
- you cannot access the camera from another Linux PC

One problem I see immediately is that according to your initial issue description the IP address of the Linux PC is 132.68.135.92 (netmask 255.255.254.0), so a camera configured to 192.168.0.69 is outside of that network. Have you already tried to set the camera's IP address to an address within that subnet, e.g. 132.68.135.93?

HeloiseFFMH, 2016-05-25:
Hi, of course I tried different IPs in the good network. @cfreundl  I tried in different networks so it looked strange.
It seems that the camera has its own local network that doesn't fit with mine. Is it possible? How can I fix it? The camera IP is 169.254.48.250 or  sometimes 169.254.135.139 I tried to connect while disabling everything, but even if only the Ethernet connection is available I can't connect to it.
Thanks!

cfreundl, 2016-05-25:
The camera always provides a link-local address in the network 169.254.0.0/16, in addition to the configured IP address. This second address is meant as a fallback in cases like yours where no "regular" connection is possible. The camera tries to obtain the address using the last two octets of its MAC address, which in your case (MAC 00:02:01:40:30:FA) is 169.254.48.250. However, this address is not necessarily constant (after every reboot) as it involves a protocol to avoid address clashes which might change the camera's address within the link-local address range.

If you try to connect to the camera with the ifm Vision Assistant using "Find sensor" on the start screen, a broadcast is sent to the network interfaces to which all cameras respond - of course only those that the broadcast gets to: this should work if the camera is directly connected to the PC, with any network devices (hubs, switches etc.) in between this cannot be guaranteed. The camera reports its IP address in the answer to this broadcast, and it chooses the IP that fits best to the broadcast sender's (i.e. the PC's) network address. So if you have configured the camera to be in the same subnet as the PC, then the "regular" IP address should be reported. If they are not in the same subnet it can happen that it reports is link-local address, so that is why you might see only those IP addresses.

Let's try to resolve the situation a little more: from what I have understood so far, it happens also on your Windows PC that, even if you configure the camera for the same network that the PC is in, you still can connect to the camera only using the link-local address, right?

In that case, we can try the following: connect to the correctly configured camera with the Vision Assistant and try to get the following from the camera:

- a service report, which can be obtained from the monitor screen you see immediately after connecting to the camera, it's on the bottom next to "View options" and "Record options" -> "Save" that report
- the complete camera configuration: go to the "Device" configuration and there choose "General" -> "Settings" -> "Export"

If you can provide any of these two, I can take a look at it and hopefully get an idea what is going wrong.

HeloiseFFMH, 2016-05-25:
Hi, here are the files, thank you very much!

[deviceConfig_O3D300.zip](https://github.com/lovepark/libo3d3xx/files/281347/deviceConfig_O3D300.zip)
[O3D3xx_2016-05-25_113910_00-02-01-40-30-fa.zip](https://github.com/lovepark/libo3d3xx/files/281345/O3D3xx_2016-05-25_113910_00-02-01-40-30-fa.zip)

cfreundl, 2016-05-25:
I can see that the camera is running ok, and the network configuration is currently IP address 192.168.0.69 with netmask 255.255.224.0. That is hopefully what you have configured and I assume that the Windows PC is configured for the same network (right?)

Now with the camera still connected to the Windows PC:

- does the Vision Assistant report the camera with IP address 192.168.0.69 when doing a "Find sensor"?
- can you ping the camera at 192.168.0.69?
- can you open the URL http://192.168.0.69/api/rpc/v1/com.ifm.efector/ in a browser?

HeloiseFFMH, 2016-05-25:
Hi, from the windows PC, the Vision Assistant still reports the camera IP with 169.254.48.250 (and not 192.168.0.69).
I can't open the URL http://192.168.0.69/api/rpc/v1/com.ifm.efector/ in the browser:

> This page can't be displayed
> 
> •Make sure the web address http://192.168.0.69 is correct.
> •Look for the page with your search engine.
> •Refresh the page in a few minutes.
>

cfreundl, 2016-05-25:
So you cannot connect to camera and also the camera thinks that the PC's broadcast is coming from another network...
In that case, can you provide the Windows PC's address and netmask (maybe just by running `ipconfig` in a command window) for the sake of completeness in order to verify that both devices are in the same network?

HeloiseFFMH, 2016-05-25:
[ipconfig.txt](https://github.com/lovepark/libo3d3xx/files/281474/ipconfig.txt)
Here it is

cfreundl, 2016-05-25:
So the PC has an address 192.168.23.52, but for a WiFi network adapter, and the camera is directly connected via Ethernet, right? That would make sense as the adapter "Ethernet 2" (which is possibly set to use DHCP maybe?) has the link-local IP address 169.254.9.244, so the camera will always present its link-local address as well.

HeloiseFFMH, 2016-05-25:
Hi, it is what I get when fixing the IP address in Ethernet2, it still get an autoconfig IPv4. We can see the gateway that I set, but the IP and also the subnet mask are not the ones I set

> Ethernet adapter Ethernet 2:
> 
>    Connection-specific DNS Suffix  . : 
>    Link-local IPv6 Address . . . . . : fe80::94da:8c48:49cc:9f4%22
>    Autoconfiguration IPv4 Address. . : 169.254.9.244
>    Subnet Mask . . . . . . . . . . . : 255.255.0.0
>    Default Gateway . . . . . . . . . : 192.168.0.1

cfreundl, 2016-05-25:
Ok, I guess we can leave that as a Windows configuration issue for the network adapter. 

The more interesting part for you is probably to fix the camera connection issue with the Linux PC. Can you manage to configure the network adapter in Linux to use an address in the 192.168.0.0/19 network which is currently configured in the camera and directly connect the camera to the Linux PC? Then check again by both pinging and opening the URL mentioned above.

HeloiseFFMH, 2016-05-25:
Hi, I added

> auto eth0
> iface eth0 inet static
> 	address 192.168.0.60
> 	netmask 255.255.224.0
> 	gateway 192.168.0.1

in /ets/network/interfaces and I get: 

> $ ifconfig
> eth0      Link encap:Ethernet  HWaddr 00:21:cc:ce:db:44  
>           inet6 addr: fe80::221:ccff:fece:db44/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:12 errors:0 dropped:1 overruns:0 frame:0
>           TX packets:70 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:768 (768.0 B)  TX bytes:11849 (11.8 KB)
>           Interrupt:20 Memory:f2500000-f2520000 
> 
> lo        Link encap:Local Loopback  
>           inet addr:127.0.0.1  Mask:255.0.0.0
>           inet6 addr: ::1/128 Scope:Host
>           UP LOOPBACK RUNNING  MTU:65536  Metric:1
>           RX packets:949 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:949 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:0 
>           RX bytes:134598 (134.5 KB)  TX bytes:134598 (134.5 KB)
> 
> wlan0     Link encap:Ethernet  HWaddr 60:67:20:66:8a:f8  
>           inet addr:192.168.8.150  Bcast:192.168.31.255  Mask:255.255.224.0
>           inet6 addr: fe80::6267:20ff:fe66:8af8/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:1035796 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:2770 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:117286550 (117.2 MB)  TX bytes:464706 (464.7 KB)

but I can't ping the address 192.168.0.69 and can't open the link http://192.168.0.69/api/rpc/v1/com.ifm.efector/

cfreundl, 2016-05-25:
This looks quite similar, it seems that your configuration for eth0 does not have any effect, instead the adapter wlan0 is configured for that network. Can you try and use a network different from the one used in wlan0, let's say e.g. 172.16.0.0/16, and configure both the Linux PC and the camera for that?

HeloiseFFMH, 2016-06-06:
Hi,
I finally succeeded in conecting to the camera. I simply had to set the netmask to 255.255.255.0
Thanks for your help!

closed 2016-06-06 by HeloiseFFMH