ifm/libo3d3xx #55: Cannot connect to camera without DHCP server

cquennouelle, 2016-05-18:
Hello,

For more than one month, we've been successfully using the IFM camera (and the ROS node) with the camera and the PC connected to a network with a DHCP server. 
Now, we want to use the camera off the network, so we've connected it to the PC through an USB/Ethernet converter. For this network eth1, the PC has a static IP address, and we've given static IP address 192.168.1.51 to the camera (disabling the DCHP Server option) with the IFM Vision Assistant (running with wine).

In this configuration, we can visualize the point cloud from the camera in the IFM Vision Assistant. We can also successfully ping the camera at 192.168.1.51.
However, after having exported the correct value to variable O3D3XX_IP=192.168.1.51, neither the tests of the "make check", nor the o3d3xx_node manage to connect to the camera (the correct IP being displayed when launching "roslaunch o3d3xx camera.launch")

Can someone help us to find what we missed in the configuration ?
Thank you.

tpanzarella, 2016-05-18:
Can we see the kernel routing table (i.e., paste in the output from `netstat -r` from a Linux shell)?

graugans, 2016-05-19:
You can check the link-local address, like described in #54

cquennouelle, 2016-05-19:
Hi @tpanzarella, 
thank you for you answer. The output of `netstat -r` is

-when connected to both network (usual network and USB/Ethernet) :
```
$ netstat -r
Table de routage IP du noyau
Destination     Passerelle      Genmask         Indic   MSS Fenêtre irtt Iface
default         gw35.bas.       0.0.0.0         UG        0 0          0 eth0
192.168.1.0     *               255.255.255.0   U         0 0          0 eth1
192.168.35.0    *               255.255.255.0   U         0 0          0 eth0
192.168.38.0    *               255.255.255.0   U         0 0          0 vmnet8
192.168.127.0   *               255.255.255.0   U         0 0          0 vmnet1
```

-when connected only with the USB/Ethernet converter :
```
$ netstat -r
Table de routage IP du noyau
Destination     Passerelle      Genmask         Indic   MSS Fenêtre irtt Iface
192.168.1.0     *               255.255.255.0   U         0 0          0 eth1
192.168.38.0    *               255.255.255.0   U         0 0          0 vmnet8
192.168.127.0   *               255.255.255.0   U         0 0          0 vmnet1
```

As for the ROS Node, I have :
```
$ roslaunch o3d3xx camera.launch 
... logging to /home/localuser/.ros/log/ae9392ee-1d96-11e6-87e4-00133b99a4cd/roslaunch-BA14-995-4769.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://BA14-995:54355/

SUMMARY
========

PARAMETERS
 * /o3d3xx/camera/frame_id_base: o3d3xx/camera
 * /o3d3xx/camera/ip: 192.168.1.51
 * /o3d3xx/camera/password: 
 * /o3d3xx/camera/publish_viz_images: True
 * /o3d3xx/camera/schema_mask: 15
 * /o3d3xx/camera/timeout_millis: 500
 * /o3d3xx/camera/timeout_tolerance_secs: 5.0
 * /o3d3xx/camera/xmlrpc_port: 80
 * /rosdistro: indigo
 * /rosversion: 1.11.19

NODES
  /o3d3xx/
    camera (o3d3xx/o3d3xx_node)
    camera_tf (tf2_ros/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[o3d3xx/camera-1]: started with pid [4787]
process[o3d3xx/camera_tf-2]: started with pid [4788]
[ WARN] [1463644419.605300570]: Timeout waiting for camera!
[ WARN] [1463644420.105825119]: Timeout waiting for camera!

```

@graugans I've read #54, but I'm bit lost. When connected to both network, my pc has two 2 ip adress. How can I know which one is used by the camera to reply ?

cquennouelle, 2016-05-20:
@graugans I have a response either by doing `ping 192.168.1.51` or `ping O3D3xx-40-1A-78.local` (as you said in #54). 

```
$ ping O3D3xx-40-1A-78.local
PING O3D3xx-40-1A-78.local (192.168.1.51) 56(84) bytes of data.
64 bytes from 192.168.1.51: icmp_seq=1 ttl=64 time=0.232 ms
```

I even tried to `export O3D3XX_IP=O3D3xx-40-1A-78.local`, but the result remains the same : 
`[ WARN] [1463728501.794319025]: Timeout waiting for camera!`

graugans, 2016-05-20:
@cquennouelle I guess you already tried: ``export O3D3XX_IP=192.168.1.51``

graugans, 2016-05-20:
@cquennouelle can you please try to access this in your browser?

```
http://O3D3xx-40-1A-78.local/api/rpc/v1/com.ifm.efector/
```

cquennouelle, 2016-05-24:
@graugans sorry for the delay. Here is a capture of what is displayed in the browser at `http://O3D3xx-40-1A-78.local/api/rpc/v1/com.ifm.efector/` :

[image]

cquennouelle, 2016-05-27:
In our current network configuration, we have to set the following environment variables : `http_proxy`, `https_proxy` and `no_proxy` as follows: 
```
export http_proxy=http://proxy.bas.com:8080
export https_proxy=$http_proxy
export no_proxy="localhost,127.0.0.1,192.168.0.0/16,172.16.0.0/12,10.0.0.0/    8,.bas.com"
```
These variables seem to be problematic: Today, we've made a test with another DHCP server on a another network and we firstly met the same problem. By removing these variables, it finally worked. 

We haven't tried with the static ip configuration yet, but I guess this can help, am I right ?

graugans, 2016-05-27:
Looks, like the underlying http library from libxmlrpc tries to connect through your proxy. Have you tried to exclude the IP by adding it to the ``no_proxy`` variable?

cquennouelle, 2016-05-31:
Finally, it works ! The solution is to unset variable `http_proxy` in the current shell :
`$ unset http_proxy`
This variable seems to be set no matter the proxy configuration. This remains a bit obscur to me, but at least we can use the camera and the ROS node with a static IP.

Thank you for your help.

closed 2016-05-31 by cquennouelle