ifm/libo3d3xx #57: What's the effect of the rectifications flags?

bibarz, 2016-05-26:
I can't find much documentation about the EnableRectificationDistanceImage and EnableRectificationAmplitudeImage flags in the application options. I can see some differences in the distance readings when I set EnableRectificationDistanceImage to true, for example. What exactly do they do and why are they false by default? Thanks.

graugans, 2016-05-26:
The [rectification]( https://en.m.wikipedia.org/wiki/Image_rectification) is some kind of image correction. Due to the optics the distance data does not form a perfect rectangle. The image looks more like a pillow. In case of the camera the rectification is disabled by default. For the sensor the rectification is activated. 

It is deactivated to provide the unmodified point cloud   to the user

bibarz, 2016-05-26:
So if I understand correctly,
- rectification should be off to compute an accurate pointcloud from the depth
- rectification should be on if we want the depth (or amplitude) 2d image to look good on the screen (i.e., to be a correct flat projection)
Is that right?

tpanzarella, 2016-05-26:
The guys at IFM are obviously the authority on this, however, my understanding of the rectification parameters are that they apply the camera's intrinsic calibration (e.g., correction for lens distortion and other systemic errors) to the data prior to serializing and returning them over the wire.  I believe they are off by default on the camera because doing the image warps are not computationally cheap and will affect frame rate. Again, this is just my understanding, @graugans would know the details better.

bibarz, 2016-05-26:
Thanks @graugans and @tpanzarella for the feedback. I have checked that the unit vectors change when the rectification flag changes (or at least when both rectification flags change, I'm changing both together; I don't know what the behavior is supposed to be when you rectify only depth or only amplitude, should the unit vectors change in that case?). So it looks like rectification is meant to correct not just the depth for display but also the pointcloud. I have the camera mounted together with a lidar and I was looking at the correspondence between them; it seems (I'm merely eyeballing) that with rectification the correspondence is slightly better. I have also checked that the frame rate drops a bit (about 3%) when rectification is on.

It would be useful if you can confirm that rectification is intended to make the pointcloud more accurate, and if the reason for it being off by default is to increase the frame rate.

graugans, 2016-05-27:
Yes framerate will be lower because of the additional calculation. In the end we need a default value for this parameter the decision was to turn it off in case of the camera. I can not remember the arguments but can ask if it is important for you or @cfreundl can explain. 

The unit vectors will change when rectification is turned on. Because the optical model changes.

graugans, 2016-05-27:
@bibarz related to accuracy I'll ask the algo team on Monday. Those are the guys with deep knowledge on the 3D data. I am the firmware guy not the algorithm expert.

bibarz, 2016-05-27:
Thanks, @graugans, confirmation about the accuracy would be great, that's all I really want.

graugans, 2016-05-30:
So after a quick chat with the algorithm team the arguments why rectification is disabled by default are the following:

- Performance
- The field of view gets smaller with rectification turned on
- The image is based on interpolated pixels

Regarding the accuracy, it will not improve because of the interpolation step. 
You can turn on/off amplitude and distance image rectification separately to spare some CPU cycles if you do not need amplitude data.

bibarz, 2016-06-03:
Ok, thanks

closed 2016-06-03 by bibarz