ifm/libo3d3xx #58: Expected behavior with multiple PCIC connections and S/W Triggering

tpanzarella, 2016-05-31:
This is a technical question for @graugans and cfreundl as to what the expected behavior should be under a particular configuration.

Let's assume the active application on the camera is configured for software triggering. What should happen when more than one TCP connection is made to PCIC with potentially different return schemas configured for their respective sessions. For our purposes here, let's assume there are two PCIC clients connected. When a S/W trigger is sent on one socket, should both clients get the triggered image frame? Or, does each connected client need to trigger independently?

tpanzarella, 2016-06-01:
I have identified this situation as a bug in the `0.4.4` framegrabber code. Currently, if you are using software triggering, only a single TCP connection to a given camera's PCIC server is supported. I have reviewed the code and will issue a patch (tomorrow -- I hope) that will address this.

graugans, 2016-06-01:
One design goal of the ``PCIC`` was to have multiple independent configured connections. You can establish up to 5 connections in parallel. Each connection you defines its own JSON output  string. The PCIC pseudo code will look like this

```cpp
void onNewResultEvent(const ResultData& data) {
     for(connection &con : m_connections) {
        sendFormatedResult(con.getFormatSchema(),data)
     }
}
```
The ``onNewResultEvent`` is published by a trigger event. A trigger event can be one of the following ones:

- Continuous Trigger
- Hardware Trigger
- Software Trigger (PCIC)

Due to the fact the very first Trigger source was the hardware trigger with a ``ready`for trigger`` line we do not queue triggers. In the hardware trigger configuration an event called ``trigger overrun`` will be emitted when you trigger the device while the ```ready`for trigger`` line is in low state. In case of the PCIC trigger you'll receive a ``!``.

In case of PCIC Trigger we distinguish between to types of Trigger

- synchronous trigger (``T?``)
- asynchronous trigger (``t``)

In case of the synchronous Trigger the PCIC response to the trigger command ``T?`` is the ``data`` itself or ``!``. But what happens to a second or a third PCIC connection in this case? Those connections will receive the data as asynchronous events. This is ticket number ``0000`` in case of protocol v3.

When you use the asynchronous trigger commad ``t`` all connections will receive asynchronous data.

tpanzarella, 2016-06-01:
Thanks for the feedback @graugans 

I'm currently testing a patch now on the `dev-0.4.5` branch. This allows for multiple PCIC sockets connected while operating under s/w triggering. It also follows that, per your email above, our PCIC trigger command is of the async variety, i.e., `t`.

tpanzarella, 2016-06-02:
Version `0.4.5` is now available as a *Release*. Async s/w triggering supporting multiple PCIC clients has been tested and is now supported.

closed 2016-06-02 by tpanzarella

graugans, 2016-06-03:
I've just updated my [meta-ros](https://github.com/graugans/meta-ros/commit/90cf13b819344b4f3741918b6fd6a17f07317721) layer with those changes