ifm/libo3d3xx #59: Error while installing libo3d3xx framegrabber

REBAUD, 2016-06-09:
Hello,

i follow the instruction to install libo3d3xx on my linux 14.04 and i do it without problems for camera but when i try to do it for frammegrabber i have the following issue:

-- UBUNTU_VERSION: 14.04.4
-- UBUNTU_MAJOR: 14
-- UBUNTU_MINOR: 04
-- UBUNTU_PATCH: 4
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   program_options
--   system
CMake Error at /home/vrebaud/devel/libo3d3xx/cmake/modules/Findo3d3xx_camera.cmake:22 (include):
  include could not find load file:

    /o3d3xx_camera/o3d3xx_camera-config-version.cmake
Call Stack (most recent call first):
  CMakeLists.txt:69 (find_package)

-- Could NOT find O3D3XX_CAMERA (missing:  O3D3XX_CAMERA_LIBRARIES O3D3XX_CAMERA_INCLUDE_DIRS) 
CMake Error at /home/vrebaud/devel/libo3d3xx/cmake/modules/Findo3d3xx_camera.cmake:54 (message):
  o3d3xx_camera: 0.4.5 !=
Call Stack (most recent call first):
  CMakeLists.txt:69 (find_package)

-- Configuring incomplete, errors occurred!

it didn't happen during my last installation of the 0.4.0 version.
Do you have any idea ??

thanks

tpanzarella, 2016-06-09:
Where did you install `o3d3xx-camera`? It is not in a standard location. The `Find` file will look for it in: `/opt/libo3d3xx` or `/usr`.

HeloiseFFMH, 2016-07-14:
Hello, I deal with the same problem. I tried to depackage the deb file of the camera module with:
sudo dpkg -i --instdir=. libo3d3xx-camera_0.4.5_ubuntu-14.04.4_amd64.deb

But then I still get the error while cmake the framegrabber module. How do I fix it?
Thanks

REBAUD, 2016-07-14:
I follow exactly the installing instruction.
It works with an older version 0.4.0 that I compiled few month ago.
A friend did the same because he had the same issue.

I install o3d3xx-camera by doing a : 
sudo dpkg -i libo3d3xx-camera_0.4.5_ubuntu-14.04.4_i386.deb 

and then issues while doing the "make" for libo3d3xx-framegrabber

REBAUD, 2016-07-14:
@HeloiseFFMH
For the moment e only advice i can give you is to checkout to an older version :s
v0.4.0 works fine for me

tpanzarella, 2016-07-14:
When installing each module, run your cmake command line with
-DCMAKE_INSTALL_PREFIX=/usr And it should work. I will try to get a patch
out in the next day or so, but that should work.

HeloiseFFMH, 2016-07-14:
Hi @REBAUD 
Thank you! Right I am installing version 4.5 too, how to install version 4.0?

REBAUD, 2016-07-14:
Hi

Try first the solution from @tpanzarella he's probably right :)

I think i did 
git checkout tags/v0.4.0 
on the folder which i clone from https://github.com/lovepark/libo3d3xx.git

Hope it's helping you

HeloiseFFMH, 2016-07-14:
Hi @tpanzarella , I am trying with cmake .. -DCMAKE_INSTALL_PREFIX=/usr. Where is the deb package installed? Thank you very much!

HeloiseFFMH, 2016-07-14:
Great
I added -DCMAKE_INSTALL_PREFIX=/usr to the cmake commands and it works!
Thank you very much!

tpanzarella, 2016-07-15:
Just patched `master` so that specifying `-DCMAKE_INSTALL_PREFIX=/usr` is not necessary -- albeit, still preferred. See the commit 255781b for details.

closed 2016-07-15 by tpanzarella