ifm/libo3d3xx #62: FrameGrabberTest.SoftwareTriggerMultipleClients fails

s-hemer, 2016-07-20:
```
[ RUN      ] FrameGrabberTest.SoftwareTriggerMultipleClients
/home/s_hemer/libo3d3xx/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:213: Failure
Value of: fg2->WaitForFrame(buff.get(), 1500)
  Actual: false
Expected: true
/home/s_hemer/libo3d3xx/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp:213: Failure
Value of: fg2->WaitForFrame(buff.get(), 1500)
  Actual: false
Expected: true
[  FAILED  ] FrameGrabberTest.SoftwareTriggerMultipleClients (14519 ms)
[----------] 8 tests from FrameGrabberTest (52595 ms total)

[----------] Global test environment tear-down
[==========] 8 tests from 1 test case ran. (52595 ms total)
[  PASSED  ] 7 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] FrameGrabberTest.SoftwareTriggerMultipleClients

´´´
But unfortunately specifing `cmake -DBUILD_TESTS=OFF ..´ gives the feedback that  manually-specified variables were not used by the project and the test run anyhow.

I am Kubuntu 16.04 with the latest Git version.

s-hemer, 2016-07-20:
Switching to tags v0.4.5 solved the problem somehow !?

tpanzarella, 2016-07-20:
> But unfortunately specifing `cmake -DBUILD_TESTS=OFF ..` gives the feedback that  manually-specified variables were not used by the project and the test run anyhow.

I cannot duplicate this behavior. 

Also, if you need to skip a particular test for some reason, you can use standard `gtest` tooling. So, for this case you can: `$ export GTEST_FILTER=-FrameGrabberTest.SoftwareTriggerMultipleClients`. You'll note the `-` sign tells `gtest` to *not* run this particular test.

In terms of *why* this particular test fails for you, I do not have the time right now to investigate. Perhaps @graugans could shed some light (if he has the time).  @graugans - this test exercises async software triggering for image acquisition. So, it establishes two PCIC socket connections, then one of the two sends a software trigger, and it expects that both of the connected sockets receives the frame. If that is not the case, it fails. [Here](https://github.com/lovepark/libo3d3xx/blob/master/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp#L171) is the relevant code. Upon looking at it quickly, assuming nothing weird on the camera-side (which I don't expect there to be), perhaps the test should allocated separate `ByteBuffer`s for each of the `FrameGrabbers`. Maybe some sort of mutex contention in parsing the byte stream is causing a timing failure -- I'm just speculating and would have to dig into it to make a determination.

tpanzarella, 2016-07-20:
@graugans  I just pushed a change to `master` for this test. Take a look. Seems to have a better success rate, but does fail every once in a while -- inconsistent. The "working theory" of this change was that the loop was potentially exceeding the max limits of the imager configuration in terms of frame rate even though it is s/w triggered. So, I lowered the loop attempts from 10 to 5 and also added a 250ms pause between sending the triggers. Seems to pass more reliably now, but, still not 100%. I need to move on to other things for now, but, this is where I am on it.

[Link](https://github.com/lovepark/libo3d3xx/blob/master/modules/framegrabber/test/o3d3xx-frame_grabber-tests.cpp#L173) to updated test code.

graugans, 2016-07-22:
Sorry, for the delay. I'll dig into this.

graugans, 2016-07-25:
After checking the [SWTrigger()](https://github.com/lovepark/libo3d3xx/blob/ce2d2931c40780a0ae6d286aa9e9b8e8d4154690/modules/framegrabber/src/libo3d3xx_framegrabber/frame_grabber.cpp#L153) method I can not see any handling of return codes. What camera firmware are you testing with?

tpanzarella, 2016-07-25:
Thanks for taking a look into this @graugans 

> After checking the SWTrigger() method I can not see any handling of return codes. What camera firmware are you testing with?

The return code handling is in [TicketHandler](https://github.com/lovepark/libo3d3xx/blob/dd823fb90ebb11e156bf79827094c593e7c0292c/modules/framegrabber/src/libo3d3xx_framegrabber/frame_grabber.cpp#L286). Basically, the frame grabber runs an asynchronous ASIO event loop on a separate thread from the main application thread -- i.e., the thread from which `SWTrigger()` would be called. To that end, in `SWTrigger` we `post` a function to the I/O loop running on the frame grabber's worker thread. Then the `TicketHandler` callback gets called to handle the response from the camera.

Camera firmware is: `1.4.1639`.

The above being said, I saw that when I was getting failures, I was seeing the `!` error from the camera instead of the `*` success indicator. According to the docs `!` means:

- Device busy with evaluation
- Device is in an invalid state for this command
- Device is set to a different trigger source
- No active application

I'd imagine our actual error is one of the top two. I think you have a way to look at the latest errors on the device which would likely shed additional light into this. Can you send that `curl` command to me?

All of the above being said, I think the problem is actually a poorly written test. I have committed a change to the test that I believe tests precisely what this test is trying to assert as true w/o getting into unnecessary complexities in the unit tests with threads and whatnot. Take a look at the [diff](https://github.com/lovepark/libo3d3xx/commit/dd823fb90ebb11e156bf79827094c593e7c0292c). The idea here is that, when we trigger the image acquisition with `fg1`, we retrieve new frame data with `fg2` and vice versa. Again, I think this captures the spirit of the test w/o trying to do concurrent data captures in separate threads. @graugans are you comfortable with this as a proper means of testing the async software trigger across multiple clients?

graugans, 2016-07-25:
The testcase looks pretty valid. Your changes in [diff](https://github.com/lovepark/libo3d3xx/commit/dd823fb90ebb11e156bf79827094c593e7c0292c) are correct. When you have not disabled async messages by ``p`` command any connection will receive Trigger messages. The curl command string is documented [here](https://github.com/lovepark/libo3d3xx/issues/33#issuecomment-165365604), but you can also use the web interface. I'll take a deeper look tomorrow when back in the office

graugans, 2016-07-28:
Sorry It seems like I took a look at the old test code. (Mobile version).  I wrote a [comment](https://github.com/lovepark/libo3d3xx/commit/dd823fb90ebb11e156bf79827094c593e7c0292c) on the commit.
Is there a easy way to just run this particular test-case?

closed 2016-08-25 by tpanzarella