ifm/libo3d3xx #63: Strange temperature values in device parameter node

graugans, 2016-08-12:
When I use either the web interface or curl to retrieve the temperature values provided in the device parameters I get reasonable values. 

```
curl -q --data "method=getAllParameters" http://172.25.125.26/api/rpc/v1/com.ifm.efector/ | grep -C2 TemperatureFront2
        "SessionTimeout": "30",
        "TemperatureFront1": "3276.7",
        "TemperatureFront2": "3276.7",
        "TemperatureIMX6": "33.8520011901855",
        "TemperatureIllu": "38.2",
```
When I do the same with ``o3d3xx-dump`` I get some strange response
```
o3d3xx-dump --ip 172.25.125.26 | grep -C2 TemperatureFront2
            "ImageTimestampReference": "32564960",
            "TemperatureFront1": "6.952798554347323e-310",
            "TemperatureFront2": "1.608934656994936e-316",
            "TemperatureIMX6": "1.608946514570436e-316",
            "TemperatureIllu": "6.952764967629592e-310"
```

tpanzarella, 2016-08-12:
Yes. I discovered this bug myself this week. It is fixed on the HEAD in
Github, and will be part of the official 0.4.6 release that supports the
1.6.x firmware.

tpanzarella, 2016-08-12:
0.4.6 release will address this:

```
$ o3d3xx-version 
libo3d3xx: version=0.4.6

$ o3d3xx-dump  | grep -C2 TemperatureFront2
            "ImageTimestampReference": "0",
            "TemperatureFront1": "3276.6999999999998",
            "TemperatureFront2": "3276.6999999999998",
            "TemperatureIMX6": "43.287998199462898",
            "TemperatureIllu": "46.5",
```

closed 2016-08-12 by tpanzarella

graugans, 2016-08-13:
Sorry for not checking HEAD first...