ifm/libo3d3xx #66: Reflectivity issue off metal objects

metallo25, 2016-08-31:
Hello, 
Thanks for the ros driver; its been helpful for many of my efforts.
I like to have more understanding on how to deal with metal objects. Any metal surface in the scene or the side of the sensor; get reflected off and creates strong grains of points originating from the sensor. This leads to false positives etc. 
Please see the picture below
[image]

I am filtering out the noises; but the noise is much stronger forcing me to get false positives. Also I am using using the version 1.4 driver, not the latest one. Will I be able to solve this issue with the latest driver?. Thank you.

tpanzarella, 2016-09-01:
The driver makes no accommodations to discern noise from a particular surface material. This would have to be handled in your application code. Typically you would exploit _a-priori_ information you have about your application space and integrate the signal processing into your pipeline. That said, some formal studies have been done specific to the O3D (and other sensors) WRT the sensor response versus particular surface materials. I believe @jumarini has performed one such formal study and may be able to share his results.

tpanzarella, 2016-09-01:
In follow up to my comment above, I should note that not all of the tuning needs to be custom code you write into your pipeline. The O3D has exposes many tunable parameters of the PMD chip that you can play with. The `libo3d3xx` driver exposes that stuff to you via the JSON dump from `o3d3xx-dump`. You can edit the JSON and reload into the camera with `o3d3xx-config`.

tpanzarella, 2016-10-27:
@metallo25 Any updates you'd like to supply on this? Can we close the issue?

metallo25, 2016-10-28:
Hello, 
The mode was at "under5m_low", changing that to " upto30m_moderate"(or high), seems to reduce the noise to a level where I can handle through my filters. Yes we can close this issue.

closed 2016-10-28 by tpanzarella