ifm/libo3d3xx #67: Accessing Data Outside Allocation

sgieseking, 2016-09-28:
In the function o3d3xx::get_check_index(), it is accessing data beyond what has been allocated.  The problem is that the buffer includes the end sequence "stop\r\n".  This gets treated as another chunk in which the CHUNK_TYPE is the string "stop".  At the end of the while loop, it adds the CHUNK_SIZE to idx where this is composed of the two bytes "\r\n" and then two bytes passed the end of the allocated data.

One possible fix is to change the comparison at the beginning of the loop to buff.end()-6 to account for the extra data.

This problem was discovered by compiling the code with address sanitizer enabled.

tpanzarella, 2016-09-28:
Eyeballing the code and reading your analysis, this sounds right. As does your proposed fix. Could you:

1. Create a PR
2. In said PR, beyond the the fix, write a unit test that triggers this fix and ensures the correct behavior is happening.

Thanks for pointing this out.

sgieseking, 2016-09-28:
I do work with other open source projects but think that it would be easier for you to write an acceptable unit test and include this one line fix.

sgieseking, 2016-09-28:
Getting the unit tests to compile is a bit difficult without documentation.  The CMakeLists.txt file assume a default location of /usr/src/gtest with binaries in gtest_bin.  There also seems to be an assumption that you have an environment variable set to include the gtest/include directory implicitly in the compilation.  It might be easier to understand if CMakeLists.txt used:

    find_package(GTest) 
    include_directories(${GTEST_INCLUDE_DIR})
    target_link_libraries($(GTEST_LIBRARIES})

tpanzarella, 2016-09-28:
The *a priori* assumption of the location of GTest in `/usr/src/gtest` is a legacy artifact. The location should probably be made into a cache variable that can be overridden on the cmake command line. The presumed location is the location where the Ubuntu package `libgtest-dev` installs -- NOTE: when I originally wrote this library, the intention was to be used as a driver underneath [the ROS package](https://github.com/lovepark/o3d3xx-ros). At the time, using ROS implied Ubuntu. Again, legacy artifact -- but, it hasn't surfaced as an issue until now. If this is a *real* problem for you, please create a new issue for it, or better yet, submit a PR. If you do that latter, please keep reading...

Now, using the approach you are suggesting is something that I would *not* recommend. While it would work for local builds, it makes cross-compiling for embedded systems difficult. The intent in the approach employed by the library is that the gtest code gets built along with your source using your compiler (which may be for a different architecture). When cross-compiling for an embedded system, this lets us deploy the unit tests (along with a properly built GTest) to the embedded target to test directly on the H/W. The assumption is that your embedded target will *not* have a GTest library available.

sgieseking, 2016-09-28:
On the gtest inclusion, I was just noting the difficulty in determining how you assumed it was configured on the computer and a suggestion to either document (since it is turned on by default) or an alternative.  By using find_package(GTest), the required entries will be visible in cmake-gui and can be modified.  The method mentioned is how I have included it in other projects and is just a suggestion.

tpanzarella, 2016-10-27:
Thanks @sgieseking for catching this bug. I have validated it using `valgrind`:

```
[...]

==31689== Conditional jump or move depends on uninitialised value(s)
==31689==    at 0x50FF71A: o3d3xx::get_chunk_index(std::vector<unsigned char, std::allocator<unsigned char> > const&, o3d3xx::image_chunk) (in /usr/lib/libo3d3xx_framegrabber.so.0.4.9)
==31689==    by 0x531AC10: o3d3xx::ImageBuffer::Organize() (in /usr/lib/libo3d3xx_image.so.0.4.9)
==31689==    by 0x5104B18: o3d3xx::FrameGrabber::WaitForFrame(o3d3xx::ByteBuffer*, long, bool, bool) (in /usr/lib/libo3d3xx_framegrabber.so.0.4.9)
==31689==    by 0x40BBAE: main (in /home/tpanzarella/dev/libo3d3xx/modules/examples/build/ex-cartesian_compare)

[...]
```

A [patch](https://github.com/lovepark/libo3d3xx/commit/3b107efd81cc600490bc25d8b9bc994b70ad5834) has been applied on [dev-0.4.9](https://github.com/lovepark/libo3d3xx/tree/dev-0.4.9). This will (hopefully) get merged to the HEAD w/in 1 - 2 weeks.

closed 2016-10-27 by tpanzarella