ifm/libo3d3xx #68: Framegrabber stops when saving PCDs

Obbart, 2016-09-29:
Hi, 
I tried saving some in my Qt+PCL+OpenCV application.. saving .png or .bmp with cv::imwrite works just fine, but when I try to use pcd::io::savePCDFileASCII() the framegrabber stops.
The function that saves the images is called by a Qt button and the most strange thing is that framegrabber stops even if the function is not called. Seems that only the "insertion" of pcd::io::savePCDFileASCII() causes the problem. My code is very similar to ex-file_io.cpp

Thanks

tpanzarella, 2016-09-29:
Without seeing your code, there is not much meaningful troubleshooting I can help you with. Posting your code, sending some relevant information from the log file (found in `/tmp` by default), and stating the version of `libo3d3xx` that you are using will be helpful.

Now, you are suggesting that the `FrameGrabber` thread __stops__ as opposed to __crashes__. If it does in fact __crash__ and it is triggered by using PCL, please read Issue #7 and see if it applies. If the `FrameGrabber` just __stops__ as you state, the one thing I would ask you to do is check how you are managing ownership of the `FrameGrabber` within your Qt event loop. If you gain access to the `FrameGrabber` using the recommended

```
o3d3xx::FrameGrabber::Ptr fg = std::make_shared<o3d3xx::FrameGraber>(...)
```
Recall that `fg` is a `std::shared_ptr` to an `o3d3xx::FrameGrabber` pointer. So, if somewhere in your event loop, the reference count to `fg` drops to 0, the smart pointer will call the `fg` dtor and, yes, by design, the frame grabbing thread will stop.

Again, w/o seeing the code, I am only guessing at what could be going on. Good luck.

Obbart, 2016-09-29:
Initialization
```
...
...
// instantiate the camera with default ip, xmlrpc port, and password
	o3d3xx::Camera::Ptr cam;

	// associate the camera with a frame grabber
	o3d3xx::FrameGrabber::Ptr fg;

	// create an buffer for holding the image data from the camera
	o3d3xx::ImageBuffer::Ptr buff;

...
...

bool RoboVis_VISION::cam_Setup() {
	try {
		cam = std::make_shared<o3d3xx::Camera>(m->camIp.toStdString());
		fg = std::make_shared<o3d3xx::FrameGrabber>(cam);
		buff = std::make_shared<o3d3xx::ImageBuffer>();
		return true;
	} catch (const o3d3xx::error_t& e) {
		printf("%s\n",e.what());
		return false;
	}
}
```
Image Capture
```
cv::Mat RoboVis_VISION::capture(bool *valid) {
	cv::Mat temp;
	uchar i = 0;
	emit wait();

#ifndef TEST  //trigger camera
	while (m->gpio.digitalRead(0) == false){
		usleep(500);
	}
	m->gpio.digitalWrite(0,true);
#else
	fg->SWTrigger();
#endif

	while (true) {
		if (!fg->WaitForFrame(buff.get(),1500,true)) {
			if (i++ < MAX_ATTEMPT)
				continue;
			else {
				error(CAMERA_ERROR);
				*valid = false;
				fg ->Stop();
				cam->CancelSession();
				is_live = false;
				sleep(0.1);
				this->cam_Setup();
				return old_img;
			}
		} else
			break;
	}

#ifndef TEST
	m->gpio.digitalWrite(0,false);
#endif

	//Capture New Image
	temp = buff->XYZImage();
	m->cloud = buff->Cloud();
	temp.copyTo(old_img);
	*valid = true;
	return temp;
}
```
Image Save
```
void RoboVis_GUI::on_btnSaveImage(){
	QDateTime t= QDateTime::currentDateTime();
	QString tt=t.toString("yyyyMMdd-hh:mm:ss");
	QString bmp = tt.append(".bmp");
	QString pcd = tt.append(".pcd");

	if (m->mutex.tryLock(100)){
		cv::imwrite(cv::String(std::string(BASE_DIR) + std::string(IMG_SAVE) + bmp.toStdString()), m->vis_image);
		pcl::io::savePCDFileASCII(std::string(BASE_DIR) + std::string(IMG_SAVE) + pcd.toStdString(), *(m->cloud));
	}
	MYLOG(QString("SAVE -> ") + tt);
}
```

m is an istance of a class that contains:
```
pcl::PointCloud<pcl::PointXYZI>::Ptr cloud;
cv::Mat vis_image;
```

Hope this is useful :)

tpanzarella, 2016-09-29:
Can you paste in the output of the log file just after the issue you are describing occurs? By default it is written to `/tmp/libo3d3xx.INFO`. Also, which version of libo3d3xx are you using? You can get that by running  `o3d3xx-version`.

Obbart, 2016-09-29:
`version=0.4.5`
in /tmp/libo3d3xx.INFO
```
Running on machine: RoboVis-PC
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0929 16:13:20.319000  6521 frame_grabber.cpp:70] Camera connection info: ip=192.168.0.69, port=50010
I0929 16:13:21.253365  6521 frame_grabber.cpp:91] Assumed trigger mode: 2
I0929 16:13:21.290552  6525 frame_grabber.cpp:458] Framegrabber thread done.
```

tpanzarella, 2016-09-29:
And now the output from: `o3d3xx-dump`.

Obbart, 2016-09-29:
```
{
    "o3d3xx":
    {
        "libo3d3xx": "405",
        "Date": "Thu Sep 29 16:17:43 2016",
        "HWInfo":
        {
            "Mainboard": "#!02_M381_003_02178445_008022954",
            "Illumination": "#!02_I300_001_02161610_008001175",
            "MiraSerial": "0030-53b9-0022-03bc",
            "MACAddress": "00:02:01:40:30:49",
            "Frontend": "#!02_F342_C32_16_00005_008022797",
            "Diagnose": "#!02_D322_C30_02165464_008023093",
            "Connector": "#!02_A300_001_02064651_008001175"
        },
        "SWVersion":
        {
            "IFM_Software": "1.4.1639",
            "Linux": "Linux version 3.11.0-00001-gda3d004 (eswadmin@dettlx79) (gcc version 4.7.2 (GCC) ) #1 SMP Wed Feb 17 15:55:11 CET 2016",
            "Diagnostic_Controller": "0.3.150",
            "ELDK": "GOLDENEYE_ELDK-2016-03-17_16-21-02-8baf2bee4a4eebca0d2ad7c7345d3a4f971032ec",
            "Calibration_Version": "0.9.160314",
            "Calibration_Device": "00:02:01:40:30:49",
            "Main_Application": "0.4.164",
            "Algorithm_Version": "1.2.14"
        },
        "Device":
        {
            "Name": "RoboVis_SENSOR",
            "Description": "",
            "ActiveApplication": "2",
            "PcicEipEnabled": "false",
            "PcicTcpPort": "50010",
            "PcicProtocolVersion": "3",
            "IOLogicType": "1",
            "IODebouncing": "true",
            "IOExternApplicationSwitch": "0",
            "SessionTimeout": "30",
            "ServiceReportPassedBuffer": "15",
            "ServiceReportFailedBuffer": "15",
            "ExtrinsicCalibTransX": "0",
            "ExtrinsicCalibTransY": "0",
            "ExtrinsicCalibTransZ": "0",
            "ExtrinsicCalibRotX": "0",
            "ExtrinsicCalibRotY": "0",
            "ExtrinsicCalibRotZ": "0",
            "EvaluationFinishedMinHoldTime": "10",
            "SaveRestoreStatsOnApplSwitch": "true",
            "IPAddressConfig": "0",
            "PasswordActivated": "false",
            "OperatingMode": "0",
            "DeviceType": "",
            "ArticleNumber": "",
            "ArticleStatus": "",
            "UpTime": "9.140475314724065e-317",
            "ImageTimestampReference": "18500768",
            "TemperatureFront1": "6.952798554347323e-310",
            "TemperatureFront2": "9.140712466234069e-317",
            "TemperatureIMX6": "9.140831041989071e-317",
            "TemperatureIllu": "6.952764967629592e-310"
        },
        "Net":
        {
            "MACAddress": "",
            "NetworkSpeed": "0",
            "StaticIPv4Address": "192.168.0.69",
            "StaticIPv4SubNetMask": "255.255.255.0",
            "StaticIPv4Gateway": "192.168.0.201",
            "UseDHCP": "false"
[... 83 lines cut ...]
                    "SymmetryThreshold": "0",
                    "TemporalFilterType": "1",
                    "ThreeFreqMax2FLineDistPercentage": "80",
                    "ThreeFreqMax3FLineDistPercentage": "80",
                    "TwoFreqMaxLineDistPercentage": "80",
                    "Type": "under5m_high",
                    "MaxAllowedLEDFrameRate": "21.5",
                    "SpatialFilter":
                    {
                        "Type": "0",
                        "TypeStr_": "Off"
                    },
                    "TemporalFilter":
                    {
                        "Type": "1",
                        "TypeStr_": "Temporal Mean Filter",
                        "NumberOfImages": "4"
                    }
                }
            }
        ]
    }
}

```

tpanzarella, 2016-09-29:
Hmmm.... I've configured my camera to look like yours (most notably w/ the 100k pixel imager, single frequency, triple exposure, 4-image mean filter, software triggered, etc.). I do not run into any problems. NOTE: I am using libo3d3xx 0.4.8 linked to PCL 1.7. I modified the ex-file_io example to run in a loop and software trigger the camera for image aquisition. That looks like this:

```
#include <iostream>
#include <memory>
#include <opencv2/opencv.hpp>
#include <pcl/io/pcd_io.h>
#include <pcl/point_cloud.h>
#include "o3d3xx_camera.h"
#include "o3d3xx_framegrabber.h"
#include "o3d3xx_image.h"

int main(int argc, const char **argv)
{
  o3d3xx::Logging::Init();

  o3d3xx::Camera::Ptr cam = std::make_shared<o3d3xx::Camera>();
  o3d3xx::ImageBuffer::Ptr img = std::make_shared<o3d3xx::ImageBuffer>();
  o3d3xx::FrameGrabber::Ptr fg =
    std::make_shared<o3d3xx::FrameGrabber>(
      cam, o3d3xx::IMG_AMP|o3d3xx::IMG_RDIS|o3d3xx::IMG_CART);

  for (int i = 0; i < 20; i++)
    {
      std::cout << i << std::endl;
      fg->SWTrigger();
      if (! fg->WaitForFrame(img.get(), 1500))
        {
          std::cerr << "Timeout waiting for camera!" << std::endl;
          return -1;
        }

      std::string i_s(std::to_string(i));
      pcl::io::savePCDFileASCII(
        "/tmp/point_cloud_" + i_s + ".pcd", *(img->Cloud()));

      imwrite("/tmp/amplitude_" + i_s + ".png", img->AmplitudeImage());
    }

  return 0;
}
```

With this code, I am not having any problems. Could you run this program and see what you get -- hopefully a bunch of files in your `/tmp` directory :-)

Obbart, 2016-09-29:
Ehm.. I'm not able to get your code running... I'm not a mekefile guru and g++ does not resolve the includes.. How should I compile it? Thanks :)

tpanzarella, 2016-09-29:
Easiest would be to edit the file in the examples module directory to look like what I have above (or similar) then just compile it using the provided `CMakeLists.txt`. For example:

```
$ cd modules/examples
$ .... EDIT THE FILE AS APPROPRIATE ...
$ mkdir build
$ cd build
$ cmake ..
$ make
```

The binary will be dropped into your current working directory.

Obbart, 2016-09-29:
ok, your code is working fine.. mine not, same problem..
The log output changed:
```
I0929 18:20:55.602010  4343 frame_grabber.cpp:70] Camera connection info: ip=192.168.0.69, port=50010
I0929 18:20:56.706789  4343 frame_grabber.cpp:91] Assumed trigger mode: 2
I0929 18:20:56.747426  4344 frame_grabber.cpp:458] Framegrabber thread done.
W0929 18:21:08.513485  4343 frame_grabber.cpp:171] pcic_ready_ flag not set!
```

tpanzarella, 2016-09-29:
OK. The last line indicates that your code is trying to do a software trigger prior to the PCIC schema being set on the camera -- odd that it is not set as `SWTrigger()` tries for two seconds before giving up and emitting that error. This tells me the usage pattern is a little different. No big deal, it should still work, once  I get a better mental model of how your code is structured.

A few thoughts/questions:

1. Could you update to the latest version of libo3d3xx? Now since you want to use the 100K pixel imager this could be tricky, I understand. You may want to install two parallel copies of libo3d3xx on your computer. For example, the version you are using, 0.4.5, could be installed in `/opt/libo3d3xx-0.4.5` and you can use this install for the command line tools (i.e., so it can speak the proper dialect of XML-RPC for the firmware you are running). Then, use 0.4.8 for your runtime code -- the `FrameGrabber` code in this version is more robust to seeing errors in the data stream back from the camera.

2. Looks like you are using this in a multi-threaded environment. You should take care to handle all synchronization yourself. The core objects are not thread safe.

3. Related to the previous point, in your `cam_Setup()`, I'd suggest using the `.reset(..)` call on the `shared_ptr` rather than using the assignment operator -- particularly on the `FrameGrabber`. Again, be sure to grab a mutex or some other synchronization tool before resetting the objects.

4. For troubleshooting purposes, I would be interested to see if you set your camera into FREE_RUN mode and also removed the call to `SWTrigger()` from your code, to see if that would work. I understand you likely have a good reason for wanting to software trigger the camera (thermal management, etc.) but again, just for troubleshooting purposes, I'd be curious to see what happens.

My gut tells me, we need to think about how you are managing the `FrameGrabber` across threads, but, the specific root of the problem is not yet clear to me. Let me know if you try any/some/all of the above -- and your results.

Obbart, 2016-09-29:
OK. Tried points 3 & 4 also using a mutex as synchronization tool with no success.
My code has 3 threads (QThread) and a class common to all the 3 threads for sharing data between them. Framegrabber is a private variable in only one of the threads and is not accessed by any of the others.
I'll try to upgrade sensor firmware and libo3d3xx version.
Sorry if I'm a bit confusing but my english is not the best..

tpanzarella, 2016-09-29:
> Sorry if I'm a bit confusing but my english is not the best..
Your English is fine, not confusing.

OK, just be aware that the latest firmware will not allow you to set the camera into 100k pixel mode. This is why I suggested my confusing point 1 above. Doing that will let you keep the firmware you have on the camera and use a version of `libo3d3xx` that can speak that dialect of XML-RPC for configuration purposes. Then, use the updated version of `libo3d3xx` for your runtime code and to get the benefit of the better `FrameGrabber`. I'm not saying this is the solution to our problem here, but, there were some recent changes to the `FrameGrabber` that make it more robust.

tpanzarella, 2016-10-27:
@Obbart Did you get this resolved?

Obbart, 2016-10-28:
Not yet, I had many other projects to attend to this month and we needed this software to start a plant. 
However I spoke with an Ifm tecnician and he said me that the factory default 100k imager will be ready soon. I will try your solution as soon as possible.

closed 2017-08-25 by tpanzarella