ifm/libo3d3xx #69: o3d3xx_camera_shared not found

sjorsjee, 2016-10-14:
I'm trying to install libo3d3xx but when i run the make command I get this error. It looks like I'm missing two files. 

someone who can help ? 

sjors@sjors-VirtualBox:~/libo3d3xx/build$ make
[  3%] Performing configure step for 'O3D3XX_CAMERA'
-- UBUNTU_VERSION: 14.04.5
-- UBUNTU_MAJOR: 14
-- UBUNTU_MINOR: 04
-- UBUNTU_PATCH: 5
-- XMLRPC was not found. Make sure the entries XMLRPC_* are set.
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   program_options
--   system
-- DEB_FILE: libo3d3xx-camera_0.4.8_ubuntu-14.04.5_amd64.deb
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIB_glog
    linked by target "o3d3xx_camera_shared" in directory /home/sjors/libo3d3xx/modules/camera/src/libo3d3xx_camera
    linked by target "o3d3xx_camera_static" in directory /home/sjors/libo3d3xx/modules/camera/src/libo3d3xx_camera
LIB_xmlrpc_clientxx
    linked by target "o3d3xx_camera_shared" in directory /home/sjors/libo3d3xx/modules/camera/src/libo3d3xx_camera
    linked by target "o3d3xx_camera_static" in directory /home/sjors/libo3d3xx/modules/camera/src/libo3d3xx_camera
LIB_xmlrpcxx
    linked by target "o3d3xx_camera_shared" in directory /home/sjors/libo3d3xx/modules/camera/src/libo3d3xx_camera
    linked by target "o3d3xx_camera_static" in directory /home/sjors/libo3d3xx/modules/camera/src/libo3d3xx_camera

-- Configuring incomplete, errors occurred!
See also "/home/sjors/libo3d3xx/build/O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-build/CMakeFiles/CMakeOutput.log".
See also "/home/sjors/libo3d3xx/build/O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-build/CMakeFiles/CMakeError.log".
make[2]: *** [O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-stamp/O3D3XX_CAMERA-configure] Error 1
make[1]: *** [CMakeFiles/O3D3XX_CAMERA.dir/all] Error 2
make: *** [all] Error 2

tpanzarella, 2016-10-14:
Looks like you are missing `libxmlrpc-c++8-dev` and `libgtest-dev`. You can install them with `apt`.

tpanzarella, 2016-10-27:
@sjorsjee Can we close this?

sjorsjee, 2016-10-28:
Yes thank you very much

closed 2016-10-28 by tpanzarella