ifm/libo3d3xx #70: Handling of asynchronous error and notifications messages

graugans, 2016-10-26:
I would like to discuss a new feature for libo3d3xx. Within the latest firmware release we provide a mechanism called asynchronous error messages and notifications. The asynchronous messages are used to signalize stuff like application and status changes. The error messages as the name implies are used when the camera hits a state where it can not operate any more. Those include 

- Over temperature
- Over voltage detection
- Under volt detection
- Short circuit on the outputs

The messages are turned off by default. They can be enabled by sending a ``p<state>`` command over the PCIC TCP/IP socket. The ``state`` value can be one of the following ones

- 0 deactivate all asynchronous output
- 1 activate asynchronous result output
- 2 activate asynchronous error output
- 3 activate asynchronous error and data output
- 4 activate asynchronous notifications
- 5 activate asynchronous notifications and asynchronous result
- 6 activate asynchronous notifications and asynchronous error output
- 7 activate all outputs

The each type of asynchronous notification event has its own ticket number

- asynchronous error codes *0001*
- asynchronous notifications *0010*

## The Format for asynchronous notifications

``<ticket=0010>L<length>CR+LF<ticket=0010><unique message ID>:<JSON content>CR LF``

The message id is formated like this ``"%09u"`` nine digits padded with leading zeros.

### Available Asynchronous Message IDs

| ID |Description| Example  |Description|
|---|---|---|---|
| 000500000|Application changed|{"ID": 1034160761,"Index":1,"Name": "Pos 1","valid":true}|   |
| 000500001|Application is not valid|{"ID": 1034160761,"Index":1,"Name": "Pos 1","valid":false}|If a application exists on given index but it is invalid, the ID and Name are filled accoring to the application. If there is no application on given index the application ID will contain 0 and the name an empty string ""|

## Format of asynchronous error messages

For error messages there is only the error id transmitted in the following format ``"%09u"`` nine digits padded with leading zeros. 

At the moment the following error codes are supported

|Code-ID|	Error|
|---|---|
|100000001|Maximum number of connections exceeded |
|110001001|Boot timeout|
|110001002|Fatal  error|
|110001003|Unknown hardware|
|110001006| Trigger overrun|
|110002000|Short circuit on Ready for Trigger|
|110002001|Short circuit on OUT1|
|110002002|Short circuit on OUT2|
|110002003|Reverse feeding|
|110003000|Vled overvoltage|
|110003001|Vled undervoltage|
|110003002|Vmod overvoltage|
|110003003|Vmod undervoltage|
|110003004|Mainboard overvoltage|
|110003005|Mainboard undervoltage|
|110003006|Supply overvoltage|
|110003007|Supply undervoltage|
|110003008|VFEMon alarm|
|110003009|PMIC supply alarm|
|110004000|Illumination overtemperature|
  
I am very open to discuss the possibility to add new notification messages on request.

After a quick check of the current [Framegrabber](https://github.com/lovepark/libo3d3xx/blob/master/modules/framegrabber/include/o3d3xx_framegrabber/frame_grabber.h) code  of the lib I thought the best idea would be to add a ``FrameGrabber``  interface and a  ``PCICConnection`` base class . The ``PCICFramegrabber`` realizes the  ``FrameGrabber`` interface and inherits the  ``PCICConnection``. After this we can provide a ``AsyncMessage`` interface which provides a method like [WaitForFrame](https://github.com/lovepark/libo3d3xx/blob/master/modules/framegrabber/include/o3d3xx_framegrabber/frame_grabber.h#L122). Another class  let's call it ``PCICAsyncMessage`` can realize the ``AsyncMessage`` interface and sub-class ``PCICConnection``.

tpanzarella, 2016-10-26:
Thanks for sharing these details @graugans 

Given the asynchronous nature of these messages, perhaps an alternate interface to this would be the addition of two public methods on the `FrameGrabber`:

1. RegisterAsyncErrorHandler(...)
2. RegisterAsyncNotificationHandler(...)

The signature of these methods would have to be sorted out, however, I envision them basically providing a `mask` and a callback function. The `mask` would indicate which errors or notifications should fire the particular handler, and the  callback function would be the handler itself (i.e., I would envision that the majority of the time the caller would likely just pass a lambda to be called). The handler would, of course, get passed the payload of the message from the sensor.

Internally, the `FrameGrabber` could, for example, manage two hash tables mapping `masks` to a vector of handlers to call for the given mask -- (i.e., we need to prepare for `FrameGrabber` instances shared across multiple clients). In order to not introduce (too much) latency in processing the pixel data, the firing of the handlers could be dispatched to a future or explicitly a thread -- basically, for poorly crafted handler functions (i.e., those that are *long running*) we do not want to execute the handlers in the context of the main frame grabbing thread.

It seems to me that regardless of the implementation, once we are in there adding these enhancements to the interface, this class should employ PIMPL as the implementation is going to start getting slightly more complicated. Further, the details of things like ASIO, etc. is really of no interest to the clients.

These are just my initial thoughts. Thanks for starting this discussion.

graugans, 2016-10-26:
This is another option, I thought about introducing a new class, because I had the on-board framegrabber changes in mind. For the onboard code we'll still need *PCIC* for asynchronous messages. There is no interface in the sync library for this.

tpanzarella, 2016-10-26:
For the on-board framegrabber, is it possible to efficiently use the sync library for pixel data and a separate PCIC socket simply for the async messages? Could we set the PCIC schema to "null" which, I presume, would result in only the async messages being transmitted over that connection?

graugans, 2016-10-27:
Sending an empty ``PCIC Schema`` is not enough. When this feature was introduced we decided to not send the asynchronous notifications by default. Some implementations do not handle the asynchronous results very well. We expected those to crash when sending new asynchronous events with different ticket numbers. To activate the the asynchronous notifications and error messages one have to enable them with the ``p<state>`` command.  By sending ``p6`` only notifications and error messages pass by.

The sync client can be used in parallel to the traditional ``PCIC`` socket on the camera. A on board sync client can also work in parallel with for example a *ROS* client or the ifm Vision Assistant.

tpanzarella, 2016-10-27:
It sounds like this could be a nice way forward to implement this feature. Specifically, processing the async messages on the ASIO event loop while proxying retrieval of image data through to the sync lib via `WaitForFrame(..)`. We could certainly implement this so that it is transparent to the calling code in terms of on-board vs. off-board clients.

graugans, 2016-10-27:
Sounds like a plan, so I try to prepare some proposal (PR) by end of next week. The ``FrameGrabber``class will be extended by two methods:

- ``RegisterAsyncErrorHandler(...)``
- ``RegisterAsyncNotificationHandler(...)``

As a starter I'll switch the class to Pimpl. Regarding the use of a future or thread I have to evaluate what are the pros or con of each aproach.

tpanzarella, 2016-10-27:
@graugans I think we should wait as I will be changing / updating the `FrameGrabber` as well to accommodate for the changes that Ville has requested in the near-term. I do not want to make the merge overly difficult. We will have to coordinate our efforts. FWIW, I'll be working on Ville's requests on [this branch](https://github.com/lovepark/libo3d3xx/tree/dev-0.4.9)

closed 2017-08-25 by tpanzarella