ifm/libo3d3xx #71: How to set 100k mode

jamesdebono, 2016-11-04:
Sorry if this is the wrong place to post such a question, I haven't found any user forum etc.
I am trying to set 100k mode per this: "We note that you really should use o3d3xx-dump and o3d3xx-config to create an application that employs the 100k imager"
I must be getting the syntax wrong, but can't find what is correct. I keep getting "XMLRPC value out of range" when I try changing resolution. What should I be setting it to?

graugans, 2016-11-04:
I am very sorry but 100k was removed from the regular O3D30x series.There will be a dedicated O3D3xx version with 100k support. Please contact your local ifm contact or maybe  @garrettplace can provide some details.

jamesdebono, 2016-11-07:
Thanks I will discuss with the local IFM, but for now downgrading to firmware 1.4.1639 has worked

closed 2017-08-25 by tpanzarella