ifm/libo3d3xx #72: Error after upgrade firmware

zfxw0206, 2016-11-08:
After I upgrade the firmware to 1.6.2038, I got the following error:

****_terminate called after throwing an instance of 'o3d3xx::error_t'
  what():  XMLRPC device already has an edit session active
[o3d3xx70/camera-2] process has died [pid 7029, exit code -6, cmd /home/visiongm/wspace/devel/lib/o3d3xx/o3d3xx_node /cloud:=/o3d3xx70/camera/cloud /depth:=/o3d3xx70/camera/depth /depth_viz:=/o3d3xx70/camera/depth_viz /amplitude:=/o3d3xx70/camera/amplitude /raw_amplitude:=/o3d3xx70/camera/raw_amplitude /confidence:=/o3d3xx70/camera/confidence /good_bad_pixels:=/o3d3xx70/camera/good_bad_pixels /hist:=/o3d3xx70/camera/hist /xyz_image:=/o3d3xx70/camera/xyz_image /GetVersion:=/o3d3xx70/camera/GetVersion /Dump:=/o3d3xx70/camera/Dump /Config:=/o3d3xx70/camera/Config /Rm:=/o3d3xx70/camera/Rm __name:=camera __log:=/home/visiongm/.ros/log/1491db9c-a5cc-11e6-b6fd-003018c8f466/o3d3xx70-camera-2.log].
log file: /home/visiongm/.ros/log/1491db9c-a5cc-11e6-b6fd-003018c8f466/o3d3xx70-camera-2*.log_****

What does this mean? How can I fix it?

tpanzarella, 2016-11-08:
You already have an active edit session with the camera (i.e., you are configuring its parameters over XML-RPC), and you are trying to create another one (edit session) without closing your previous one. Only one active edit session is allowed by the camera.

Now, how you got into that situation, I don't know. We don't have nearly enough info to help. Let's start with the version stack you are running: IFM Firmware, libo3d3xx, o3d3xx-ros

zfxw0206, 2016-11-08:
After I got this error, I just wait for about 5 min, and run the camera.launch again, the error is gone. Thank you anyway.

tpanzarella, 2016-11-08:
That is because the session with the camera timed out, and the camera cleaned it up for you.

You should still be sure you are running a supported software stack by looking at the README files for both `libo3d3xx` and `o3d3xx-ros`. On those pages, there is a compatibility matrix. I highly encourage you to make sure your stack is supported. Closing this for now.

closed 2016-11-08 by tpanzarella

nathangeorge1, 2017-10-17:
I found this error pops up when running multiple thread (-j3) while trying to make the package as well. I'd suggest some kind of note in the quick start to not use -j3 when building.