ifm/libo3d3xx #73: Failed Unit Tests

Yarion, 2016-11-09:
Hello,

I'm trying to build the libo3d3xx lib for the IFM OD3D303 camera.
However 6 unit tests fail during the build.
Can anyone help me with these issues 

Camera info:
O3D303 AB Camera 
Version 1.4.1639

The make:
[  3%] Performing test step for 'O3D3XX_CAMERA'
[  5%] Built target gtest
[ 10%] Built target gtest_main
[ 65%] Built target o3d3xx_camera
[ 65%] Built target o3d3xx_camera_shared
[100%] Built target o3d3xx-camera-tests
[==========] Running 44 tests from 6 test cases.
[----------] Global test environment set-up.
[----------] 4 tests from TemporalFilterTest
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_General
[       OK ] TemporalFilterTest.TemporalFilterConfig_General (994 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameters
[       OK ] TemporalFilterTest.GetTemporalFilterParameters (1711 ms)
[ RUN      ] TemporalFilterTest.GetTemporalFilterParameterLimits
[       OK ] TemporalFilterTest.GetTemporalFilterParameterLimits (2013 ms)
[ RUN      ] TemporalFilterTest.TemporalFilterConfig_JSON
[       OK ] TemporalFilterTest.TemporalFilterConfig_JSON (1895 ms)
[----------] 4 tests from TemporalFilterTest (6613 ms total)

[----------] 2 tests from General_Tests
[ RUN      ] General_Tests.Exceptions
[       OK ] General_Tests.Exceptions (0 ms)
[ RUN      ] General_Tests.Version
[       OK ] General_Tests.Version (0 ms)
[----------] 2 tests from General_Tests (0 ms total)

[----------] 4 tests from SpatialFilterTest
[ RUN      ] SpatialFilterTest.SpatialFilterConfig_General
[       OK ] SpatialFilterTest.SpatialFilterConfig_General (891 ms)
[ RUN      ] SpatialFilterTest.GetSpatialFilterParameters
[       OK ] SpatialFilterTest.GetSpatialFilterParameters (1785 ms)
[ RUN      ] SpatialFilterTest.GetSpatialFilterParameterLimits
[       OK ] SpatialFilterTest.GetSpatialFilterParameterLimits (1946 ms)
[ RUN      ] SpatialFilterTest.SpatialFilterConfig_JSON
[       OK ] SpatialFilterTest.SpatialFilterConfig_JSON (1874 ms)
[----------] 4 tests from SpatialFilterTest (6497 ms total)

[----------] 15 tests from AppImagerTest
[ RUN      ] AppImagerTest.CopyDeleteApplication
[       OK ] AppImagerTest.CopyDeleteApplication (1246 ms)
[ RUN      ] AppImagerTest.CreateApplication
[       OK ] AppImagerTest.CreateApplication (866 ms)
[ RUN      ] AppImagerTest.ChangeAppNameAndDescription
[       OK ] AppImagerTest.ChangeAppNameAndDescription (1016 ms)
[ RUN      ] AppImagerTest.EditApplication
[       OK ] AppImagerTest.EditApplication (1128 ms)
[ RUN      ] AppImagerTest.GetAppParameters
/home/maarten/3dCameraLib/newlibo3d3xx/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:131: Failure
Value of: 9
Expected: params.size()
Which is: 8
[  FAILED  ] AppImagerTest.GetAppParameters (1552 ms)
[ RUN      ] AppImagerTest.AppConfig
[       OK ] AppImagerTest.AppConfig (1864 ms)
[ RUN      ] AppImagerTest.AppConfig_JSON
[       OK ] AppImagerTest.AppConfig_JSON (1697 ms)
[ RUN      ] AppImagerTest.GetAvailableImagerTypes
[       OK ] AppImagerTest.GetAvailableImagerTypes (1497 ms)
[ RUN      ] AppImagerTest.ChangeImagerType
[       OK ] AppImagerTest.ChangeImagerType (2367 ms)
[ RUN      ] AppImagerTest.GetImagerParameters
/home/maarten/3dCameraLib/newlibo3d3xx/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:270: Failure
Expected: params.at("UseSimpleBinning") doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.GetImagerParameters (1700 ms)
[ RUN      ] AppImagerTest.GetImagerParameterLimits
[       OK ] AppImagerTest.GetImagerParameterLimits (2951 ms)
[ RUN      ] AppImagerTest.ImagerConfig
/home/maarten/3dCameraLib/newlibo3d3xx/libo3d3xx/modules/camera/test/o3d3xx-app-imager-tests.cpp:420: Failure
Expected: cam_->SetImagerConfig(im.get()) doesn't throw an exception.
  Actual: it throws.
[  FAILED  ] AppImagerTest.ImagerConfig (2136 ms)
[ RUN      ] AppImagerTest.ImagerConfigValueOutOfRange
[       OK ] AppImagerTest.ImagerConfigValueOutOfRange (2861 ms)
[ RUN      ] AppImagerTest.ImagerConfig_JSON
[       OK ] AppImagerTest.ImagerConfig_JSON (1521 ms)
[ RUN      ] AppImagerTest.Exposure
[       OK ] AppImagerTest.Exposure (2881 ms)
[----------] 15 tests from AppImagerTest (27284 ms total)

[----------] 18 tests from Camera_Tests
[ RUN      ] Camera_Tests.Ctor
[       OK ] Camera_Tests.Ctor (0 ms)
[ RUN      ] Camera_Tests.GetXMLRPCURLPrefix
[       OK ] Camera_Tests.GetXMLRPCURLPrefix (0 ms)
[ RUN      ] Camera_Tests.GetAllParameters
[       OK ] Camera_Tests.GetAllParameters (374 ms)
[ RUN      ] Camera_Tests.GetParameter
[       OK ] Camera_Tests.GetParameter (1149 ms)
[ RUN      ] Camera_Tests.GetSWVersion
[       OK ] Camera_Tests.GetSWVersion (32 ms)
[ RUN      ] Camera_Tests.GetApplicationList
[       OK ] Camera_Tests.GetApplicationList (15 ms)
[ RUN      ] Camera_Tests.RequestSession
[       OK ] Camera_Tests.RequestSession (56 ms)
[ RUN      ] Camera_Tests.CancelSession
[       OK ] Camera_Tests.CancelSession (50 ms)
[ RUN      ] Camera_Tests.Heartbeat
[       OK ] Camera_Tests.Heartbeat (97 ms)
[ RUN      ] Camera_Tests.SetOperatingMode
[       OK ] Camera_Tests.SetOperatingMode (469 ms)
[ RUN      ] Camera_Tests.GetDeviceConfig
/home/maarten/3dCameraLib/newlibo3d3xx/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:182: Failure
Value of: 34
Expected: params.size()
Which is: 32
unknown file: Failure
C++ exception with description "_Map_base::at" thrown in the test body.
[  FAILED  ] Camera_Tests.GetDeviceConfig (636 ms)
[ RUN      ] Camera_Tests.ActivateDisablePassword
[       OK ] Camera_Tests.ActivateDisablePassword (462 ms)
[ RUN      ] Camera_Tests.SetDeviceConfig
/home/maarten/3dCameraLib/newlibo3d3xx/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:269: Failure
Expected: cam->SetDeviceConfig(dev.get()) doesn't throw an exception.
  Actual: it throws.
/home/maarten/3dCameraLib/newlibo3d3xx/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:275: Failure
Value of: dev->EthernetFieldBus()
  Actual: 0
Expected: tmp_eth_field_bus
Which is: 2
/home/maarten/3dCameraLib/newlibo3d3xx/libo3d3xx/modules/camera/test/o3d3xx-camera-tests.cpp:276: Failure
Value of: dev->EthernetFieldBusEndianness()
  Actual: 0
Expected: tmp_eth_field_bus_endianness
Which is: 1
[  FAILED  ] Camera_Tests.SetDeviceConfig (2774 ms)
[ RUN      ] Camera_Tests.DeviceConfig_JSON
[       OK ] Camera_Tests.DeviceConfig_JSON (261 ms)
[ RUN      ] Camera_Tests.GetNetParameters
[       OK ] Camera_Tests.GetNetParameters (372 ms)
[ RUN      ] Camera_Tests.NetConfig
[       OK ] Camera_Tests.NetConfig (429 ms)
[ RUN      ] Camera_Tests.NetConfig_JSON
[       OK ] Camera_Tests.NetConfig_JSON (414 ms)
[ RUN      ] Camera_Tests.TemporaryParameters
unknown file: Failure
C++ exception with description "XMLRPC method not found" thrown in the test body.
[  FAILED  ] Camera_Tests.TemporaryParameters (51 ms)
[----------] 18 tests from Camera_Tests (7641 ms total)

[----------] 1 test from ImportExport_Tests
[ RUN      ] ImportExport_Tests.ImportExportApp
[       OK ] ImportExport_Tests.ImportExportApp (859 ms)
[----------] 1 test from ImportExport_Tests (859 ms total)

[----------] Global test environment tear-down
[==========] 44 tests from 6 test cases ran. (48894 ms total)
[  PASSED  ] 38 tests.
[  FAILED  ] 6 tests, listed below:
[  FAILED  ] AppImagerTest.GetAppParameters
[  FAILED  ] AppImagerTest.GetImagerParameters
[  FAILED  ] AppImagerTest.ImagerConfig
[  FAILED  ] Camera_Tests.GetDeviceConfig
[  FAILED  ] Camera_Tests.SetDeviceConfig
[  FAILED  ] Camera_Tests.TemporaryParameters

 6 FAILED TESTS
make[6]: *** [test/CMakeFiles/check] Error 1
make[5]: *** [test/CMakeFiles/check.dir/all] Error 2
make[4]: *** [test/CMakeFiles/check.dir/rule] Error 2
make[3]: *** [check] Error 2
make[2]: *** [O3D3XX_CAMERA-prefix/src/O3D3XX_CAMERA-stamp/O3D3XX_CAMERA-test] Error 2
make[1]: *** [CMakeFiles/O3D3XX_CAMERA.dir/all] Error 2
make: *** [all] Error 2

tpanzarella, 2016-11-09:
You need to make sure you are running a supported software stack -- I.e,
libo3d3xx version and IFM firmware version. See the projects main page or
README for the software compatibility matrix.

Yarion, 2016-11-09:
Thank you for the quick response.

I used the wrong version of the lib. Downloaded the version 4.5 and compiled it without any trouble.

closed 2016-11-09 by Yarion