ifm/libo3d3xx #74: Cross compilation for Raspberry Pi - pthread_create

dlaurent12, 2016-11-14:
The computer I use to cross compile is Debian stretch/sid.

The rootfs is based on the last raspbian (26-09-2016) based on Debian Jessie 8.

when I call cmake an error message appears linked to pthread_create. Probably the same with is describe in the article cross compile.

Please find attach the log files.

Best Regards

graugans, 2016-11-14:
You may forgot to attach the logs?

dlaurent12, 2016-11-14:
[logfile.zip](https://github.com/lovepark/libo3d3xx/files/589088/logfile.zip)

dlaurent12, 2016-11-17:
Solved. Issue in toolchains.cmake file

closed 2016-11-17 by dlaurent12

graugans, 2016-11-17:
@dlaurent12 I am glad to here you solved zour problem. Could you please be so kind and share your toolchain and maybe some details in form of a README? For example as a gist or in this issue. So others can use this as a starter.